agherzan/meta-raspberrypi #950: Is there a plan for CM4 dunfell to support kernel version 5.10?

bigbearishappy, 2021-11-19:
I can find that the branch master has supported the kernel 5.10.But the current LTS line dunfell still don't support the 5.10.
And I want to use the dunfell with 5.10.
Is there a plan for dunfell to support the 5.10 kernel?

agherzan, 2021-11-29:
@bigbearishappy That would be a pretty major upgrade and it would break the assumptions on an LTS branch. For now, I recommend using a `mixin` layer that bumps the kernel version (and probably the firmware too) and set the kernel version preference manually (or through a distro configuration for example).

closed 2021-11-29 by agherzan

bigbearishappy, 2021-11-30:
I got it. Thank you for your reply.
For now.I use the branch master of meta-raspberrypi and other layers are in branch of dunfell.
It can work fine after I make some change to both meta-raspberrypi and other layers.So I think it's not a good idea.
I will try your suggesstion later.
Thank you again.