agherzan/meta-raspberrypi #952: CAN-Bus Driver for PI4

lemmi25, 2021-11-19:
**Description**

I am trying to add CAN-Bus support to my Pi4. 

I use a [PiCAN 3 Board](https://www.elektor.com/pican-3-can-bus-board-for-raspberry-pi-4-with-3-a-smps-rtc). I know it is not officially tested (not in the docu) but the config setup is the following:

```
dtparam=spi=on
dtoverlay=mcp2515-can0,oscillator=16000000,interrupt=25
dtoverlay=spi-bcm2835-overlay 
```

And this is not to different from what is written in the [docu](https://meta-raspberrypi.readthedocs.io/en/latest/extra-build-config.html#enable-can) (except the last line). I Added it manually.

this is my yocto local.config setup:

```
MACHINE ??= "raspberrypi4-64"
ENABLE_UART = "1"
ENABLE_SPI_BUS = "1"
ENABLE_CAN = "1"

...

RPI_EXTRA_CONFIG = ' \n\
# CAN Custom config \n\
dtoverlay=spi-bcm2835-overlay \n\'

...
```

The repo to all my PI4 Yocto Setup can be found [here](https://github.com/lemmi25/yocto-docker-buildenv).

When I start the Pi4 the **can0** interface is not showing up.

```
ifconfig -a
```

is not listing **can0**. And when I look into the boot output:

```
dmesg | fgrep -i mcp
```

nothing shows up. Also for:

```
dmesg | fgrep -i can
```

Maybe someone can help me how to set up the driver properly. My local.conf can be found [here](https://github.com/lemmi25/yocto-docker-buildenv/blob/master/conf/local.conf).

Many thanks 

Moritz

lemmi25, 2021-11-22:
Hi everyone,

I managed to solve it. I was building with 

```
core-image-minimal
```

after changing to 

```
core-image-base
```

it worked. Thanks anyway for this cool yocto layer ;).

closed 2021-11-22 by lemmi25

agherzan, 2021-11-29:
You were missing the kernel modules. I'm glad you have sorted it out.