agherzan/meta-raspberrypi #957: Kernel 5.10.78 bug with vc4-kms-v3d on Raspberrypi cm3+

newmedFabio, 2021-11-22:
Hello,
I've spent last week trying to figure out what was wrong, hope this post will avoid all that time to someone else.

I'm trying to build up a simple core-image-weston with Pocky Honister for a raspberry cm3+ board.
This is my bblayer.conf:

_POKY_BBLAYERS_CONF_VERSION = "2"
BBPATH = "${TOPDIR}"
BBFILES ?= ""
BBLAYERS ?= " \
  /home/fabio/poky-honister/meta \
  /home/fabio/poky-honister/meta-poky \
  /home/fabio/poky-honister/meta-openembedded/meta-oe \
  /home/fabio/poky-honister/meta-openembedded/meta-networking \
  /home/fabio/poky-honister/meta-openembedded/meta-multimedia \
  /home/fabio/poky-honister/meta-openembedded/meta-python \
  /home/fabio/poky-honister/meta-raspberrypi \
  "_

This is the uncommented local.conf that differs from default only for a couple of lines at the end:

_MACHINE ??= "raspberrypi-cm3"
DISTRO ?= "poky"
PACKAGE_CLASSES ?= "package_rpm"
EXTRA_IMAGE_FEATURES ?= "debug-tweaks"
USER_CLASSES ?= "buildstats"
PATCHRESOLVE = "noop"
BB_DISKMON_DIRS ??= "\
    STOPTASKS,${TMPDIR},1G,100K \
    STOPTASKS,${DL_DIR},1G,100K \
    STOPTASKS,${SSTATE_DIR},1G,100K \
    STOPTASKS,/tmp,100M,100K \
    ABORT,${TMPDIR},100M,1K \
    ABORT,${DL_DIR},100M,1K \
    ABORT,${SSTATE_DIR},100M,1K \
    ABORT,/tmp,10M,1K"
PACKAGECONFIG:append:pn-qemu-system-native = " sdl"
CONF_VERSION = "2"
MACHINE_ESSENTIAL_EXTRA_RDEPENDS += " kernel-modules udev-rules-rpi"
EXTRA_IMAGE_FEATURES += " ssh-server-openssh"_

**What does not work:**
After correctly building and flashing, I can only see a quick glimpse of boot writings since the displays truns off.
Then I connect with ssh to the raspberry and run: 
`dmesg | grep vc4`

The output is:

[image]

It seems that gpu vc4 driver can't be loaded for a vec issue.
After serching for this issue I found this:

[https://github.com/raspberrypi/linux/issues/4693](https://github.com/raspberrypi/linux/issues/4693)
And after, this:
[https://github.com/raspberrypi/linux/commit/46c99e3d7cf38446491b3c2a826fc05dcebc588d](https://github.com/raspberrypi/linux/commit/46c99e3d7cf38446491b3c2a826fc05dcebc588d)

Can you please fix also for meta-raspberry?

closed 2021-11-23 by agherzan

newmedFabio, 2021-11-29:
I was able to try with the new kernel only today.
I've modified meta-raspberry with new 5.10.81 version, unfortunately the issue is still there.

[image]

Tomorrow I'll try to downgrade to 5.10.17, which is working on the same hardware with a raspberrypi os distribution and let you know.

Thanks

kraj, 2021-11-30:
> I tried. Vc4 works fine with Kernel 5.10.17. The only thing I've changed is kernel version in meta-raspberrypi/recipes-kernel/linux/linux-raspberrypi_5.10.bb: SRCREV_machine = "ec967eb45f8d4ed59bebafb5748da38118383be7" SRCREV_meta = "5833ca701711d487c9094bd1efc671e8ef7d001e"
> 
> [image]
> 
> Now I should investigate what is wrong with recent kernels....

seems some sort of regression, perhaps start with device tree.

tewarid, 2021-12-01:
Seeing the same issue on raspberrypi0-wifi with master and honister branches

```
sh-5.1# uname -a 
Linux raspberrypi0-wifi 5.10.81 #1 Mon Nov 22 14:00:03 UTC 2021 armv6l

sh-5.1# dmesg | grep vc4
[    2.196533] fb0: switching to vc4drmfb from simple
[    2.200995] vc4-drm soc:gpu: bound 20400000.hvs (ops vc4_hvs_ops)
[    3.152202] vc4-drm soc:gpu: bound 20400000.hvs (ops vc4_hvs_ops)
[    3.288743] vc4-drm soc:gpu: bound 20400000.hvs (ops vc4_hvs_ops)
[    3.326939] rc rc0: vc4 as /devices/platform/soc/20902000.hdmi/rc/rc0
[    3.334095] input: vc4 as /devices/platform/soc/20902000.hdmi/rc/rc0/input0
[    3.348383] vc4-drm soc:gpu: bound 20902000.hdmi (ops vc4_hdmi_ops)
[    3.360491] [drm:vc4_vec_bind] *ERROR* Failed to get clock: -2
[    3.366781] vc4-drm soc:gpu: failed to bind 20806000.vec (ops vc4_vec_ops): -2
[    3.396633] vc4-drm soc:gpu: master bind failed: -2
[    3.401909] vc4-drm: probe of soc:gpu failed with error -2
```

Does not happen with hardknott branch

```
sh-5.1# uname -a 
Linux raspberrypi0-wifi 5.10.31 #1 Fri Apr 23 15:16:49 UTC 2021 armv6l

sh-5.1# dmesg | grep -i vc4
[    2.193210] vc4-drm soc:gpu: bound 20400000.hvs (ops vc4_hvs_ops)
[    3.131373] vc4-drm soc:gpu: bound 20400000.hvs (ops vc4_hvs_ops)
[    3.262615] vc4-drm soc:gpu: bound 20400000.hvs (ops vc4_hvs_ops)
[    3.301256] rc rc0: vc4 as /devices/platform/soc/20902000.hdmi/rc/rc0
[    3.308438] input: vc4 as /devices/platform/soc/20902000.hdmi/rc/rc0/input0
[    3.324300] debugfs: Directory '20902000.hdmi' with parent 'vc4-hdmi' already present!
[    3.342159] vc4-drm soc:gpu: bound 20902000.hdmi (ops vc4_hdmi_ops)
[    3.349231] vc4-drm soc:gpu: bound 20806000.vec (ops vc4_vec_ops)
[    3.357324] vc4-drm soc:gpu: bound 20004000.txp (ops vc4_txp_ops)
[    3.364272] vc4-drm soc:gpu: bound 20206000.pixelvalve (ops vc4_crtc_ops)
[    3.372725] vc4-drm soc:gpu: bound 20207000.pixelvalve (ops vc4_crtc_ops)
[    3.380357] vc4-drm soc:gpu: bound 20807000.pixelvalve (ops vc4_crtc_ops)
[    3.388550] vc4-drm soc:gpu: bound 20c00000.v3d (ops vc4_v3d_ops)
[    3.396661] fb0: switching to vc4drmfb from simple
[    3.429712] [drm] Initialized vc4 0.0.0 20140616 for soc:gpu on minor 0
[    3.584515] vc4-drm soc:gpu: [drm] fb0: vc4drmfb frame buffer device
[    3.609076] #0: vc4-hdmi
```

@agherzan should perhaps keep this issue open for now.

agherzan, 2022-01-04:
Thanks @tewarid for looking into this,

PR #971 (2021-12-08): linux-raspberrypi: Upgrade to 5.10.83
Upgrade linux-raspberrypi to 5.10.83.

Fixes #957.
commit ce12544b (2021-12-08): linux-raspberrypi: Upgrade to 5.10.83

Fixes #957
PR #975 (2021-12-26): linux-raspberrypi: Upgrade to 5.10.83
Fixes #957
commit 378d4b6e (2021-12-26): linux-raspberrypi: Upgrade to 5.10.83

Fixes #957