agherzan/meta-raspberrypi #959: Using raspberry pi camera through picamera or any other library

bhargavthriler, 2021-11-25:
**Description**
Currently controlling raspberry pi camera is quite straight forward using picamera module if rasberry pi os is used. But I want to use the camera using yocto. picamera module is not for yocto. I even tried creating a recipe for picamera but due to some other dependencies it doesn't seem to work.

Is there any other library or tool to control the camera.

agherzan, 2021-11-29:
I reckon you are referring to https://github.com/waveform80/picamera, right? I don't think there is an integration of it in Yocto yet so we would need someone to look into it. If you get it working, feel free to submit a PR.

bhargavthriler, 2021-12-04:
I will look into it. The picamera seems to be building fine but it needs some other modules to work. Will update the progress here.

bhargavthriler, 2021-12-04:
Btw is there any other way to use the raspberry pi camera from yocto (poky distro) ?

bhargavthriler, 2021-12-06:
**Update** : picamera needs the raspberrypi firmware's (https://github.com/raspberrypi/firmware) precompiled shared libraries, working on it.

@agherzan Btw don't we have these precompiled binaries for aarch64 ? since raspberrypi os now has 64bit support.

bhargavthriler, 2021-12-24:
Everything is working after porting the picamera module. Now I just have to organize the recipes and the raise PR.

PR #983 (2022-01-25): Add picamera support
python3-picamera : recipe for python picamera library.
python3-colorzero: recipe for python colorzero library, required by picamera
picamera-libs    : recipe raspberry pi firmware which are required by picamera.

Closes: #959

**- What I did**
Created recipe for python picamera library. Along with it created 2 more, python colorzero and picamera-libs.
commit bfac0b9c (2022-01-25): python3-picamera: Add picamera library

python3-picamera : recipe for python picamera library.
picamera-libs    : recipe raspberry pi firmware which are required
by picamera.

Closes: #959
closed 2022-01-25 by agherzan