agherzan/meta-raspberrypi #965: raspberrypi0-2w-32

sm-oralnum, 2021-12-03:
Hello, I  writed a conf file  to get the raspberrypi0-2 to work in 32 mode to have the camera packages raspivid. It works just fine. But the problem I don't have wlan0 interface. I tried everything but I didn't succeed in getting to work. Here is my conf file:

#@TYPE: Machine
#@NAME: RaspberryPi0 2 Wifi Development Board
#@DESCRIPTION: Machine configuration for the RaspberryPi0 2 Wifi in 32 bits mode

include conf/machine/raspberrypi3.conf

MACHINEOVERRIDES := "${@'${MACHINEOVERRIDES}'
.replace(':${MACHINE}',':raspberrypi3:${MACHINE}')}"

MACHINE_EXTRA_RRECOMMENDS += "\
    linux-firmware-rpidistro-bcm43436 \
    linux-firmware-rpidistro-bcm43436s \
    bluez-firmware-rpidistro-bcm43430b0-hcd \
    linux-firmware-rpidistro-bcm43430 \
"
also i tried to add following lines  to machine conf file.

RPI_KERNEL_DEVICETREE = " \
    broadcom/bcm2710-rpi-zero-2.dtb \
    "
But i get an error :
| make[2]: *** No rule to make target 'arch/arm/boot/dts/broadcom/bcm2710-rpi-zero-2.dtb'.  Stop.

It works just fine. The problem that I have is with the interface wlan0. I don't know why i don't have it ?

maurosanjo, 2021-12-10:
Hello, the problem is you need the dtb, when in 32 bits file structure is different. Take a look at my PR: #972

sm-oralnum, 2021-12-10:
Hello, i got it fixed a few days ago.

Best regards,

Safouane

Envoyé à partir de Courrier<https://go.microsoft.com/fwlink/?LinkId=550986> pour Windows

Envoyé le :vendredi 10 décembre 2021 10:47
À : ***@***.***>
Cc : Safouane ***@***.***>; ***@***.***>

agherzan, 2022-01-04:
Support is now available in master.

closed 2022-01-04 by agherzan