agherzan/meta-raspberrypi #968: Display_rotate=1 doesn't work for 1080p on CM3+ with u-boot

timemaster5, 2021-12-06:
**Description**

When I use DISPLAY_VC4GRAPHICS=1 together with display_rotate=1, the picture is damaged.
Without u-boot, all is fine.

**Steps to reproduce the issue:**
1. DISABLE_VC4GRAPHICS=1
2. RPI_USE_U_BOOT = "1"
3. hdmi_group=1, hdmi_mode=16 

**Describe the results you received:**
The Linux kernel splash screen renders unreadable.

**Describe the results you expected:**
Correctly showed a readable picture.

**Additional information you deem important (e.g. issue happens only occasionally):**
I am on Dunfell, would like to test master as there is a newer u-boot. But don't think it will change much.
720p and display_rotate=1 is perfectly fine. I think it has to do something with the framebuffer size.

**Additional details (revisions used, host distro, etc.):**
Dunfell branch
Compute Module 3+ (I would like to confirm the same behaviour on RPi3+).

If I disable u-boot, it is fine with the original loader, which I assume knows how to set up framebuffer correctly.

agherzan, 2022-01-04:
There seems to be multiple issues when using u-boot and I don't think anyone managed to dig into them properly. Have you raised this with the uboot community?

timemaster5, 2022-01-10:
Sorry for the late reply. No, do you know where to send bug reports? I found the source code here on GitHub, but the issues are disabled.

Also, do you know about some other bugs in u-boot on raspberry which we need to resolve? Or is it even interesting to develop u-boot support? (this is more about this BSP layer goal). I know their bootloader is getting better and could be sufficient for many things, but I still see a lot of flexibility in u-boot.

agherzan, 2022-01-10:
@timemaster5 If you search the issues in this repository you will find a set of them on uboot. The upstream community is available on their mailing list: https://www.denx.de/wiki/U-Boot/Patches

timemaster5, 2022-01-13:
Ok, it was a stupid question from me, I know... Thank you :)

What is the strategy now? I saw you recommended someone to raise an issue with the u-boot community, but as they are hard to reach for classic GitHub users, wouldn't it be a solution to add required patches here into this layer?

It seems like this layer is an excellent place to patch things, and the original author can then do the official support based on our internal work. If I look into Weston, for example, it is the same. However, patches for raspberry and userland stuff are implemented here, not in the Weston main tree.

I am asking because I can imagine some of the u-boot bugs we have here have a solution already that could be included in this layer instead of dealing with mailing lists. I have started working on this one and hope for a good result soon, and I want to patch it here ideally and potentially send the u-boot team a link to the mailing list so they know about the problems and fixes.

agherzan, 2022-02-23:
@timemaster5 Generally I welcome patches on top of third part components but the issue generally is maintenance. I don't what those patches to start adding maintenance work with each uboot bump. And certainly don't want to lock the uboot version here either. Generally, I'm fine with patches that were submitted upstream and they are planned to be included in later releases.

timemaster5, 2022-02-23:
Good point, thank you for the explanation. I will try to propose my patch to the u-boot upstream.

agherzan, 2022-02-24:
@timemaster5 Do keep us updated. Thanks for understanding.