agherzan/meta-raspberrypi #969: mesa: build fails with clang after #934

tewarid, 2021-12-07:
Build of machine raspberrypi4 on master fails in recipe mesa_21.3.0 after #934. See [log.do_compile.2608103.txt](https://github.com/agherzan/meta-raspberrypi/files/7670761/log.do_compile.2608103.txt). Build works without the commit. Building with poky master at 6245f65fa300f7f3e4cc13c7fe272a9d16668485 and meta-clang master at 1b1640576a5e299d8675024b083dd794f5f082c9.

kraj, 2021-12-07:
```
../mesa-21.3.0/src/broadcom/vulkan/v3dv_bo.c:71:15: error: format specifies type 'long' but the argument has type 'time_t' (aka 'long long') [-Werror,-Wformat]
              time.tv_sec);
              ^~~~~~~~~~~
```

This is in newly added code and happens with musl, since it has 64bit time_t for all 32bit systems as well.

kraj, 2021-12-07:
@tewarid can you test this patch in oe-core https://github.com/YoeDistro/openembedded-core/commit/4528b3c04acfb7872dd188688bd798ca6a2fc498

tewarid, 2021-12-07:
> @tewarid can you test this patch in oe-core [YoeDistro/openembedded-core@4528b3c](https://github.com/YoeDistro/openembedded-core/commit/4528b3c04acfb7872dd188688bd798ca6a2fc498)

Builds without error. Thanks.

closed 2021-12-09 by agherzan