agherzan/meta-raspberrypi #970: Touchscreen doesn't work if keyboard or mouse plugged on on start up

GWLindberg, 2021-12-07:
**Description**

We are building for Yocto Zeus, raspberry pi4.
The touchscreen works properly unless a mouse or keyboard is plugged in on boot. The touch screen continues to work properly idfa mouse or keyboard is plugged in after the system is up and running.

1. Connect touchscreen
2. boot system, touchscreen works
2. Connect Keyboard
3. reboot system, no touchscreen

**Describe the results you expected:**

We would like to be able to use the touchscreen even if a keyboard or mouse is plugged in when the system boots.

**Additional details (revisions used, host distro, etc.):**

Yocto Poky version 3.0.4, Zeus, compiled on Ubuntu 18.04

agherzan, 2021-12-09:
Zeus is not a version that we maintain anymore. Have you tried honister or master?

GWLindberg, 2021-12-09:
Hello Andrei,
We are using the QT boot2qt version of the Yocto system, so I am stuck on zeus for now. The next version of boot2qt will be using Hardknot.

Do you know if there are any changes between zeus and the current master that might affect this?

Regards,
Greg

agherzan, 2021-12-09:
It's very hard to say given how old zeus is. What is the graphical stack you are using?

GWLindberg, 2021-12-09:
The top level is EGL. I don't know what the pieces in the middle are, boot2qt does not run a window manager, just the application.

Is there a way to figure what is loaded for that?

aurelian17, 2021-12-10:
@GWLindberg Indeed, the bottom level is EGL. The interface between Qt and the EGL is the QPA_PLUGIN, I guess in your case is eglfs. Qt has its own way of handling the touch/mouse events. It's the tslib that does the magic. I suggest starting investigating from there.

GWLindberg, 2021-12-10:
Hi Aurelian,
Thank you for the insight, I will contact Qt support to try to figure out what is going on there.

Regards,
Greg

agherzan, 2022-01-04:
I'm going to close this issue as zeus is not a branch we maintain anymore. Nevertheless, I'm happy to pull in changes that would fix this on zeus as long as you manage to figure out the issue.

closed 2022-01-04 by agherzan