agherzan/meta-raspberrypi #974: No UART output on raspberry pi 3

bhargavthriler, 2021-12-10:
**Description**
No UART output on raspberry pi 3

Can anyone confirm if raspberry pi 3 UART out is working with the latest builds.But 
I have not tested the HDMI out. But the green and red led are blink which confirm that the board is booting.

My last build was working fine, these are commit of the last build
poky                          : `bcc9d8f837d6c1c69d1321e65aa5c74c6dd3dfab` 
meta-raspberrypi      : `63a3d8cb17c5d1affe8f2848f45fcc6a706f9412`
meta-openembedde : `e0ab236378f8a734b72e3b392c900540bd41f87e`

**Configs**
```
MACHINE = "raspberrypi3"
ENABLE_UART = "1"
RPI_USE_U_BOOT = "1"
GPU_FREQ = "250"
```

timemaster5, 2021-12-12:
Hello, it does work well for me. Try to set up serial consoles like this:

SERIAL_CONSOLES = "115200;ttyAMA0"

bhargavdas, 2021-12-13:
> Hello, it does work well for me. Try to set up serial consoles like this:
> 
> SERIAL_CONSOLES = "115200;ttyAMA0"

Is this change added recently ? earlier it worked fine without this.

bhargavthriler, 2021-12-13:
> Hello, it does work well for me. Try to set up serial consoles like this:
> 
> SERIAL_CONSOLES = "115200;ttyAMA0"

Is this change added reccently ? earlier it worked fine.

bhargavthriler, 2021-12-14:
@timemaster5 
Where did you change the serial console's setup ?
I added in local.conf `SERIAL_CONSOLES = "115200;ttyAMA0"`   -  No output
Manually changed `boot/config.txt` from `console=serial0,115200` to `console=ttyAMA0,115200`  - Still no output 
Can you share your local.conf 's additions ?

bhargavthriler, 2021-12-24:
Now working fine latest build after changing multiple sd cards, didn;t understand what was the issue. Maybe my sd card had gave up . Closing this issue.

closed 2021-12-24 by bhargavthriler