agherzan/meta-raspberrypi #977: Can't build one machine with userland and another with vc4 drivers

timemaster5, 2021-12-12:
**Description**

I would like to have two machines, one would use userland and one vc4. I can build them one by one if I remove tmp directory, but if I want to build both continuously, I receive this error:

```
ERROR: mesa-gl-2_20.0.2-r0 do_package_write_ipk: The recipe mesa-gl is trying to install files into a shared area when those files already exist. Those files and their manifest location are:
  /home/builder/smartboxos/build/tmp/deploy/ipk/cortexa7t2hf-neon-vfpv4/libglapi0_20.0.2-r0_cortexa7t2hf-neon-vfpv4.ipk
    (matched in manifest-cortexa7t2hf-neon-vfpv4-mesa.package_write_ipk)
  /home/builder/smartboxos/build/tmp/deploy/ipk/cortexa7t2hf-neon-vfpv4/libgl-mesa-dev_20.0.2-r0_cortexa7t2hf-neon-vfpv4.ipk
    (matched in manifest-cortexa7t2hf-neon-vfpv4-mesa.package_write_ipk)
  /home/builder/smartboxos/build/tmp/deploy/ipk/cortexa7t2hf-neon-vfpv4/libgbm1_20.0.2-r0_cortexa7t2hf-neon-vfpv4.ipk
    (matched in manifest-cortexa7t2hf-neon-vfpv4-mesa.package_write_ipk)
  /home/builder/smartboxos/build/tmp/deploy/ipk/cortexa7t2hf-neon-vfpv4/libegl-mesa-dev_20.0.2-r0_cortexa7t2hf-neon-vfpv4.ipk
    (matched in manifest-cortexa7t2hf-neon-vfpv4-mesa.package_write_ipk)
  /home/builder/smartboxos/build/tmp/deploy/ipk/cortexa7t2hf-neon-vfpv4/libgbm-dev_20.0.2-r0_cortexa7t2hf-neon-vfpv4.ipk
    (matched in manifest-cortexa7t2hf-neon-vfpv4-mesa.package_write_ipk)
  /home/builder/smartboxos/build/tmp/deploy/ipk/cortexa7t2hf-neon-vfpv4/libglapi-dev_20.0.2-r0_cortexa7t2hf-neon-vfpv4.ipk
    (matched in manifest-cortexa7t2hf-neon-vfpv4-mesa.package_write_ipk)
  /home/builder/smartboxos/build/tmp/deploy/ipk/cortexa7t2hf-neon-vfpv4/mesa-megadriver_20.0.2-r0_cortexa7t2hf-neon-vfpv4.ipk
    (matched in manifest-cortexa7t2hf-neon-vfpv4-mesa.package_write_ipk)
Please verify which recipe should provide the above files.

The build has stopped, as continuing in this scenario WILL break things - if not now, possibly in the future (we've seen builds fail several months later). If the system knew how to recover from this automatically it would, however there are several different scenarios which can result in this and we don't know which one this is. It may be you have switched providers of something like virtual/kernel (e.g. from linux-yocto to linux-yocto-dev), in that case you need to execute the clean task for both recipes and it will resolve this error. It may be you changed DISTRO_FEATURES from systemd to udev or vice versa. Cleaning those recipes should again resolve this error, however switching DISTRO_FEATURES on an existing build directory is not supported - you should really clean out tmp and rebuild (reusing sstate should be safe). It could be the overlapping files detected are harmless in which case adding them to SSTATE_DUPWHITELIST may be the correct solution. It could also be your build is including two different conflicting versions of things (e.g. bluez 4 and bluez 5 and the correct solution for that would be to resolve the conflict. If in doubt, please ask on the mailing list, sharing the error and filelist above.
ERROR: mesa-gl-2_20.0.2-r0 do_package_write_ipk: If the above message is too much, the simpler version is you're advised to wipe out tmp and rebuild (reusing sstate is fine). That will likely fix things in most (but not all) cases.
```

**Steps to reproduce the issue:**
1. create two machines
2. one with DISABLE_VC4GRAPHICS = "1"
3. one with default configuration

**Describe the results you received:**
A mesa-gl package collides somehow between those two builds.

**Describe the results you expected:**
I think this should work, I haven't found any change of DISTRO_FEATURES or MACHINE_FEATURES, which is unsupported to do between builds, but this should be only about choosing virtual providers. 

**Additional information you deem important (e.g. issue happens only occasionally):**
I am on dunfell currently.

**Additional details (revisions used, host distro, etc.):**
compute module 3

timemaster5, 2021-12-18:
I have temporarily fixed this by adding:

```
SSTATE_DUPWHITELIST += "\
${DEPLOY_DIR_IPK}/\
"
```

to the mesa and mesa-gl bbappend.

I know this is a wrong solution, but also not that bad as these two recipes are meant to serve the same-named files.

I couldn't find a correct full path for this, though, so I had to add the whole ${DEPLOY_DIR_IPK}. 
I can imagine for someone who uses RPM's. He will need to add ${DEPLOY_DIR_RPM}.

Some better resolution is very welcome, but since I don't see anything better, I am happy to build for two machines without deleting tmp.

agherzan, 2022-01-04:
This indeed shouldn't happen. Can you reproduce this on master?

timemaster5, 2022-01-14:
It's not easy to try for me now. My distro is on Dunfell, I've tried to just checkout master, and add a Dunfell as a compatible version to the layer.conf, but without success. So I made a new clean build env, but there I am failing on the issue reported here: https://github.com/agherzan/meta-raspberrypi/issues/987

timemaster5, 2022-01-15:
Ha, the same for honister.

timemaster5, 2022-02-04:
OK, so far, this is not happening on Honister, and expecting a master is also ok. So I will probably leave it here, and once we upgrade to Honister, I will check again whether the issue persists or not.

timemaster5, 2022-02-12:
I am sorry, the first answer related to Honister was that I couldn't build Honister and master at all, as both had the same error during compilation as described in #987. In that bug, I have received a patch, so I was able to continue testing, and once I was able to compile Honister, I haven't seen this issue so far.

agherzan, 2022-02-12:
So you have tested it in `honister` in the meanwhile and you couldn't reproduce it anymore? I'm asking because I suspect that this was fixed in core.

timemaster5, 2022-02-12:
I tested in `honister` once I resolved the compilation issue from bug #987, I couldn't test before. But yes, correct, I couldn't reproduce it in `honister`. It is great if this has been fixed in the core already, do you have a concrete patch in mind? I won't be able to upgrade my project from `dunfell` soon, so maybe I could backport their fix.

agherzan, 2022-02-12:
@timemaster5 it's not an easy one to figure out without digging into it a bit. All I can see is that it seems like both `mesa` and `mesa-gl` are pulled in. I would suspect that you only want `mesa-gl` so I'd start by investigating what pulls in the other by checking the dependency graph

timemaster5, 2022-02-14:
I've spent already quite some time on this problem, and couldn't find it. That's why I am interested in any clue on what could be different in the current core. I suspect it has something to do with the rework of the overrides system. I've been through the dependency charts and so on, and I think the configuration is OK. `mesa` and `mesa-gl` are being pulled in as resolution of a virtual package `mesa` if I remember it correctly, and this configuration is no different from the configuration in `honister`. Based on the VC4 variable, the system decides which one to use, and that's it basically.

agherzan, 2022-02-19:
@timemaster5 are you building this with a minimum layer set? oe-core/poky + meta-raspberrypi? If yes, can you describe a minimum setup/configuration to try to reproduce?