agherzan/meta-raspberrypi #979: hot set raspberry pi4  wifi as access point(hotspot) in yocto image

gitgonewithwind, 2021-12-15:
sorry to bother you all,
my colleagues want to use the p4 as an access point(hotspot) in my yocto image. 
before I misunderstood it and thought we only use wifi to connect routers with wifi,but it's wrong. how Can I do?
do you have some guide about what components should add in con/local.conf and how to config it?
I searched online, but not clear now.
any help?

Thank you very much.

lesti, 2021-12-23:
@gitgonewithwind
This sounds like it's not really a problem with that layer.
However, it seems you'd want to use something like `hostapd` (already available for oe), copy in a config file and create a init file so it runs at startup.
So the steps you want to run are roughly the following:
- add `hostapd` to `IMAGE_INSTALL:append` (plus maybe `dnsmasq`, if you need the Pi acting as DHCP server)
- create the corresponding config files (examples can be found [here](https://w1.fi/cgit/hostap/plain/hostapd/hostapd.conf) and [here](https://github.com/imp/dnsmasq/blob/master/dnsmasq.conf.example), adapt to your needs)
- copy the configs into the image via your own recipe (and add that to your `local.conf`)
- (if it doesn't work it's maybe your wifi country not set, e.g. RaspberryPi OS behaves a bit weird regarding this)
  - you should also be able to achieve this in `local.conf`
- the init file depends on the init system you're using, examples should be easy to find

gitgonewithwind, 2021-12-23:
@lesti 
thanks
I done .

please close it. 
I post my solution(no bridge) here if somebody needs it later.
refer: https://thepi.io/how-to-use-your-raspberry-pi-as-a-wireless-access-point/

add below lines at local.conf
CORE_IMAGE_EXTRA_INSTALL += " \
 
       hostapd dnsmasq \
         "
modified below files in /etc    
dnsmasq.conf  
dhcpcd.conf  hostapd.conf
/etc/default/hostapd
/etc/network/interface

file content:
1. /etc/default/hostapd:
EMON_CONF="/etc/hostapd.conf"
2. /etc/dhcpcd.conf
interface wlan0
static ip_address=192.168.0.10/24
3./etc/dnsmasq.conf 
interface=wlan0
  dhcp-range=192.168.0.11,192.168.0.30,255.255.255.0,24h
4./etc/hostapd.conf
interface=wlan0
driver=nl80211
ssid=my_ap
hw_mode=g
channel=6
macaddr_acl=0
auth_algs=1
ignore_broadcast_ssid=0
wpa=3
wpa_passphrase=1234%#@l7890
wpa_key_mgmt=WPA-PSK
wpa_pairwise=TKIP
rsn_pairwise=CCMP

5. /etc/network/interfaces
# The loopback interface
auto lo
iface lo inet loopback

# Wired or wireless interfaces
auto eth0
iface eth0 inet static
	address 192.168.1.100
	netmask 255.255.255.0

#auto wlan0                                                                 
#iface wlan0 inet dhcp                                                      
#wireless_mode managed                                                      
#  wireless_essid any 
#  wpa-driver nl80211
#  wpa-conf /etc/wpa_supplicant.conf

auto wlan0
iface wlan0 inet static
address 192.168.0.1
netmask 255.255.255.0

agherzan, 2022-01-04:
Indeed you can use hostapd and I'm pretty sure NetworkManager also support hotspot mode.

closed 2022-01-04 by agherzan