agherzan/meta-raspberrypi #982: U-Boot, and swupdate

wasabii, 2022-01-03:
Howdy.

Trying to put together a build for a RPI4 that supports using swupdate to update images with an A/B swap. Got everything working basically. rootfs on p2 and p3 swapped successfully.

What I'm trying to figure out now is how to get the kernel updating. And potentially u-boot itself.

So for the kernel, I think what might make sense is for me to disable /boot being a separate partition. Just include u-boot stuff on p1, but put kernel on 2/3 in /boot. I can't come up with a good reason to actually have the kernel in the boot partition, given uboot being the thing RPI needs to load.

Can there maybe just be an option for this? RPI_USE_U_BOOT_ROOTFS or something? It seems to make little sense to actually have a mountable /boot with the kernel when using UBoot.

How might I approach doing this myself?

agherzan, 2022-01-04:
@wasabii there are multiple ways of achieving what you want, You are probably looking for an atomic operation and if you do, you can take a look at tryboot in RaspberryPi bootloader. You can find an example as part of https://git.ostc-eu.org/OSTC/OHOS/components/sysota/ but sadly the official RaspberryPi documentation is lacking. You'll have to dig a bit through GitHub issues or so. Anyway, I hope this gives you a good lead.

closed 2022-01-04 by agherzan

wasabii, 2022-01-04:
Maybe I included too much information.

My question is really about how I can edit the meta-raspberrypi layer to drop the Kernel onto the rootfs instead of the bootfs.

agherzan, 2022-01-04:
Dropping it in the rootfs is the easiest part of it. The issue is how are you going to load it? Are you going to use uboot? Nevertheless, including the kernel image can be accomplished by packaging the image file in the kernel recipe. I'm pretty sure there is already an image package that gets generated. Check out the packages created by the linux-raspberrypi recipe.

agherzan, 2022-01-04:
Unless you have a requirement on uboot (and even if you do) you should really take a look at tryboot. That gives you all you need for rpi with a a/b strategy.

anthonio9, 2024-04-29:
Hi @wasabii  and @agherzan, I'm having a similar problem today - trying to do the A/B swap with RAUC instead of swupdate. It seems like including everything, the kernel, the device trees and other contents of /boot partition on the rootfs is the way to go. @wasabii how have you made that work in the end?
What was your way of telling uboot to look for the kernel in