agherzan/meta-raspberrypi #986: Using BOOT_SPACE to increase boot partition for placing cpio file

arrow53, 2022-01-13:
**Description**

**Steps to reproduce the issue:**
This is the relevant part of my `local.conf`
```
MACHINE ?= "raspberrypi4-64"

ENABLE_UART = "1"
RPI_USE_U_BOOT = "1"
ENABLE_SPI_BUS = "1"

DISTRO_FEATURES_remove = " x11 wayland vulkan directfb"
MACHINE_FEATURES_remove = " vc4graphics"
MACHINE_FEATURES_append = "tpm2"
DISTRO_FEATURES_append = " systemd "
VIRTUAL-RUNTIME_init_manager = "systemd"

RPI_EXTRA_CONFIG = "dtoverlay=lets-trust"

INITRAMFS_IMAGE = "tpm-initramfs-image"
INITRAMFS_IMAGE_BUNDLE = "1"
INITRAMFS_MAXSIZE = "315400"
IMAGE_FSTYPES_pn-${INITRAMFS_IMAGE} = "${INITRAMFS_FSTYPES}"
```

**Describe the results you received:**
the `/boot` partition size is always `62592`

**Additional details (revisions used, host distro, etc.):**

I'm working with dunfell

arrow53, 2022-01-15:
thanks @kraj i'll just use the sd image `IMAGE_FSTYPES = " rpi-sdimg"`

closed 2022-01-15 by arrow53

agherzan, 2022-01-17:
@arrow53 the SD card class might be one up for deprecation soon so I recommend you to go with the wic approach.