agherzan/meta-raspberrypi #987: MESA/drm.h no such file or directory - can't build core-image-base on master

timemaster5, 2022-01-14:
**Description**

Mesa can't include drm.h when building anything mesa-related on master

**Steps to reproduce the issue:**
1. follow readme
2. bitbake core-image-base

**Describe the results you received:**
```
| In file included from ../mesa-21.3.3/src/gbm/main/backend.c:37:
| /home/timemaster/raspberry-test/rpi-build/tmp/work/cortexa7t2hf-neon-vfpv4-poky-linux-gnueabi/mesa-gl/2_21.3.3-r0/recipe-sysroot/usr/include/xf86drm.h:40:10: fatal error: drm.h: No such file or directory
|    40 | #include <drm.h>
|       |          ^~~~~~~
| compilation terminated.
| ninja: build stopped: subcommand failed.
```
**Describe the results you expected:**
Clean build

EasyIP2023, 2022-01-24:
@timemaster5 what I was doing to bypass issue was

add into ``recipes-graphics/mesa/mesa-gl_%.bbappend``

```
TARGET_CFLAGS:append = " -I${STAGING_INCDIR}/drm"
```

and I added an ifdef inside ``src/gbm/main/backend.c``

```
#ifdef HAVE_DRI
extern const struct gbm_backend gbm_dri_backend;
#endif

static const struct gbm_backend_desc builtin_backends[] = {
#ifdef HAVE_DRI
  { "dri", &gbm_dri_backend },
#endif
};

```

Should get you past compile on master.

timemaster5, 2022-02-04:
Thank you for this. I confirm this helped.

What's the plan? Shall we do the patch and a pull request for this? The problem is in both the master and Honister branches.

agherzan, 2022-02-23:
@timemaster5 I've just built this `master` / `raspberrypi4` and it builds fine. Sounds like a fix landed in `core`.

closed 2022-02-23 by agherzan

timemaster5, 2022-05-31:
This is still valid at least for the current kirkstone. I saw a lot of movement in the core around mesa's versions, maybe it was fixed in one and broken again in the next. Anyway your point @EasyIP2023 is still helping @ 👍