agherzan/meta-raspberrypi #991: RPIDistro seems to change default branch

nandra, 2022-01-24:
With dunfell I'm getting:
```
[Step 1/1] ERROR: linux-firmware-rpidistro-0.0+gitAUTOINC+f0ad1a42b0-r0 do_fetch: Fetcher failure: Unable to find revision f0ad1a42b051aa9da1d9e1dc606dd68ec2f163a5 in branch master even from upstream
[Step 1/1] ERROR: linux-firmware-rpidistro-0.0+gitAUTOINC+f0ad1a42b0-r0 do_fetch: Fetcher failure for URL: 'git://github.com/RPi-Distro/firmware-nonfree'. Unable to fetch URL from any source.
[Step 1/1] ERROR: linux-firmware-rpidistro-0.0+gitAUTOINC+f0ad1a42b0-r0 do_fetch: 
[Step 1/1] ERROR: linux-firmware-rpidistro-0.0+gitAUTOINC+f0ad1a42b0-r0 do_fetch: Function failed: base_do_fetch
[Step 1/1] ERROR: Logfile of failure stored in: /home/builder/buildAgent/work/2bf058ae8f7d2c6d/.build-rpi3/tmp/work/all-poky-linux/linux-firmware-rpidistro/0.0+gitAUTOINC+f0ad1a42b0-r0/temp/log.do_fetch.11072
[Step 1/1] NOTE: recipe linux-firmware-rpidistro-0.0+gitAUTOINC+f0ad1a42b0-r0: task do_fetch: Failed
```

otski, 2022-01-24:
Have you pulled the latest changes to you local dunfell branch? The current SHA in dunfell is `98e815735e2c805d65994ccc608f399595b74438`
https://github.com/agherzan/meta-raspberrypi/blob/dunfell/recipes-kernel/linux-firmware-rpidistro/linux-firmware-rpidistro_git.bb#L38

Also the branch in the `SRC_URI` has changed from `master` to `buster`

nandra, 2022-01-24:
@otski ok thanks for info. Seems I was on older dunfell release.

closed 2022-01-24 by nandra