agherzan/meta-raspberrypi #996: my yocto image can not work for Model  Raspberry Pi 4 Model B Rev 1.4 2GB boards ,  but work well in Rev 1.2 4GB boards

gitgonewithwind, 2022-02-01:
Hi all,
I built the yocto  image about  2 years ago, it works well for  work well in Rev 1.2 4GB boards.
cat /proc/cpuinfo
...
Hardware        : BCM2711                                                     
Revision        : c03112                                                      
Serial          : 10000000e25f2e75                                            
Model           : Raspberry Pi 4 Model B Rev 1.2

but now we buy new board  Rev 1.4 2GB, boards , not work!
...
Hardware        : BCM2711                                                     
Revision        : b03114                                                      
Serial          : 10000000e7e3b6db                                            
Model           : Raspberry Pi 4 Model B Rev 1.4

how can I do?  do I have  to update to the  latest  my raspberry layer to  commit e1014d41ef459af807f95cb7ff9105f077bc6daa ?
it will take a long time for me to patch a lot for my rtc , chrome ... ,I think. 
do you have some shortcut?

Thank you very much.

agherzan, 2022-02-02:
I'd recommend you first try a minimum setup with the latest layers. I suspect you need an upgraded revision and you will need to adapt your layer/changes on top.

gitgonewithwind, 2022-02-02:
@agherzan 

thanks.
Do you mean I can ONLY change the latest meta-raspberry in my old yocto? and then  to change (adapt) to the other meta-(such as poky,meta-openembedded... )

I tried a little bit that I directly only changed the latest meta-raspberry in my old yocto. it could not pass the building. 

It seems I have to update everything,such as poky,meta-openembedded ... the latest yocto syntax has a few differences.

agherzan, 2022-02-02:
Yes. You will need to update the entire layers setup you have.

gitgonewithwind, 2022-02-02:
@agherzan 
got it. I did before to update all layers, but there was an issue to block me. https://github.com/agherzan/meta-raspberrypi/issues/944 somebody said he find the solution. I am building to try.

by the way, there is a little error, maybe you have fixed in today latest commit(now I am using commit e31fba4c96cdf819958d05875fb8733481997806 ):
ERROR: linux-firmware-rpidistro-20190114-1+rpt11-r0 do_fetch: Bitbake Fetcher Error: FetchError('Unable to fetch URL from any source.', 'git://github.com/RPi-Distro/firmware-nonfree;branch=master;protocol=https')

I think I need  to change  the website  to  https://github.com/RPi-Distro/firmware-nonfree  . I am doing it.

agherzan, 2022-02-23:
Both the issues you are referencing should be fixed in `master`. Do come back if you still encounter issues.

closed 2022-02-23 by agherzan