agherzan/meta-raspberrypi #997: Cannot set static eth0 IP address

ricbrue, 2022-02-02:
eth0 does not take fixed ip set on /etc/systemd/network configuration file.

I had followed a tutorial in order to set a static IP address to the eth0 interface, [here ](https://hub.mender.io/t/how-to-configure-networking-using-systemd-in-yocto-project/1097)are the steps, generally what I've done is:

1. Used "systemd";
2. Created custom layer with the "recipes-core/systemd" folder;
3. Created "systemd-conf_%.bbappend" file which copies .network files inside the "/etc/systemd/network/" folder;
4. Created the "eth.network" configuration file as follows:
```
[Match]
Name=eth0

[Network]
Address=192.168.10.200/24
Gateway=192.168.10.1
DNS=8.8.8.8
```
5. Built the image;
6. Boot

When I boot the board, it takes a DHCP IP ignoring the static IP set in the configuration files. How can I solve this issue?
I've already checked the following:

1. NetworkManager is not present in the image;
2. Tried to `systemctl restart systemd-networkd.service` without success, the IP remains to the one obrained by the DHCP;
3. Tried to specifically set "DHCP=no" in the configuration file;
4. Tried to use "eth*" in the configuration file, nothing to do;

agherzan, 2022-02-02:
That should work fine. Are you sure that there is no NM running? Are you sure that networkd is up and running? Sounds like something is conflicting there.

ricbrue, 2022-02-02:
> That should work fine. Are you sure that there is no NM running? Are you sure that networkd is up and running? Sounds like something is conflicting there.

Yes I am sure, here is my custom image recipe if it could help:

```
IMAGE_FEATURES += " \
    splash x11-base \
    hwcodecs ssh-server-openssh \
    tools-debug debug-tweaks \
    eclipse-debug tools-profile \
    tools-sdk nfs-server \
"

IMAGE_INSTALL += " \
    tslib libtar libcrypto qtbase qtbase-tools qtbase-plugins \
    qtcharts boost qwt-qt5 openssl sudo vim curl wget unzip glibc \ 
    binutils gcc packagegroup-core-buildessential \
    qtbase qtcharts qwt-qt5 openssh-sftp-server \
    net-tools ethtool evtest fbset \
    memtester boost-dev linux-firmware-rpidistro-bcm43430
" 
```

agherzan, 2022-02-23:
I will close this one for now but feel free to come back if you figure this out or you get some progress.

closed 2022-02-23 by agherzan