agherzan/meta-raspberrypi #1007: how to correctly disable rainbow colour display at bootup on pi4 hardware res 1.4 yocto image

gitgonewithwind, 2022-02-11:
hi all,
sorry to bother you all.
I am playing pi4 hardware res 1.4 boards and meta-raspberrypi commit commit e1014d41ef459af807f95cb7ff9105f077bc6daa
I changed meta-raspberrypi/recipes-core/psplash/psplash_%.bbappend to my company background picture.
but after startup, the rainbow colour display is still there. 
there is no progress bar for my company logo picutre at startup  and only I reboot the board, the progress bar is on display to my company logo.
but there older hardware res 1.2 worked well(older  meta-raspberrypi commit)

my questions are:
1.how to correctlly disable the rainbow display?
2 why no progress bar  picture at startup but only at reboot?

Thank you very much.

agherzan, 2022-02-11:
1. Use `disable_splash=1` in the `config.txt`.
2. You will need to debug that a bit more. It could be due to insufficient power to the display? Have you tried to debug the splash service at runtime when it doesn't show? Nevertheless, that would need some debugging.

gitgonewithwind, 2022-02-14:
@agherzan 
Thanks
1. works with disable_splash=1 in the config.txt   or DISABLE_SPLASH = "1" in local.conf  to disable rainbow color
2. ln -s  ../init.d/psplash.sh S00psplash.sh     in /etc/rc2.d     /etc/rc3.d /etc/rc4.d /etc/rc5.d   to show the progress bar in startup.  originally ,there is K00psplash.sh  in /etc/rc6.d so that it can show it at reboot

please close it

agherzan, 2022-02-19:
> @agherzan
> Thanks
> 1. works with disable_splash=1 in the config.txt   or DISABLE_SPLASH = "1" in local.conf  to disable rainbow color
> 2. ln -s  ../init.d/psplash.sh S00psplash.sh     in [...]

I'm glad it works. We sadly not C `sysvinit` configuratios and there are less and less users on that front. My advice is to switch to `systemd` where most of the support is validated.

closed 2022-02-19 by agherzan