agherzan/meta-raspberrypi #1010: MACHINE_FEATURES_append = " rtc" won't sync the sys clock to hwclock

nucatus, 2022-02-15:
I'm not sure if this functionality should be included or not in meta-raspberry.

To my understanding, setting `MACHINE_FEATURES_append = " rtc"` should add the full capability to the board, including setting the system time from the RTC device. At the moment, this is not possible without adding a dedicated init script / systemd unit to initialize the system time from the time stored in the RTC device _before_ trying the NTP synchronization.

Would it be wise to add such initialization logic in order to set the system time from the RTC device?

gitgonewithwind, 2022-02-15:
@nucatus  see here  https://github.com/agherzan/meta-raspberrypi/issues/913.  maybe  help you:
-----
after reading /etc/init.d/hwclock.sh , I add IMAGE_INSTALL_append = "tzdata" in local.conf and ln -s /usr/share/zoneinfo/America/Toronto /etc/localtime
I make it works well.

agherzan, 2022-02-15:
@nucatus if you check the bitbake environment, does `rtc` end up in `MACHINE_FEATURES`?

nucatus, 2022-02-16:
@agherzan,

`$ bitbake -e my-image | grep MACHINE_FEATURES`
```
# $ANY_OF_MACHINE_FEATURES
#     [doc] "When a recipe inherits the features_check class, at least one item in this variable must be included in MACHINE_FEATURES."
#     [doc] "A set of features common between MACHINE_FEATURES and DISTRO_FEATURES."
#     "${@oe.utils.set_intersect('DISTRO_FEATURES', 'MACHINE_FEATURES', d)}"
#     [vardeps] "DISTRO_FEATURES MACHINE_FEATURES"
#   "${@oe.utils.set_intersect('DISTRO_FEATURES', 'MACHINE_FEATURES', d)}"
# $CONFLICT_MACHINE_FEATURES
#     [doc] "When a recipe inherits the features_check class, no item in this variable can be included in MACHINE_FEATURES."
# $MACHINE_FEATURES [5 operations]
MACHINE_FEATURES=" apm usbhost keyboard vfat ext2 screen touchscreen alsa bluetooth wifi sdio vc4graphics rtc qemu-usermode"
# $MACHINE_FEATURES_BACKFILL [2 operations]
#     [doc] "Features to be added to MACHINE_FEATURES if not also present in MACHINE_FEATURES_BACKFILL_CONSIDERED. This variable is set in the meta/conf/bitbake.conf file and is not intended to be user-configurable."
MACHINE_FEATURES_BACKFILL="rtc qemu-usermode"
# $MACHINE_FEATURES_BACKFILL_CONSIDERED [2 operations]
#     [doc] "Features from MACHINE_FEATURES_BACKFILL that should not be backfilled (i.e. added to MACHINE_FEATURES) during the build."
MACHINE_FEATURES_BACKFILL_CONSIDERED="rtc"
#     "${@bb.utils.contains("MACHINE_FEATURES", "vc4graphics", "mesa", "userland", d)}"
#     "${@bb.utils.contains("MACHINE_FEATURES", "vc4graphics", "mesa", "userland", d)}"
#     "${@bb.utils.contains("MACHINE_FEATURES", "vc4graphics", "mesa", "mesa-gl", d)}"
#     "${@bb.utils.contains("MACHINE_FEATURES", "vc4graphics", "mesa", "mesa-gl", d)}"
#     "${@bb.utils.contains("MACHINE_FEATURES", "vc4graphics", "mesa", "mesa-gl", d)}"
#     "${@bb.utils.contains("MACHINE_FEATURES", "vc4graphics", "mesa", "userland", d)}"
#   "${@bb.utils.contains("MACHINE_FEATURES", "vc4graphics", "mesa", "userland", d)}"
#     "${@bb.utils.contains("MACHINE_FEATURES", "vc4graphics", "mesa", "mesa-gl", d)}"
#     "${@bb.utils.contains("MACHINE_FEATURES", "vc4graphics", "mesa", "mesa-gl", d)}"
#   "${@bb.utils.contains("MACHINE_FEATURES", "vc4graphics", "mesa", "mesa-gl", d)}"
#     "${@bb.utils.contains("MACHINE_FEATURES", "vc4graphics", "mesa", "userland", d)}"
#   "${@bb.utils.contains("MACHINE_FEATURES", "vc4graphics", "mesa", "userland", d)}"
#     "${@bb.utils.contains("MACHINE_FEATURES", "vc4graphics", "mesa", "mesa-gl", d)}"
#   "${@bb.utils.contains("MACHINE_FEATURES", "vc4graphics", "mesa", "mesa-gl", d)}"
# $REQUIRED_MACHINE_FEATURES
#     [doc] "When a recipe inherits the features_check class, all items in this variable must be included in MACHINE_FEATURES."
#     "     xserver-xorg     ${@bb.utils.contains("MACHINE_FEATURES", "vc4graphics", "xserver-xorg-extension-glx", "", d)}     ${@bb.utils.contains("MACHINE_FEATURES", "vc4graphics", "xf86-video-modesetting", "xf86-video-fbdev", d)}     "
#     [depends] "     parted-native:do_populate_sysroot     mtools-native:do_populate_sysroot     dosfstools-native:do_populate_sysroot     virtual/kernel:do_deploy     ${IMAGE_BOOTLOADER}:do_deploy     rpi-config:do_deploy     ${@bb.utils.contains('MACHINE_FEATURES', 'armstub', 'armstubs:do_deploy', '' ,d)}     ${@bb.utils.contains('RPI_USE_U_BOOT', '1', 'u-boot:do_deploy', '',d)}     ${@bb.utils.contains('RPI_USE_U_BOOT', '1', 'u-boot-default-script:do_deploy', '',d)}     ${RPI_SDIMG_EXTRA_DEPENDS} "
    oe.utils.features_backfill("MACHINE_FEATURES", d)
 ```

nucatus, 2022-02-16:
Going through the logs, I see that the rtc feature is present in the `MACHINE_FEATURES_BACKFILL` variable, as well as into the `MACHINE_FEATURES_BACKFILL_CONSIDERED` variable. According to the [documentation](https://www.yoctoproject.org/docs/3.1/ref-manual/ref-manual.html#ref-features-backfill), when the feature is present in both of these, the feature is actually _removed_ from the MACHINE_FEATURES. Please correct me if I'm wrong.

nucatus, 2022-02-16:
Here is the outcome of my research on the subject.

The rtc feature was added at a later time in the standard distribution and for back compatibility reasons the feature was not added to the `MACHINE_FEATURES` variable, but to the `MACHINE_FEATURES_BACKFILL` variable. The explanation behind it is in the [documentation](https://www.yoctoproject.org/docs/3.1/ref-manual/ref-manual.html#ref-features-backfill).

Now, for developers, to deactivate a feature that is added in the `MACHINE_FEATURES_BACKFILL` you have to add this feature to the `MACHINE_FEATURES_BACKFILL_CONSIDERED` variable. (If you're the person who chose the name of this variable, plase stay away!).

The raspberrypi machine removes the feature by [adding](https://github.com/agherzan/meta-raspberrypi/blob/2e3ed68f9d37a269cfdebee3bbca354fd635c8ea/conf/machine/include/rpi-base.inc#L96) "rtc" to the the `MACHINE_FEATURES_BACKFILL_CONSIDERED` variable.

However, if your board features a RTC device, you can still add the rtc feature by undoing the setting in the raspberrypi machine.conf. In your local.conf or other global conf file, add `MACHINE_FEATURES_BACKFILL_CONSIDERED_remove = "rtc"`.

For this to properly work, you also need to compile the kernel with the RTC driver included and other protocol drivers that are required by the device. In my case, where I used the pcf8563 device, I compiled the kernel with these settings:

```
CONFIG_I2C_CHARDEV=y
CONFIG_RTC_DRV_PCF8563=y

# Automatically set the system clock from hwclock
CONFIG_RTC_HCTOSYS=y
```

Of course, you need to enable the overlay device tree accordingly.

agherzan, 2022-02-19:
That is correct @nucatus. Can you remind me what takes into consideration this machine configuration? Wasn't there also an equivalent distro feature? I'm wondering if we should just drop this assumption in `meta-raspberrypi`. Sure, the boards don't have an RTC but there are a good set of RPI-based boards with onboard RTC modules and as you said, you can always attach one.

PS I'm not the one naming that variable :) and it's not ideal. But as you know, figuring out names is the hardest part of any implementation.

nucatus, 2022-02-21:
@agherzan, strictly speacking, I guess the decision to deactivate the rtc module in case of `meta-raspberrypi` is wise, taking into account the standard board. However, I don't have the knowledge of how reinstating the feature will affect the standard build on a board where the device is absent.

In case of a board featuring an rtc device, as I mentioned in the previous post, there's not much you have to do in order to get the feature working, besides reinstating the feature by removing it from `MACHINE_FEATURE_BACKFILL_CONSIDERED`. 

Regarding the naming, my intention was not to point at you, rather to an imaginary person who happens to be both, the person to name it, and somebody reading this post. It's a safty valve to disipate unneeded frustration :)

agherzan, 2022-02-23:
@nucatus So basically, everything that this `MACHINE_FEATURES` actually does is to pull in `VIRTUAL-RUNTIME_base-utils-hwclock` - which by default is from `busybox`. Obviously, this configuration can't know the kernel configuration, module, etc associated with it so it assumes that it is done separately. The annoyance is that if I start to enable this, everyone will start deploying a runtime service for `hwclock`. So I reckon the way forward now is to just add it in your own image. I don't see another easy way out of this. @kraj, what do you think?

kraj, 2022-02-23:
you can do something like 
```
KERNEL_FEATURES:append = " ${@bb.utils.contains('MACHINE_FEATURES', 'rtc', 'features/rtc/rtc.scc', '', d)}"
```

in a kernel bbappend. You might have to create rtc.scc if it does not exist.

agherzan, 2022-02-23:
@kraj The issue here fundamentally is that we disable a feature by default for all machines (rtc) even though there could be rtc device attached to basically any of these boards.

kraj, 2022-02-23:
> @kraj The issue here fundamentally is that we disable a feature by default for all machines (rtc) even though there could be rtc device attached to basically any of these boards.

That deserves to be a new machine IMO unless we want to make it runtime dynamic, which could be quite complex.

agherzan, 2022-02-23:
@kraj That is hard to justify as a new configuration only for a peripheral attached. It would easily blow in configurations. I would say that for now a custom image/layer with the right kernel configuration + the right `hwclock` provider should be the way.

kraj, 2022-02-23:
> @kraj That is hard to justify as a new configuration only for a peripheral attached. It would easily blow in configurations. I would say that for now a custom image/layer with the right kernel configuration + the right `hwclock` provider should be the way.

`rtc` is a MACHINE_FEATURE, so if you tend to ignore that then you will end up with sub-optimal solution.

agherzan, 2022-02-24:
@kraj there are issues with that. You saw in the past how creating machines based on existing ones is showing side effects. Practically you are proposing for downstream to create a duplicated machine only for a machine feature?

kraj, 2022-02-24:
if we have a smarter way to make the service ineffective during runtime perhaps thats fine but it will be a complex solution.

agherzan, 2022-02-24:
Ok, @nucatus. I think I will take back my initial idea of removing the RTC machine feature "considered". This is because it would either have an impact on the majority of the layer's users or need way more work to make this runtime-based implementation (with minimum runtime impact). And currently, there are at least a couple of avenues that can be pursued. Having a custom machine would be the cleanest - we have some machines that are based on existing ones so you could take inspiration from that. Another option would be to go down the image/configuration route (I mentioned this before).

Given that there are workarounds/solutions I'd say we can for now close this issue. What do you think?

nucatus, 2022-03-04:
@agherzan, yes, this totally makes sense. The only recommendation I would add is to somehow improve the documentaiton to make this obvious for people going into the matter. 

Yes, the issue can be closed. Thanks.

agherzan, 2022-03-14:
@nucatus Makes sense. Check this one out for a short review: https://github.com/agherzan/meta-raspberrypi/pull/1035