agherzan/meta-raspberrypi #1013: Different boot behaviour between rpi-sdimg and wic images

JeanPhilippeLebel, 2022-02-17:
Raspberry Pi 4 (32-bit build) using the Honister branch. Building/testing the core-image-weston image.

A 7 inch/touchscreen display (official pi display) is connected via the DSI connector and I use the HDMI port next to the USB-C connector.

My objective is to see the desktop on the touchscreen.

If I boot from the wic image,  I see the rainbow on both the touchscreen and HDMI then the touchscreen shuts down and the desktop is displayed on the HDMI output. I never see the "progress"  bar.

When I boot from the rpi-sdimage, both screens display the rainbow image, I then see the progress bar on the touchscreen (while the HDMI output is still displaying the rainbow). On the touchscreen, the progressbar disappears and displays the boot log (rainbow on the hdmi).

After a rapid comparison of the two images, on the wic image the overlays folder is empty. 

To sum up, I think there are two issues
1 - Behaviour is different between the rpi-sdimage and the wic image.
2 - When using the rpi-sdimage, the desktop is never shown (not on HDMI, not on the touchscreen)

config.conf only defines the following

MACHINE = 'raspberrypi4'
IMAGE_FSTYPES += " rpi-sdimg wic"

bblayers.conf contains meta, meta-poky, meta-yocto-bsp and meta-raspberrypi.

agherzan, 2022-02-19:
@JeanPhilippeLebel That sounds like a bug. When you say the overlays folder, do you refer to the overlay files on the boot partition? Can you confirm that none are deployed there?

JeanPhilippeLebel, 2022-02-21:
Hello,

The two images were not built exactly at the same time so to be sure I have a coherent setup, I deleted the tmp, cache and sstate-cache folders and rebuilt everything.

The overload folder on the boot partition of the wic image is no longer empty BUT they are not identical (the wic image as an extra "overlay_map.dtb”).

I've placed the two generated images here:

https://www.dropbox.com/s/lkeuytzl0jgxg3g/core-image-weston-raspberrypi4-20220217180533.rootfs.rpi-sdimg?dl=0
https://www.dropbox.com/s/5gt4ns3nrx8c2h9/core-image-weston-raspberrypi4-20220217180533.rootfs.wic?dl=0

What I'm chasing is not quite the overlay folder but the difference in the boot behaviour between the two image type, especially regarding having the GUI displayed on the touchscreen which works using the sdimg and not the wic image. Any ideas?

JeanPhilippeLebel, 2022-02-21:
Actually, it's the other way around. The sdimg is missing the "overlay_map.dtb” and boots fine to the touchscreen.

I will run two tests:
- Removing the overlay_map.dtb file from the _wic_ image to see if there is any change.
- Adding the overlay_map.dtb file from the _sdimg_ image to see if there is any change.

JeanPhilippeLebel, 2022-02-21:
Removing the overlay_map.dtb file from the _wic_ image RESOLVES the issue.

I was not able to have Yocto force-adding the overlay_map.dtb to the sdimage (any idea of that can be done?)

agherzan, 2022-02-21:
@JeanPhilippeLebel that sounds like the vc4 configuration fails. And basically not using fixing it. Are you using the default vc4 configuration? The map dtb is mapping the right vc4 dtb for rpi4. Sounds like something is misbehaving there.

JeanPhilippeLebel, 2022-02-21:
@agherzan Yes, default everything. Even if there was a particular configuration, why would the sdimg behaves differently than the wic image?

Is there more info I can provide you with? 

Thanks.

agherzan, 2022-02-21:
So what I think is happening is that there is an issue either in the dtb mapping or when using vc4. The sdimg works because it doesn't have that mapping hence not using vc4. Would you be able to give `master` a spin? It could be that a later kernel update has this fixed.

JeanPhilippeLebel, 2022-02-22:
Tested with meta-raspberrypi master (157f72b8084756d9ca994e9ca30a07cfc4074137) and yocto master (6c23bd5c8113c1a8feddf2171626cb02e421b71d)

The behaviour is the same.

agherzan, 2022-02-22:
I'd probably need to reproduce to dig a bit into this. It sounds like a VC4 issue. I'm not sure when I'd be about to though but I'll keep you updated if I figure something out.

JeanPhilippeLebel, 2022-02-23:
Thanks, let me know if I can be of any help

agherzan, 2022-02-23:
@JeanPhilippeLebel So the first issue is that `overlay_map.dtb` is deployed in the wrong place for the `sdimg`. It is supposed to be in `overlays` and it is actually deployed in the root. I know that this is making things worse but that is the issue of the different content. Now for the display, we would need to look a bit more into it.

JeanPhilippeLebel, 2022-02-23:
Shall I open a different issue regarding the weston/vc4 issue?

agherzan, 2022-02-23:
I have a feeling that the vc4/graphics issue is the same as https://github.com/agherzan/meta-raspberrypi/issues/1022

closed 2022-02-25 by agherzan