agherzan/meta-raspberrypi #1015: Build images for RPi CM4 on dunfell branch

Sourabh1107, 2022-02-19:
**Description**

I have few doubts about booting RPi CM4 with raspberrypi4-64.conf machine configuration. I am on "Dunfell" branch and there is difference between machine config between dunfell and master branch. I am wondering whether can we use `dunfell `branch to build and boot images for RPi CM4? 

`linux-firmware-rpidistro-bcm43456 \ bluez-firmware-rpidistro-bcm4345c5-hcd \` in `MACHINE_EXTRA_RRECOMMENDS` are not included in the `raspberrypi4-64.conf` dunfell branch. And also many of the `RPI_KERNEL_DEVICETREE_OVERLAYS` and `bcm2711-rpi-cm4.dtb \` are not available in `meta-raspberrypi/conf/machine/include/rpi-base.inc`. Will these be an issue?

I modified `raspberrypi4-64.conf` file and matched with `raspberrypi4-64.conf` of master branch. But while bitbaing it gave couple of errors like:

`ERROR: Nothing RPROVIDES 'linux-firmware-rpidistro-bcm43456' (but /root/sources/iot-demonstrator/poky/meta/recipes-` `core/packagegroups/packagegroup-base.bb RDEPENDS on or otherwise requires it)`
`NOTE: Runtime target 'linux-firmware-rpidistro-bcm43456' is unbuildable, removing...`
`Missing or unbuildable dependency chain was: ['linux-firmware-rpidistro-bcm43456']`

And also:

`ERROR: ParseError at /root/sources/iot-demonstrator/meta-raspberrypi/conf/machine/raspberrypi4-64.conf:17: Could not include required file conf/machine/include/arm/armv8a/tune-cortexa72.inc`

Can anyone please let me know how to work with `dunfell` branch for RPi CM 4?

Your help will be much appreciated.

**Steps to reproduce the issue:**
1. Modify `raspberrypi4-64.conf` and match with `raspberrypi4-64.conf` of master branch.

**Describe the results you received:**
ParseError and unbuildable targets.

**Describe the results you expected:**
Successfully bitbake the image recipe and booting the board from the image.

**Additional information you deem important (e.g. issue happens only occasionally):**

**Additional details (revisions used, host distro, etc.):**
My other meta-layers are on dunfell branch and switching to new branch is not an option at the moment.

agherzan, 2022-02-19:
This machine is not supported on dunfell and I fear that backporting that support won't be trivial as it would depend on kernel/firmware updates which on an LTS branch would be undesirable. My recommendation is to create an additional layer where you can bump the relevant dependencies and bring in a copy of the cm4 machine configuration.

closed 2022-02-19 by agherzan

Sourabh1107, 2022-02-19:
Hello @agherzan thanks for the update. Can you please let me know what do you mean by "bump the relevant dependencies and bring in a copy of the cm4 machine configuration"? For example, how and where to get the files relevant for cm4? like related to `linux-firmware-rpidistro-bcm43456`? Or else which branch can I use other than master and provides LTS?

agherzan, 2022-02-19:
> Hello @agherzan thanks for the update. Can you please let me know what do you mean by "bump the relevant dependencies and bring in a copy of the cm4 machine configuration"? For example, how and [...]

That is sadly something you need to figure out by checking the relevant support in all the components that might have impact (kernel/dtbs, Raspberry Pi firmware, machine configuration, WiFi/BT firmware etc.). It shouldn't be much but I wouldn't know without digging into `master` support and the components listed above.