agherzan/meta-raspberrypi #1022: Boot issue on RPi CM4 | Black screen after boot on Honister release

Sourabh1107, 2022-02-23:
**Description**
Hello All,

I am building a basic image (core-image-base) for RPi CM4(Lite) on "Honister" release with "uboot" as bootloader.

In my `conf/local.conf` file `MACHINE = "raspberrypi4-64"` and also I have set

```
ENABLE_UART = "1"
RPI_USE_U_BOOT = "1"
PREFERRED_PROVIDER_virtual/bootloader = "u-boot"
```

Image builds successfully and I transferred the image to SD card and tried to boot. I am using WKS_FILE = "sdimage-dual-raspberrypi.wks.in"

sdimage-dual-raspberrypi.wks.in :

```
part /boot --source bootimg-partition --ondisk mmcblk0 --fstype=vfat --label boot --active --align 4096 --size 100
part / --source rootfs --ondisk mmcblk0 --fstype=ext4 --label rootfs_A --align 4096
part / --source rootfs --ondisk mmcblk0 --fstype=ext4 --label rootfs_B --align 4096
part /home --source rootfs --rootfs-dir=${IMAGE_ROOTFS}/home --ondisk mmcblk0 --fstype=ext4 --label homefs --align 1024 --size 500 --fsoptions "x-systemd.growfs"
```

But unfortunately, i am not able to boot using this image and initially it starts with 
```
.
.
Detecting USB Device.
.
.
Hit any key to stop auto-boot..
.
.
Starting kernel..

```
But right after this screen turns off and after a second screen turns on with "Loading progress bar" and as soon as it loads, a blank screen pops-up. I am not sure whether system is booted properly or not. I am not able to interact with it. And no Linux user space is accessible

Sadly, the custom carrier board I am using has no serial console, so I can only use a RPi display to see the progress and messages.

Can anyone please let me know what could be the issue here and how to resolve it?

Your help will be much appreciated.

thanks in advance..

**Steps to reproduce the issue:**
1. Use "raspberrypi4-64" as machine with "u-boot" bootloader, build "core-image-base" for RPi CM4.

**Describe the results you received:**
Black screen after boot

**Describe the results you expected:**
System bring-up 

**Additional information you deem important (e.g. issue happens only occasionally):**

**Additional details (revisions used, host distro, etc.):**
I am using Yocto "Honister". Tried with "Dunfell" but it doesn't support RPi CM4.

agherzan, 2022-02-23:
@Sourabh1107 Could you try the reproduce without uboot? And also, can you try using the included sdimg image type?

Sourabh1107, 2022-02-23:
Hello @agherzan Thanks for quick reponse.

I already tried without uboot. I removed `RPI_USE_U_BOOT = "1" \ PREFERRED_PROVIDER_virtual/bootloader = "u-boot"` from `local.conf`. But I got similar result. Blank screen with no messages.

Do you mean to use different `.wks` file? Like [sdimage-raspberrypi.wks](https://github.com/agherzan/meta-raspberrypi/tree/master/wic) ? Or booting with `sdimg` as `IMAGE_FSTYPES`?

agherzan, 2022-02-23:
I mean to just try a minimum setup with `sdimg` in `IMAGE_FSTYPES`.

Sourabh1107, 2022-02-23:
Hi @agherzan after building core-image-base with `IMAGE_FSTYPES += " rpi-sdimg"` and coping the image `sudo dd if=core-image-base-raspberrypi4-64-20220223110404.rootfs.rpi-sdimg | sudo dd of=/dev/sdb bs=1M conv=fsync status=progress` to SD card and booting from it, I have same behavior as before. Black screen with no messages. Below is the local.conf file I am using:

```
CONF_VERSION = "2"

ENABLE_I2C = "1"
ENABLE_UART = "1"

DISTRO_FEATURES:append = " rauc pam systemd wifi"
DISTRO_FEATURES_BACKFILL_CONSIDERED = "sysvinit"
VIRTUAL-RUNTIME_init_manager = "systemd"
VIRTUAL-RUNTIME_initscripts = ""

#RPI_USE_U_BOOT = "1"
#PREFERRED_PROVIDER_virtual/bootloader = "u-boot"

#WKS_FILE = "sdimage-dual-raspberrypi.wks.in"
WKS_FILE = "sdimage-raspberrypi.wks"

WKS_FILE_DEPENDS += "squashfs-tools-native"

IMAGE_FEATURES += " splash "

IMAGE_INSTALL:append = " rauc wpa-supplicant iw dhcpcd rauc-hawkbit-updater"

CORE_IMAGE_EXTRA_INSTALL += " packagegroup-base-wifi"

IMAGE_FSTYPES += "tar.bz2 ext4 wic.bz2 wic.bmap rpi-sdimg"
SDIMG_ROOTFS_TYPE = "ext4"

EXTRA_IMAGE_FEATURES += " \
    tools-debug \
    ssh-server-dropbear \
    dbg-pkgs \
    "
```

agherzan, 2022-02-23:
@Sourabh1107 It's very hard to debug if you have no serial access to the device. I'd try to get a carrier with a serial console. Otherwise, I'm not sure how to deal with this. Well, another option would be to go down the SSH route. Do all the configurations to allow SSH connection and DHCP, check your router for IP and try to connect to the target.

Sourabh1107, 2022-02-23:
Hello @agherzan Can you please let me know how to enable verbose options for kernel and bootloader? This may help to see some messages printed during booting..

agherzan, 2022-02-24:
There are kernel arguments you can play with. On the firmware side I'm not aware of flexibility in that regard. https://elinux.org/Kernel_Debugging_Tips

Sourabh1107, 2022-02-27:
Hello @agherzan I could get my hands on Official Raspberry Pi Compute Module 4 IO board and below is the log messages I got while booting.
```
[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x410fd083]
[    0.000000] Linux version 5.10.78-v8 (oe-user@oe-host) (aarch64-oe-linux-gcc (GCC) 11.2.0, GNU ld (GNU Binutils) 2.37.20210721) #1 SMP PREEMPT Tue Feb 22 15:19:24 UTC 2022
[    0.000000] random: fast init done
[    0.000000] Machine model: Raspberry Pi Compute Module 4 Rev 1.0
[    0.000000] efi: UEFI not found.
[    0.000000] Reserved memory: created CMA memory pool at 0x000000001ec00000, size 256 MiB
[    0.000000] OF: reserved mem: initialized node linux,cma, compatible id shared-dma-pool
[    0.000000] Zone ranges:
[    0.000000]   DMA      [mem 0x0000000000000000-0x000000003b3fffff]
[    0.000000]   DMA32    empty
[    0.000000]   Normal   empty
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x000000003b3fffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x000000003b3fffff]
[    0.000000] percpu: Embedded 32 pages/cpu s92312 r8192 d30568 u131072
[    0.000000] Detected PIPT I-cache on CPU0
[    0.000000] CPU features: detected: Spectre-v2
[    0.000000] CPU features: detected: Spectre-v4
[    0.000000] CPU features: detected: ARM errata 1165522, 1319367, or 1530923
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 238896
[    0.000000] Kernel command line: coherent_pool=1M 8250.nr_uarts=1 snd_bcm2835.enable_compat_alsa=0 snd_bcm2835.enable_hdmi=1  smsc95xx.macaddr=E4:5F:01:49:E3:A0 vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  dwc_otg.lpm_enable=0 console=ttyS0,115200 root=/dev/mmcblk0p2 rootfstype=ext4 rootwait
[    0.000000] Dentry cache hash table entries: 131072 (order: 8, 1048576 bytes, linear)
[    0.000000] Inode-cache hash table entries: 65536 (order: 7, 524288 bytes, linear)
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Memory: 663716K/970752K available (12416K kernel code, 1992K rwdata, 4440K rodata, 3968K init, 1267K bss, 44892K reserved, 262144K cma-reserved)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] ftrace: allocating 39462 entries in 155 pages
[    0.000000] ftrace: allocated 155 pages with 5 groups
[    0.000000] rcu: Preemptible hierarchical RCU implementation.
[    0.000000] rcu:     RCU event tracing is enabled.
[    0.000000] rcu:     RCU restricting CPUs from NR_CPUS=256 to nr_cpu_ids=4.
[    0.000000]  Trampoline variant of Tasks RCU enabled.
[    0.000000]  Rude variant of Tasks RCU enabled.
[    0.000000]  Tracing variant of Tasks RCU enabled.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 25 jiffies.
[    0.000000] rcu: Adjusting geometry for rcu_fanout_leaf=16, nr_cpu_ids=4
[    0.000000] NR_IRQS: 64, nr_irqs: 64, preallocated irqs: 0
[    0.000000] GIC: Using split EOI/Deactivate mode
[    0.000000] irq_brcmstb_l2: registered L2 intc (/soc/interrupt-controller@7ef00100, parent irq: 10)
[    0.000000] random: get_random_bytes called from start_kernel+0x3b4/0x574 with crng_init=1
[    0.000000] arch_timer: cp15 timer(s) running at 54.00MHz (phys).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0xc743ce346, max_idle_ns: 440795203123 ns
[    0.000006] sched_clock: 56 bits at 54MHz, resolution 18ns, wraps every 4398046511102ns
[    0.000247] Console: colour dummy device 80x25
[    0.000305] Calibrating delay loop (skipped), value calculated using timer frequency.. 108.00 BogoMIPS (lpj=216000)
[    0.000335] pid_max: default: 32768 minimum: 301
[    0.000481] LSM: Security Framework initializing
[    0.000658] Mount-cache hash table entries: 2048 (order: 2, 16384 bytes, linear)
[    0.000688] Mountpoint-cache hash table entries: 2048 (order: 2, 16384 bytes, linear)
[    0.001947] cgroup: Disabling memory control group subsystem
[    0.004208] rcu: Hierarchical SRCU implementation.
[    0.005195] EFI services will not be available.
[    0.005693] smp: Bringing up secondary CPUs ...
[    0.006664] Detected PIPT I-cache on CPU1
[    0.006733] CPU1: Booted secondary processor 0x0000000001 [0x410fd083]
[    0.007826] Detected PIPT I-cache on CPU2
[    0.007873] CPU2: Booted secondary processor 0x0000000002 [0x410fd083]
[    0.009043] Detected PIPT I-cache on CPU3
[    0.009088] CPU3: Booted secondary processor 0x0000000003 [0x410fd083]
[    0.009268] smp: Brought up 1 node, 4 CPUs
[    0.009312] SMP: Total of 4 processors activated.
[    0.009330] CPU features: detected: 32-bit EL0 Support
[    0.009346] CPU features: detected: CRC32 instructions
[    0.009363] CPU features: detected: 32-bit EL1 Support
[    0.044240] CPU: All CPU(s) started at EL2
[    0.044310] alternatives: patching kernel code
[    0.045475] devtmpfs: initialized
[... 440 lines cut ...]
[  414.359562] w1_master_driver w1_bus_master1: Attaching one wire slave 00.100000000000 crc 9d
[  414.371851] w1_master_driver w1_bus_master1: Family 0 for 00.100000000000.9d is not registered.
[  450.119561] w1_master_driver w1_bus_master1: Attaching one wire slave 00.900000000000 crc 11
[  450.131889] w1_master_driver w1_bus_master1: Family 0 for 00.900000000000.11 is not registered.
[  519.839564] w1_master_driver w1_bus_master1: Attaching one wire slave 00.500000000000 crc db
[... cut ...]
tmpfs                /var/volatile        tmpfs      defaults              0  0

/dev/mmcblk0p1       /boot                auto       defaults              0  0
/dev/appfs           /app                 auto       defaults              0  0
/dev/configfs        /config              auto       defaults              0  0
/dev/mmcblk0p9       /data                auto       defaults              0  0
```

Can you please let me know whata could be the issue here?

Your help will be much appreciated.

Sourabh1107, 2022-03-07:
Hi @agherzan , can you please let me know what could be the reason for the above issue?

agherzan, 2022-03-14:
@Sourabh1107 can you try to build without uboot (using the firmware directly loading the kernel image) and check how it goes?

Sourabh1107, 2022-03-14:
Hello @agherzan I am not using uboot now but still console is stuck at:

```
OpenEmbedded nodistro.0 raspberrypi4-64 ttyS0

raspberrypi4-64 login: [   14.718847] Bluetooth: HCI UART driver ver 2.3
[   14.723478] Bluetooth: HCI UART protocol H4 registered
[   14.728877] Bluetooth: HCI UART protocol Three-wire (H5) registered
[   14.735632] Bluetooth: HCI UART protocol Broadcom registered
[   14.985386] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[   14.990829] Bluetooth: BNEP filters: protocol multicast
[   14.996781] Bluetooth: BNEP socket layer initialized
[   15.016364] NET: Registered protocol family 38
[   15.024632] audit: type=1334 audit(1646676238.549:6): prog-id=9 op=LOAD
[   15.031791] audit: type=1334 audit(1646676238.549:7): prog-id=10 op=LOAD
[   15.103792] Bluetooth: RFCOMM TTY layer initialized
[   15.108976] Bluetooth: RFCOMM socket layer initialized
[   15.114255] Bluetooth: RFCOMM ver 1.11
```

I have commented below lines in conf/local.conf file:

```
#RPI_USE_U_BOOT = "1"
#PREFERRED_PROVIDER_virtual/bootloader = "u-boot"
```

Only using `VIRTUAL-RUNTIME_init_manager = "systemd"` as init system.

agherzan, 2022-03-14:
I don't see it stuck as I can read there is a login prompt.

Sourabh1107, 2022-03-14:
Yes, even I think so @agherzan but may be because of "Bluetooth" related messages, I am not able to login. Is there a way to disable this?

anilrath51, 2022-08-04:
Hi @Sourabh1107,

Could you please try the below steps?

Added below lines in local.conf file.
--------------------------------------

MACHINE ??= "raspberrypi"
ENABLE_UART = "1"
IMAGE_FSTYPES += "tar.bz2 wic wic.gz wic.bmap "
WKS_FILE="sdimage-raspberrypi.wks"

$ bitbake parted-native dosfstools-native mtools-native bmap-tools-native
$ bitbake core-image-minimal
$ wic create sdimage-raspberrypi -e core-image-minimal

Boot with SD Card on Raspberry Pi Model BNOTE: /dev/sdg is <sd-card> mount point.
-----------------------------------------------------------------------------------------------

/poky/build/tmp/deploy/images/raspberrypi$ /usr/bin/bmaptool copy --bmap core-image-minimal-raspberrypi.wic.bmap core-image-minimal-raspberrypi.wic /dev/sdg
/poky/build/tmp/deploy/images/raspberrypi$ sync