agherzan/meta-raspberrypi #1031: u-boot issue with Rev 1.5 boards

nmbath, 2022-03-08:
I am working on a project of updating an embedded system to work with the RPI boards based on OpenEmbedded dunfell. The existing u-boot was 2019.07 and kernel 4.19, so out of date.

I have moved to the 2021.01 version of u-boot, kernel 5.10 and boot files 20210527 as per the RaspberryPI meta layer for dunfell, however when used on a RPI4 rev 1.5 board I am getting error from u-boot:

```
U-Boot 2021.01-00003-g7389bb8ec2-dirty (Mar 01 2022 - 09:12:40 +0000)

DRAM:  1.9 GiB
RPI 4 Model B (0xb03115)
MMC:   mmcnr@7e300000: 1, mmc@7e340000: 0
Loading Environment from MMC... *** Warning - bad CRC, using default environment

In:    serial
Out:   serial
Err:   serial
Net:   eth0: ethernet@7d580000
PCIe BRCM: link up, 5.0 Gbps x1 (SSC)
starting USB...
Bus xhci_pci: Register 5000420 NbrPorts 5
Starting the controller
USB XHCI 1.00
scanning bus xhci_pci for devices... 2 USB Device(s) found
       scanning usb for storage devices... 0 Storage Device(s) found
Hit any key to stop autoboot:  0
** No partition table - mmc 0 **
U-Boot>
``` 

I have tested this with RPI rev 1.2, 1.4 and 1.5 boards. The 1.2 and 1.4 boards boot without any issues.

There is little information online about the revision 1.5 boards. I did come across this [link](https://forums.raspberrypi.com/viewtopic.php?p=1971022#p1971022)

As dunfell pulls in boot files of 20210527, I should be using boot files that will work with the rev 1.5 board.

Anyone got any pointers as to what might be causing the issue and what the fix is?

agherzan, 2022-03-14:
Could you check exactly what autoboot command fails? Also, list the mmc devices - maybe the mmc index is wrong?

nmbath, 2022-03-16:
The output for mmc list and mmc info is

```
U-Boot 2021.01-00006-g46663de97b-dirty (Mar 07 2022 - 20:52:53 +0000)

DRAM:  1.9 GiB
RPI 4 Model B (0xb03115)
MMC:   mmcnr@7e300000: 1, mmc@7e340000: 0
Loading Environment from MMC... *** Warning - bad CRC, using default environment

In:    serial
Out:   serial
Err:   serial
Net:   eth0: ethernet@7d580000
Hit any key to stop autoboot:  0
U-Boot> mmc list
mmcnr@7e300000: 1
mmc@7e340000: 0 (SD)
U-Boot> mmc info
Device: mmc@7e340000
Manufacturer ID: 2
OEM: 544d
Name: SA16G
Bus Speed: 25000000
Mode: MMC legacy
Rd Block Len: 512
SD version 1.0
High Capacity: Yes
Capacity: 14.5 GiB
Bus Width: 4-bit
Erase Group Size: 512 Bytes
```

The autoboot is
```
bootcmd=run set_root set_args load do_boot
set_root=if test "${version}" = 2; then setenv bootpart 0:3; setenv mmcroot /dev/mmcblk0p3; else setenv bootpart 0:2; setenv mmcroot /dev/mmcblk0p2; fi
set_args=fdt addr ${fdt_addr} && fdt get value bootargs /chosen bootargs && setenv bootargs "${bootargs} root=${mmcroot} rootwait"
```

For info on a Revision 1.4 RPI4 8Gb, which boots is

```
U-Boot 2021.01-00006-g46663de97b-dirty (Mar 07 2022 - 20:52:53 +0000)

DRAM:  7.9 GiB
RPI 4 Model B (0xd03114)
MMC:   mmcnr@7e300000: 1, mmc@7e340000: 0
Loading Environment from MMC... OK
In:    serial
Out:   serial
Err:   serial
Net:   eth0: ethernet@7d580000
Hit any key to stop autoboot:  0 
U-Boot> mmc list
mmcnr@7e300000: 1
mmc@7e340000: 0 (SD)
U-Boot> mmc info
Device: mmc@7e340000
Manufacturer ID: 41
OEM: 3432
Name: SD16G 
Bus Speed: 50000000
Mode: SD High Speed (50MHz)
Rd Block Len: 512
SD version 3.0
High Capacity: Yes
Capacity: 14.6 GiB
Bus Width: 4-bit
Erase Group Size: 512 Bytes
U-Boot> mmc dev
switch to partitions #0, OK
mmc0 is current device
```

The only difference I can find is the Bus Speed and Mode of the mmc

agherzan, 2022-04-20:
It would be worth trying the current master.