agherzan/meta-raspberrypi #1037: linux-raspberrypi: Unable to find revision cad0c67b64bb9de6bc970c0a7afbccabec0fcf9d

matthiasklein, 2022-03-17:
The Commit ID in the kernel recipe seems to be wrong: https://github.com/agherzan/meta-raspberrypi/commit/030bbc50b799a729e56b0cae20044d22773832a4

I get the following error: 
```
WARNING: linux-raspberrypi-1_5.15.27+gitAUTOINC+e1b976ee4f_cad0c67b64-r0 do_fetch: Failed to fetch URL git://github.com/raspberrypi/linux.git;name=machine;branch=rpi-5.15.y;protocol=https, attempting MIRRORS if available
ERROR: linux-raspberrypi-1_5.15.27+gitAUTOINC+e1b976ee4f_cad0c67b64-r0 do_fetch: Fetcher failure: Unable to find revision cad0c67b64bb9de6bc970c0a7afbccabec0fcf9d in branch rpi-5.15.y even from upstream
ERROR: linux-raspberrypi-1_5.15.27+gitAUTOINC+e1b976ee4f_cad0c67b64-r0 do_fetch: Bitbake Fetcher Error: FetchError('Unable to fetch URL from any source.', 'git://github.com/raspberrypi/linux.git;name=machine;branch=rpi-5.15.y;protocol=https')

```

I think instead of https://github.com/raspberrypi/linux/commit/cad0c67b64bb9de6bc970c0a7afbccabec0fcf9d it shoud be https://github.com/raspberrypi/linux/commit/ce9ff9bd6c05198446d19e6c42451a41547bd348

I used:

```
LINUX_VERSION ?= "5.15.28"
SRCREV_machine = "ce9ff9bd6c05198446d19e6c42451a41547bd348"
```

kraj, 2022-03-17:
does [this](https://github.com/agherzan/meta-raspberrypi/commit/7a9b2751f0c0822af1e18536a33358002b99cb05) fix it.

agherzan, 2022-03-17:
It should have. Closing this issue.

@kraj if this happens again I'll revert to 5.10. what do you think?

closed 2022-03-17 by agherzan

kraj, 2022-03-17:
> It should have. Closing this issue.
> 
> @kraj if this happens again I'll revert to 5.10. what do you think?

yes, I think we will have to wait and watch the public branch rebasing mess. The branch is marked as
default branch on the repo and it gets rebased :( so sad.

matthiasklein, 2022-03-18:
Yes, the commit fixes the problem.
Thanks a lot!