agherzan/meta-raspberrypi #1040: Kas does not build RPI-CM3

ckamas, 2022-03-25:
**Description**

When building with the KAS tool, the CM3 has two errors:
1. Using the MASTER pull the kernel does not build:
ERROR: linux-raspberrypi-1_5.15.28+gitAUTOINC+e1b976ee4f_ab46d8de9f-r0 do_fetch: Fetcher failure: Unable to find revision ab46d8de9f2406eb8e4ee0bd0b1b093cef59a1de in branch rpi-5.15.y even from upstream

**Steps to reproduce the issue:**
1. modify the KAS file as follows:
diff --git a/kas-poky-rpi.yml b/kas-poky-rpi.yml
index 2ab4770..a859482 100644
--- a/kas-poky-rpi.yml
+++ b/kas-poky-rpi.yml
@@ -1,17 +1,18 @@
 header:
   version: 8
 
-machine: raspberrypi4
+machine: raspberrypi-cm3
 distro: poky
 target:
   - core-image-base
 
 repos:
   meta-raspberry:
+    path: meta-raspberrypi
 
   poky:
     url: https://git.yoctoproject.org/git/poky
-    path: layers/poky
+    path: meta-poky
     refspec: master
     layers:
       meta:
@@ -20,7 +21,7 @@ repos:
 
   meta-openembedded:
     url: http://git.openembedded.org/meta-openembedded
-    path: layers/meta-openembedded
+    path: meta-openembedded
     refspec: master
     layers:
       meta-oe:
@@ -28,11 +29,6 @@ repos:
       meta-networking:
       meta-perl:
 
-  meta-qt5:
-    url: https://github.com/meta-qt5/meta-qt5/
-    path: layers/meta-qt5
-    refspec: master
-
 bblayers_conf_header:
   standard: |
     POKY_BBLAYERS_CONF_VERSION = "2"
@@ -59,3 +55,4 @@ local_conf_header:
         ABORT,${DL_DIR},100M,1K \
         ABORT,${SSTATE_DIR},100M,1K \
         ABORT,/tmp,10M,1K"
+

-----------------------------------------
Then execute kas build kas-poky-rpi.yml

lots of output.... then:
WARNING: linux-raspberrypi-1_5.15.28+gitAUTOINC+e1b976ee4f_ab46d8de9f-r0 do_fetch: Failed to fetch URL git://github.com/raspberrypi/linux.git;name=machine;branch=rpi-5.15.y;protocol=https, attempting MIRRORS if available

ERROR: linux-raspberrypi-1_5.15.28+gitAUTOINC+e1b976ee4f_ab46d8de9f-r0 do_fetch: Fetcher failure: Unable to find revision ab46d8de9f2406eb8e4ee0bd0b1b093cef59a1de in branch rpi-5.15.y even from upstream

ERROR: linux-raspberrypi-1_5.15.28+gitAUTOINC+e1b976ee4f_ab46d8de9f-r0 do_fetch: Bitbake Fetcher Error: FetchError('Unable to fetch URL from any source.', 'git://github.com/raspberrypi/linux.git;name=machine;branch=rpi-5.15.y;protocol=https')

matthiasklein, 2022-03-25:
The issue should be fixed with the latest commit: https://github.com/agherzan/meta-raspberrypi/commit/bc10842957212450023adfcbad5a73c21b28fce7

agherzan, 2022-03-25:
Pull latest changes. It should be fixed. Closing.

closed 2022-03-25 by agherzan

ckamas, 2022-03-26:
Thank you!  It works now.Chuck

Closed #1040.