agherzan/meta-raspberrypi #1050: "Could not find DRM device" when trying to run Qt apps with EGLFS on Pi4

jclsn, 2022-04-13:
**Description**
I have been trying to fix this for days now without success. I wanto run my application using eglfs, but I am getting the message ``Could not find DRM device!``

My configuration is as follow

```
# Architecture of the host machine
SDKMACHINE ?= "x86_64"

# Extra packages

LICENSE_FLAGS_WHITELIST = "commercial"

IMAGE_INSTALL:remove = " \
  x11 \
  wayland \
"

IMAGE_INSTALL:append = " \
  coreutils \
  qtbase-examples \
  qtbase-plugins \
  qtbase-tools \
  qtdeclarative \
  qtdeclarative-plugins \
  qtdeclarative-qmlplugins \
  qtdeclarative-tools \
  qtimageformats-plugins \
  qtmultimedia \
  qtmultimedia-plugins \
  qtmultimedia-qmlplugins \
  qtquickcontrols2 \
  qtquicktimeline \
  qtscript \
  qtsvg \
  qtsvg-plugins \
  qtsystems \
  qtsystems-qmlplugins \
  qtsystems-tools \
  rsync \
"

PACKAGECONFIG:append:pn-qtbase = " \
  eglfs \
  gles2 \
  kms \
  "

# Image file system types to package
IMAGE_FSTYPES = "rpi-sdimg"

MACHINE ??= "raspberrypi4-64"
DISTRO ??= "poky"
BBMULTICONFIG ?= ""
```

**Describe the results you received:**
I cannot run my Qt application although Qt is configuring right according to [this guide](https://github.com/UvinduW/Cross-Compiling-Qt-for-Raspberry-Pi-4):

```
QPA backends:
  DirectFB ............................... no
  EGLFS .................................. yes
  EGLFS details:
    EGLFS OpenWFD ........................ no
    EGLFS i.Mx6 .......................... no
    EGLFS i.Mx6 Wayland .................. no
    EGLFS RCAR ........................... no
    EGLFS EGLDevice ...................... yes
    EGLFS GBM ............................ yes
    EGLFS VSP2 ........................... no
    EGLFS Mali ........................... no
    EGLFS Raspberry Pi ................... no
    EGLFS X11 ............................ yes
```
I had no problems on Raspbian. So I am thinking that there must be something wrong with my meta-raspberrypi setup. I would think that the ``qtbase-%.bbappend`` would take care of providing the right configuration for the Pi4. I have been playing around with a lot of values, but am always coming to the same result

kraj, 2022-04-13:
I have it working well with Yoe Distro on rpi4-64

jclsn, 2022-04-13:
@kraj I am using kas. Should the tool matter? In the docs also Kas is recommended

kraj, 2022-04-13:
> @kraj I am using kas. Should the tool matter? In the docs also Kas is recommended

it matters, if you see in [meta-yoe](https://github.com/YoeDistro/yoe-distro/tree/master/sources/meta-yoe/dynamic-layers)
there are bunch of changes to make qt5 work with eglfs.

jclsn, 2022-04-13:
Looks to me like only ``linuxfb`` was added if you compare it to the .bbappend from meta-raspberrypi

```
# This does not work when wayland and x11 both are enabled in DISTRO_FEATURES and x11 image
# is built but works ok for eglfs and wayland images
QT_CONFIG_FLAGS += "${@bb.utils.contains('DISTRO_FEATURES', 'wayland', '-qpa wayland', \
                                         bb.utils.contains('DISTRO_FEATURES', 'x11', '', \
                                                          '-qpa eglfs', d), d)}"

PACKAGECONFIG:append = " libinput eglfs examples linuxfb tslib"

PACKAGECONFIG_GL = "${@bb.utils.contains('DISTRO_FEATURES', 'x11 opengl', 'gl', \
                    bb.utils.contains('DISTRO_FEATURES',     'opengl', 'eglfs gles2', \
                                     '', d), d)}"
```

And shouldn't these changes be in meta-raspberry Pi then, if they fix it?

jclsn, 2022-04-15:
@kraj I tried using the qtbase_%.bbappend from meta-yoe, but it did not change anything unfortunately

jclsn, 2022-04-15:
Problem was that I had ``dtoverlay=vc4-kms-v3d`` and not ``dtoverlay=vc4-fkms-v3d`` set in the ``config.txt```...

closed 2022-04-15 by jclsn