agherzan/meta-raspberrypi #1054: Kernel modules not loaded during boot

bhargavthriler, 2022-04-24:
**Description**
Observed that some kernel modules are not being loaded in the latest kernel `5.15.34-v7`.

So I have built a `core-image-base` with the tip of the master (`0135a02`) and while trying access the camera using Picamera got some errors. The errors mainly complain about mmal drivers not present.
```
root@raspberrypi3:~# python3
Python 3.10.4 (main, Mar 23 2022, 20:25:24) [GCC 11.3.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> from picamera import PiCamera
>>> camera = PiCamera()
mmal: mmal_vc_shm_init: could not initialize vc shared memory service
mmal: mmal_vc_component_create: failed to initialise shm for 'vc.camera_info' (7:EIO)
mmal: mmal_component_create_core: could not create component 'vc.camera_info' (7)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/lib/python3.10/site-packages/picamera/camera.py", line 408, in __init__
    self._init_revision(options)
  File "/usr/lib/python3.10/site-packages/picamera/camera.py", line 480, in _init_revision
    with mo.MMALCameraInfo() as camera_info:
  File "/usr/lib/python3.10/site-packages/picamera/mmalobj.py", line 2425, in __init__
    super(MMALCameraInfo, self).__init__()
  File "/usr/lib/python3.10/site-packages/picamera/mmalobj.py", line 696, in __init__
    mmal_check(
  File "/usr/lib/python3.10/site-packages/picamera/exc.py", line 184, in mmal_check
    raise PiCameraMMALError(status, prefix)
picamera.exc.PiCameraMMALError: Failed to create MMAL component b'vc.camera_info': I/O error
>>> 
>>> 
root@raspberrypi3:~# 
```

After digging through my system found an older build (don't know why didn't delete it but thankfully it gave some insight into the issue), I tried booting that image and everything seems to working fine.

So I checked out to the commit which the older build was using (`63a3d8cb17c5d1affe8f2848f45fcc6a706f9412`), and the camera worked fine(though I had to make few changes, which are not significant for this issue). While analyzing the bootlogs found that the latest build (`0135a02`) doesn't load all the drivers.

Also I have observed that the kernel module are compressed in the 5.15.34 kernel, eg: `root@raspberrypi3:~# ls /lib/modules/5.15.34-v7/kernel/drivers/usb/gadget/libcomposite.ko.xz` and while trying load the modules using modprobe getting the following error:
```
root@raspberrypi3:~# ls /lib/modules/5.15.34-v7/kernel/drivers/usb/gadget/legacy/
g_acm_ms.ko.xz        g_cdc.ko.xz           g_hid.ko.xz           g_midi.ko.xz          g_printer.ko.xz       g_webcam.ko.xz        gadgetfs.ko.xz
g_audio.ko.xz         g_ether.ko.xz         g_mass_storage.ko.xz  g_multi.ko.xz         g_serial.ko.xz        g_zero.ko.xz
root@raspberrypi3:~# modprobe gadgetfs
modprobe: FATAL: Module gadgetfs not found in directory /lib/modules/5.15.34-v7
```
My question is what and where the changes have happened to the kernel between `63a3d8cb17c5d1affe8f2848f45fcc6a706f9412` (5.10) and `0135a02` (5.15) , so that I can look into and adapt the changes required ?

**Note**: All the commit which are mentioned above are of this repo.

**Logs**
I have attached the bootlogs of both the kernels.
lsmod logs
- 5.15.34
```
root@raspberrypi3:~# lsmod
Module                  Size  Used by
root@raspberrypi3:~#
```
- 5.10.81
```
root@raspberrypi3:~# lsmod
Module                  Size  Used by
rfcomm                 49152  2
cmac                   16384  3
algif_hash             16384  1
nfc                    86016  0
aes_arm_bs             24576  2
crypto_simd            16384  1 aes_arm_bs
cryptd                 24576  2 crypto_simd
algif_skcipher         16384  1
af_alg                 28672  6 algif_hash,algif_skcipher
bnep                   20480  2
hci_uart               40960  1
btbcm                  16384  1 hci_uart
bluetooth             421888  31 hci_uart,bnep,btbcm,rfcomm
ecdh_generic           16384  2 bluetooth
ecc                    36864  1 ecdh_generic
ipv6                  503808  26
brcmfmac              331776  0
brcmutil               24576  1 brcmfmac
sha256_generic         16384  0
bcm2835_v4l2           49152  0
cfg80211              782336  1 brcmfmac
bcm2835_codec          40960  0
bcm2835_isp            32768  0
v4l2_mem2mem           36864  1 bcm2835_codec
rfkill                 32768  4 bluetooth,nfc,cfg80211
bcm2835_mmal_vchiq     36864  3 bcm2835_isp,bcm2835_codec,bcm2835_v4l2
videobuf2_dma_contig    20480  2 bcm2835_isp,bcm2835_codec
videobuf2_vmalloc      16384  1 bcm2835_v4l2
videobuf2_memops       16384  2 videobuf2_dma_contig,videobuf2_vmalloc
videobuf2_v4l2         32768  4 bcm2835_isp,bcm2835_codec,bcm2835_v4l2,v4l2_mem2mem
videobuf2_common       61440  5 bcm2835_isp,bcm2835_codec,bcm2835_v4l2,v4l2_mem2mem,videobuf2_v4l2
raspberrypi_hwmon      16384  0
videodev              253952  6 bcm2835_isp,bcm2835_codec,videobuf2_common,bcm2835_v4l2,v4l2_mem2mem,videobuf2_v4l2
mc                     45056  6 bcm2835_isp,bcm2835_codec,videobuf2_common,videodev,v4l2_mem2mem,videobuf2_v4l2
vc_sm_cma              32768  2 bcm2835_isp,bcm2835_mmal_vchiq
uio_pdrv_genirq        16384  0
uio                    20480  1 uio_pdrv_genirq
fixed                  16384  0
root@raspberrypi3:~# 
```

**Additional details (revisions used, host distro, etc.):**

Some more details of the builds
- meta-openembedded commit hash :  `64156a6f8f3b4527c4732c91936aee0a167acb48`
- poky commit hash :  `5ec6839709f1f9411a8334306c563675246fff35`
- local.conf
```
MACHINE = "raspberrypi3"
VIDEO_CAMERA = "1"
ENABLE_UART = "1"
RPI_USE_U_BOOT = "1"
GPU_FREQ = "250"
GPU_MEM = "128"

IMAGE_INSTALL:append = "python3-picamera openssh libcamera libffi libgphoto2"
```
- bblayers.conf
```
BBLAYERS ?= " \
  /home/bulbasaur/HDD-2-2/RPI-camera-project-23042022/poky/meta \
  /home/bulbasaur/HDD-2-2/RPI-camera-project-23042022/poky/meta-poky \
  /home/bulbasaur/HDD-2-2/RPI-camera-project-23042022/poky/meta-yocto-bsp \
  /home/bulbasaur/HDD-2-2/RPI-camera-project-23042022/meta-raspberrypi \
  /home/bulbasaur/HDD-2-2/RPI-camera-project-23042022/meta-openembedded/meta-python \
  /home/bulbasaur/HDD-2-2/RPI-camera-project-23042022/meta-openembedded/meta-oe \
  /home/bulbasaur/HDD-2-2/RPI-camera-project-23042022/meta-openembedded/meta-multimedia \
  "
```
[bootlog-v5_15_34-kernel.txt](https://github.com/agherzan/meta-raspberrypi/files/8549587/bootlog-v5_15_34-kernel.txt)
[bootlog-v5_10_18-kernel.txt](https://github.com/agherzan/meta-raspberrypi/files/8549588/bootlog-v5_10_18-kernel.txt)

bhargavthriler, 2022-04-26:
Hey @MatthiasGugel thanks for the explaination, I dig through it.

bhargavthriler, 2022-04-26:
BTW this behaviour is only for 5.15 kernel and not present in 5.10. As we still have a 5.10 recipe here, so built it and everything was working fine. Forgot to update, timecrunch ;) 

So need to check the 5.15 recipe or probably the upstream changes.

Though updates might be delayed as I only get time in weekends nowadays :)

bhargavthriler, 2022-04-26:
@MatthiasGugel check this, I think this is the change which prevents loading of the kernel
https://github.com/raspberrypi/linux/pull/4991

agherzan, 2022-04-27:
@MatthiasGugel @bhargavthriler It looks like kmod can't load the kernel modules due to the disabled compression type. `xz` is not enabled by default in `kmod`. That being said, give this MR a spin https://github.com/agherzan/meta-raspberrypi/pull/1056

agherzan, 2022-04-27:
That is unexpected - the build system should run `depmod` at rootfs type. Did you build a clean image with the change above?

bhargavthriler, 2022-04-28:
@agherzan ,yup @MatthiasGugel is correct even observing the same issue need to reboot after `depmod --all` to load the modules.

I have also observed that `modules.dep` is populated after your changes but somehow those are not loaded.

kraj, 2022-04-28:
@bhargavthriler @MatthiasGugel please test https://lists.openembedded.org/g/openembedded-core/message/164932
delete your `${TMPDIR}` dir and build the rootfs+image again so that image post processing can re-run the depmod during image build.

bhargavthriler, 2022-04-29:
@kraj , I deleted the tmpdir as you suggested and built the image again but still it's not loading all the drivers though those are present in module.deps

only these are loaded in the first boot
```
root@raspberrypi3:~# lsmod
Module                  Size  Used by
rfcomm                 49152  2
cmac                   16384  3
algif_hash             16384  1
nfc                    86016  0
aes_arm_bs             24576  2
crypto_simd            16384  1 aes_arm_bs
cryptd                 24576  2 crypto_simd
algif_skcipher         16384  1
af_alg                 28672  6 algif_hash,algif_skcipher
bnep                   20480  2
hci_uart               40960  1
btbcm                  20480  1 hci_uart
bluetooth             421888  31 hci_uart,bnep,btbcm,rfcomm
ecdh_generic           16384  2 bluetooth
ecc                    40960  1 ecdh_generic
rfkill                 32768  3 bluetooth,nfc
ipv6                  520192  26
```
The following are loaded after executing `depmod -a`
```
root@raspberrypi3:~# depmod -a
root@raspberrypi3:~# reboot

Broadcast message from root@raspberrypi3 (ttyS0) (Fri Mar  9 12:40:58 2018):

The system is going down for reboot NOW!
INIT: Sending processes configured via /etc/inittab the TERM signal
Stopping OpenBSD Secure Shell server: sshdstopped /usr/sbin/sshd (pid 421)
.
 * Stopping Avahi mDNS/DNS-SD Daemon: avahi-daemon
   ...done.
Stopping advanced power management daemon: no /usr/sbin/apmd found; none killed
apmd.
----------
reboot
----------
raspber[   20.160226] Bluetooth: RFCOMM ver 1.11
rypi3 login: [   33.770185] cam-dummy-reg: disabling
root
root@raspberrypi3:~# lsmod
Module                  Size  Used by
rfcomm                 49152  2
cmac                   16384  3
algif_hash             16384  1
nfc                    86016  0
aes_arm_bs             24576  2
crypto_simd            16384  1 aes_arm_bs
cryptd                 24576  2 crypto_simd
algif_skcipher         16384  1
af_alg                 28672  6 algif_hash,algif_skcipher
bnep                   20480  2
hci_uart               40960  1
btbcm                  20480  1 hci_uart
bluetooth             421888  31 hci_uart,bnep,btbcm,rfcomm
ecdh_generic           16384  2 bluetooth
ecc                    40960  1 ecdh_generic
ipv6                  520192  26
brcmfmac              335872  0
brcmutil               24576  1 brcmfmac
bcm2835_codec          45056  0
bcm2835_isp            32768  0
sha256_generic         16384  0
v4l2_mem2mem           36864  1 bcm2835_codec
bcm2835_v4l2           45056  0
bcm2835_mmal_vchiq     32768  3 bcm2835_isp,bcm2835_codec,bcm2835_v4l2
cfg80211              770048  1 brcmfmac
videobuf2_dma_contig    20480  2 bcm2835_isp,bcm2835_codec
videobuf2_vmalloc      16384  1 bcm2835_v4l2
videobuf2_memops       16384  2 videobuf2_dma_contig,videobuf2_vmalloc
videobuf2_v4l2         32768  4 bcm2835_isp,bcm2835_codec,bcm2835_v4l2,v4l2_mem2mem
videobuf2_common       65536  8 bcm2835_isp,bcm2835_codec,videobuf2_dma_contig,videobuf2_vmalloc,videobuf2_memops,bcm2835_v4l2,v4l2_mem2mem,videobuf2_v4l2
videodev              245760  6 bcm2835_isp,bcm2835_codec,videobuf2_common,bcm2835_v4l2,v4l2_mem2mem,videobuf2_v4l2
rfkill                 32768  4 bluetooth,nfc,cfg80211
mc                     45056  6 bcm2835_isp,bcm2835_codec,videobuf2_common,videodev,v4l2_mem2mem,videobuf2_v4l2
vc_sm_cma              32768  2 bcm2835_isp,bcm2835_mmal_vchiq
raspberrypi_hwmon      16384  0
uio_pdrv_genirq        16384  0
uio                    20480  1 uio_pdrv_genirq
fixed                  16384  0
root@raspberrypi3:~# 
```

kraj, 2022-04-29:
interesting, I have not done `raspberrypi3` builds but on rpi4 it looks  ok. What do you use for a build host ? I wonder if thats the variable factor.

bhargavthriler, 2022-04-29:
I am using Ubuntu 20.04.
Build image: core-image-base

kraj, 2022-04-29:
on pi4 the order of modules is a bit different after `depmod -a` but content is same

```diff
--- a
+++ b
@@ -1,30 +1,28 @@
-root@raspberrypi4-64:~# lsmod
-Module                  Size  Used by
-v3d                    86016  1
-bcm2835_v4l2           45056  0
-bcm2835_isp            32768  0
 bcm2835_codec          53248  0
+bcm2835_isp            32768  0
+v3d                    86016  1
 brcmfmac              339968  0
-gpu_sched              49152  1 v3d
-v4l2_mem2mem           45056  1 bcm2835_codec
+bcm2835_v4l2           45056  0
 bcm2835_mmal_vchiq     40960  3 bcm2835_codec,bcm2835_v4l2,bcm2835_isp
-brcmutil               24576  1 brcmfmac
-raspberrypi_hwmon      16384  0
 videobuf2_vmalloc      20480  1 bcm2835_v4l2
+v4l2_mem2mem           45056  1 bcm2835_codec
 videobuf2_dma_contig    24576  2 bcm2835_codec,bcm2835_isp
 videobuf2_memops       20480  2 videobuf2_vmalloc,videobuf2_dma_contig
 videobuf2_v4l2         32768  4 bcm2835_codec,bcm2835_v4l2,v4l2_mem2mem,bcm2835_isp
 videobuf2_common       69632  8 bcm2835_codec,videobuf2_vmalloc,videobuf2_dma_contig,videobuf2_v4l2,bcm2835_v4l2,v4l2_mem2mem,videobuf2_memops,bcm2835_isp
-i2c_brcmstb            20480  0
+gpu_sched              49152  1 v3d
+brcmutil               24576  1 brcmfmac
 videodev              266240  6 bcm2835_codec,videobuf2_v4l2,bcm2835_v4l2,videobuf2_common,v4l2_mem2mem,bcm2835_isp
+raspberrypi_hwmon      16384  0
+i2c_brcmstb            20480  0
 vc_sm_cma              36864  2 bcm2835_mmal_vchiq,bcm2835_isp
 mc                     61440  6 videodev,bcm2835_codec,videobuf2_v4l2,videobuf2_common,v4l2_mem2mem,bcm2835_isp
-cfg80211              876544  1 brcmfmac
 rpivid_mem             16384  0
-rfkill                 36864  2 cfg80211
-nvmem_rmem             16384  0
+cfg80211              876544  1 brcmfmac
 uio_pdrv_genirq        16384  0
 uio                    24576  1 uio_pdrv_genirq
+nvmem_rmem             16384  0
+rfkill                 36864  2 cfg80211
 sch_fq_codel           20480  7
 fuse                  139264  1
 ipv6                  565248  32

```

kraj, 2022-04-29:
> I am using Ubuntu 20.04. Build image: core-image-base

OK I dont ubuntu instead use archlinux but are you wiping the  SD card clean ? or is there some persistence there.

MatthiasGugel, 2022-04-29:
For me the kernel modules are now loaded without the need to call `depmod --all` manually. My host is Ubuntu 20.04 but I am using kas to build the image inside a docker container.

configuration.yml:
```
header:
  version: 11

machine: raspberrypi4-64
distro: my-distro
target: my-image
build_system: oe

repos:
  openembedded-core:
    url: "https://git.openembedded.org/openembedded-core"
    refspec: "master"
    path: "layers/openembedded-core"
    layers:
      meta:
...
  meta-raspberrypi:
    url: "https://github.com/agherzan/meta-raspberrypi"
    refspec: "master"
    path: "layers/meta-raspberrypi"
...
```

I manually patched `kas/layers/openembedded-core/meta/recipes-kernel/kmod/kmod_29.bb` and deleted what I believe is `${TMPDIR}` bevor building the image:

```
$ rm -rf kas/build/tmp-glibc
$ kas/kas-container build ./configuration.yml
...
WARNING  - Repo openembedded-core is dirty - no checkout
...
0: my-image-1.0-r0 do_rootfs - 5s (pid 375424)   3% |#######

```

Flash the SD Card:
```
$ sudo umount /dev/mmcblk0p*; sudo bmaptool copy my-image-raspberrypi4-64.wic.bz2 /dev/mmcblk0"
```

```
root@raspberrypi4-64:~# lsmod
Module                  Size  Used by
rfcomm                 53248  2
cmac                   16384  3
algif_hash             20480  1
aes_arm64              16384  3
algif_skcipher         20480  1
af_alg                 32768  6 algif_hash,algif_skcipher
bnep                   28672  2
hci_uart               49152  1
btbcm                  28672  1 hci_uart
bluetooth             483328  31 hci_uart,btbcm,bnep,rfcomm
ecdh_generic           16384  2 bluetooth
ecc                    36864  1 ecdh_generic
rpivid_mem             16384  0
brcmfmac              339968  0
brcmutil               24576  1 brcmfmac
hid_lg_g15             20480  0
bcm2835_v4l2           45056  0
bcm2835_codec          53248  0
v4l2_mem2mem           45056  1 bcm2835_codec
bcm2835_isp            32768  0
bcm2835_mmal_vchiq     40960  3 bcm2835_codec,bcm2835_v4l2,bcm2835_isp
videobuf2_vmalloc      20480  1 bcm2835_v4l2
videobuf2_dma_contig    24576  2 bcm2835_codec,bcm2835_isp
videobuf2_memops       20480  2 videobuf2_vmalloc,videobuf2_dma_contig
videobuf2_v4l2         32768  4 bcm2835_codec,bcm2835_v4l2,v4l2_mem2mem,bcm2835_isp
v3d                    86016  3
videobuf2_common       69632  8 bcm2835_codec,videobuf2_vmalloc,videobuf2_dma_contig,videobuf2_v4l2,bcm2835_v4l2,v4l2_mem2mem,videobuf2_memops,bcm2835_isp
gpu_sched              49152  1 v3d
videodev              266240  6 bcm2835_codec,videobuf2_v4l2,bcm2835_v4l2,videobuf2_common,v4l2_mem2mem,bcm2835_isp
raspberrypi_hwmon      16384  0
mc                     61440  6 videodev,bcm2835_codec,videobuf2_v4l2,videobuf2_common,v4l2_mem2mem,bcm2835_isp
vc_sm_cma              36864  2 bcm2835_mmal_vchiq,bcm2835_isp
i2c_brcmstb            20480  0
nvmem_rmem             16384  0
uio_pdrv_genirq        16384  0
uio                    24576  1 uio_pdrv_genirq
cfg80211              876544  1 brcmfmac
rfkill                 36864  4 bluetooth,cfg80211
sch_fq_codel           20480  6
fuse                  139264  1
ipv6                  561152  44
```

So for me https://lists.openembedded.org/g/openembedded-core/message/164932 and deleting `${TMPDIR}` resolved all my issues.

agherzan, 2022-04-29:
@MatthiasGugel Thanks for confirming. @kraj We need that in kirkstone too - that as in the oe-core change.

agherzan, 2022-04-29:
The MR automatically closed this issue - and it is fine as we fixed it from this layer's perspective, but we still need to deal with it being broken in kirkstone. It will be backported/cherry-pick in the next round.

closed 2022-04-29 by agherzan

agherzan, 2022-05-03:
@tewarid That is expected. You need the oe-core patch mentioned above.

tewarid, 2022-05-03:
> @tewarid That is expected. You need the oe-core patch mentioned above.

Thanks - sounds good. I'll wait for the patch to make it to the repo.