agherzan/meta-raspberrypi #1055: Can't mix lcd_rotate=2 with touchscreen-inverted-x=1

JeanPhilippeLebel, 2022-04-26:
Using meta-raspberrypi master, rpi4, uboot and offical 7in display.

If I set lcd_rotate=2 in my config.txt, my display is correctly oriented but the touch is not. 

Using
dtoverlay=rpi-ft5406,touchscreen-inverted-x=1,touchscreen-inverted-y=1 

the display and the touch are correctly oriented but the uboot prompt and kernel boot logs are not correctly oriented.

I tried setting both at the same time, but then touch was then incorrectly oriented.

I'm using Wayland with Cage.

Not sure if this is a but or I'm doing something wrong.

agherzan, 2022-04-28:
Could this be related to https://github.com/agherzan/meta-raspberrypi/issues/968 ?

JeanPhilippeLebel, 2022-04-28:
I don't thinks so. The image is correctly displayed.

When I'm using lcd_rotate, the parameters passed to the rpi-ft5406 are simply ignored; the touch system is not rotated.

1) lcd_rotate=0  -> uboot/boot prompt are incorrectly oriented
2) Cage/wayland starts -> The image is incorrectly rotated so I force the rotation from within Cage.
3) At this point, the touch input is correctly rotated.

1) lcd_rotate=2  -> uboot/boot prompt are correctly oriented (I can't test the touch input at this point)
2) Cage/wayland starts -> The image is incorrectly rotated so I force the rotation from within Cage.
3) At this point, the touch input is incorrectly rotated.

I also tested the display_rotate=2 but this had no impact at all.

JeanPhilippeLebel, 2022-05-24:
Ended up using this instead

dtoverlay=vc4-kms-dsi-7inch,invx=1,invy=1
lcd_rotate=2

closed 2022-05-24 by JeanPhilippeLebel

agherzan, 2022-05-26:
Thanks @JeanPhilippeLebel for the follow up.