agherzan/meta-raspberrypi #1058: Incorrect baudrate on UART using RPI Zero W

definable, 2022-05-02:
**Core image base with U-boot has a 'non standard' baudrate on the serial port**

**Steps to reproduce the issue:**
1. Bitbake core-image-base with MACHINE raspberrypi0-wifi
2. Check UART (GPIO14/GPIO15) output 

**Describe the results you received:**
bit time measured: approx 13.9us. 
Configured terminal application to 71942 bit/s, which operated correctly with this baudrate.  

**Describe the results you expected:**
115k2 or another standard baudrate.

U-boot version: 2020.01

agherzan, 2022-05-04:
What kernel version are you using? Is it the default 5.15? If yes, can you try with 5.10?

definable, 2022-05-05:
Sorry my mistake. I should have mentioned it is the u-boot UART. I do not have a console UART configured for the kernel.

agherzan, 2022-05-09:
In that case, I would follow up on the u-boot mailing list about this issue. Otherwise, I'd try similar tests on the Linux side to verify the hardware. If that doesn't work either, it's worth testing without having u-boot in the mix.

agherzan, 2022-07-29:
@definable Any news on this issue?