agherzan/meta-raspberrypi #1063: problem with using wget in branch kirkstone

bigbearishappy, 2022-05-08:
**Description**
My yocto is base on this repo.
When I used wget in [bbappend file](https://github.com/Seeed-Studio/meta-seeed-cm4/blob/90cc0cc4f8a894585224aeda2725a8e4cc9d22b8/recipes-kernel/linux/linux-raspberrypi_5.10.bbappend#L12) of the branch dunfell. It works fine.
But the wget can not work fine when I switch my code to branch kirkstone.

**Steps to reproduce the issue:**
1. prepare the code
```
git clone -b kirkstone git://git.yoctoproject.org/poky
cd poky
git clone -b kirkstone git://git.yoctoproject.org/meta-raspberrypi
git clone -b main https://github.com/Seeed-Studio/meta-seeed-reterminal.git
git clone -b kirkstone https://github.com/meta-qt5/meta-qt5.git
git clone -b kirkstone https://github.com/openembedded/meta-openembedded.git
```
2.prepare the environment
```
cd poky
source oe-init-build-env
bitbake-layers add-layer ../meta-raspberrypi
bitbake-layers add-layer ../meta-seeed-reterminal
bitbake-layers add-layer ../meta-qt5
bitbake-layers add-layer ../meta-openembedded/meta-oe
bitbake-layers add-layer ../meta-openembedded/meta-python
```
3.compile the code
```
MACHINE="seeed-reterminal" bitbake rpi-test-image
```

**Describe the results you received:**
```
| --2022-05-08 14:22:51--  https://raw.githubusercontent.com/Seeed-Studio/seeed-linux-dtoverlays/master/overlays/rpi/reTerminal-overlay.dts
| Resolving raw.githubusercontent.com (raw.githubusercontent.com)... failed: Temporary failure in name resolution.
| wget: unable to resolve host address ‘raw.githubusercontent.com’
| WARNING: exit code 4 from a shell command.
```

**Describe the results you expected:**
The compilation should complete successfully.

**Additional information you deem important (e.g. issue happens only occasionally):**
**I try to compile the branch dunfell.The compilation works fine.**

I would be appreciate if someone can help me out or give me some advice~
Thank you.

agherzan, 2022-05-09:
That doesn't seem to be a BSP issue. Have you checked your network configuration? Can you resolve DNS entries? Can you ping an internet IP? If connectivity is fine, I suspect a wrong DNS configuration.

bigbearishappy, 2022-05-09:
> That doesn't seem to be a BSP issue. Have you checked your network configuration? Can you resolve DNS entries? Can you ping an internet IP? If connectivity is fine, I suspect a wrong DNS configuration.

I have checked my network configuration.It works fine.
And I try to run the wget command to download the dts file. It works fine too.
I will retry it on another computer and let you know.

Thank you for your reply.

kraj, 2022-05-09:
kirkstone has improved in ways we access network and disables network access in all tasks except `do_fetch` by default. Its not clear where this wget request is being made, since its in your own recipe, perhaps you might want to check that the fetching needs are all moved to do_fetch task

bigbearishappy, 2022-05-09:
> kirkstone has improved in ways we access network and disables network access in all tasks except `do_fetch` by default. Its not clear where this wget request is being made, since its in your own recipe, perhaps you might want to check that the fetching needs are all moved to do_fetch task

Oh,I see.I will have a try.
Thank you for your reply.

agherzan, 2022-05-09:
@kraj is right. Your bbappend accesses the network in `configure` which won't work by default anymore.

closed 2022-05-09 by agherzan

bigbearishappy, 2022-05-22:
I tried to change the do_configre to do_fetch.There still be some compile error happened.
But I find another way to fix it.Just enable the network access in the conf file.
And it works fine now.Here is the [commit link](https://github.com/Seeed-Studio/meta-seeed-cm4/commit/48d92941931b6a718135cd989cbc26e017c7b657).

Thank you for your reply again~

agherzan, 2022-05-23:
Keep in mind that the change you pointed to will bypass the default policy for the entire build when a specific machine is selected. I would try to fix the actual problem and not rely on configure to do handle fetching steps.