agherzan/meta-raspberrypi #1064: [Kirkstone] Raspberry Pi CM4 WIFI not working

RajasJoshi, 2022-05-11:
Branch: Kirkstone 
Board: Raspberry Pi CM4 with WIFI and Bluetooth
Kernel Version: 5.15

Local.conf:
```MACHINE ??= "raspberrypi4-64"
DISTRO ?= "poky"
PACKAGE_CLASSES ?= "package_deb"

PREFERRED_VERSION_linux-raspberrypi = "5.15.%"
INHERIT += "rm_work"
EXTRA_IMAGE_FEATURES="debug-tweaks"
IMAGE_FSTYPES = "ext4.gz wic"
IMAGE_FEATURES += "read-only-rootfs"
DISTRO_FEATURES:remove = "pulseaudio"
DISTRO_FEATURES:append = " systemd ${DISTRO_FEATURES_LIBC} wifi"
DISTRO_FEATURES_BACKFILL_CONSIDERED += "sysvinit"
VIRTUAL-RUNTIME_init_manager = "systemd"
VIRTUAL-RUNTIME_initscripts = "systemd-compat-units"

#Peripheral control
ENABLE_DWC2_HOST = "1"
ENABLE_UART = "1"
RPI_USE_U_BOOT = "1"

#Overlays
RPI_KERNEL_DEVICETREE_OVERLAYS:append = " overlays/uart3.dtbo overlays/i2c-rtc.dtbo "

RPI_EXTRA_CONFIG = ' \n \
    arm_64bit=1 \n \
    dtoverlay=uart3 \n \
    '

# Userpace applications
IMAGE_INSTALL:append = " kernel-image kernel-modules hostapd dnsmasq iptables screen "
IMAGE_INSTALL:append = " openssh openssh-sftp-server openssl python3 sqlite3 rpi-gpio raspi-gpio "
IMAGE_INSTALL:append = " linux-firmware-bcm43455 busybox daemontools libmodbus zeromq "
IMAGE_INSTALL:append = " watchdog cronie openvpn usb-modeswitch initscripts init-ifupdown"

# Python libraries
IMAGE_INSTALL:append = " python3-pip python3-django python3-pyserial python3-tornado python3-pymodbus "
IMAGE_INSTALL:append = " python3-pyzmq python3-modbustk"

PREFERRED_VERSION_python3-django = "4.0.2"

USER_CLASSES ?= "buildstats "

PATCHRESOLVE = "noop"

BB_DISKMON_DIRS ??= "\
    STOPTASKS,${TMPDIR},1G,100K \
    STOPTASKS,${DL_DIR},1G,100K \
    STOPTASKS,${SSTATE_DIR},1G,100K \
    STOPTASKS,/tmp,100M,100K \
    ABORT,${TMPDIR},100M,1K \
    ABORT,${DL_DIR},100M,1K \
    ABORT,${SSTATE_DIR},100M,1K \
    ABORT,/tmp,10M,1K"

CONF_VERSION = "2" 
```

I am trying to build an image for the Raspberry Pi 4 , I have added the necessary packages in my local .conf and built the core-image-base version of the image. It compiles successfully but when I run ```ifconfig``` it doesn't shows up. Even if I try ```ifup wlan0``` it shows no such device error. 

[image]

I am trying to setup my board as an Access point. Following is my hostapd.conf, dnsmasq.conf and interfaces files which I have added in my own recipe.  

hostapd.conf
``` 
interface=wlan0
ssid=CM4
wpa_passphrase=raspberrypi
hw_mode=g
ieee80211n=1
channel=6
wmm_enabled=1
ignore_broadcast_ssid=0
auth_algs=1
wpa=2
wpa_key_mgmt=WPA-PSK
rsn_pairwise=CCMP
```

dnsmasq.conf
``` 
interface=wlan0
dhcp-range=10.0.0.2,10.0.0.5,255.255.255.0,12h
```

interfaces
``` 
# The loopback interface
auto lo
iface lo inet loopback

# Wireless interfaces
auto wlan0
iface wlan0 inet static
  address 10.0.0.1
  netmask 255.255.255.0

# Wired or wireless interfaces
auto eth0
iface eth0 inet dhcp
iface eth1 inet dhcp

# Bluetooth networking
iface bnep0 inet dhcp
```

Please let me know if I am missing any packages to be included.

RajasJoshi, 2022-05-13:
Found the solution

The issue is caused by the kernel modules not being loaded caused by xz compresssion, first found in [Issue](https://github.com/agherzan/meta-raspberrypi/issues/1054)
Updated my meta-openembedded layer which fetched the latest changes and solved this issue

closed 2022-05-13 by RajasJoshi

agherzan, 2022-05-16:
For reference, this was fixed in `oe-core/poky` not in `meta-openembeded`.