agherzan/meta-raspberrypi #1069: fuzzy black text on the red background in a 15.6 touchscreen on raspberry kernel 5.10,but kernel 4.19 works well

gitgonewithwind, 2022-06-15:
HI all,
our application need to show black text on the red background as error,

but fuzzy black text on the red background in a 15.6 touchscreen (Lilliput. The model is tk156a ) on raspberry kernel 5.10  ( ycoto meta-raspberrypi e1014d41ef459af807f95cb7ff9105f077bc6daa), the pi 4 hardware versions are  1.4 and 1.2

at same time, I tested updated raspbian image 2022-04-04-raspios-bullseye-arm64.img.xz(not yocto image) from here: https://www.raspberrypi.com/software/ (kernel 5.15)  it has the same issue on  both pi4 hardware 1.4 and 1.2

we have pi 4 hardware version 1.2 ,so I tested older yocto image (kerenl  4.19 , meta-raspberrypi  b112816e95ec59447a82a3a0ee7842c547dda335) ,it works well!( of course ,this image can not startup in hardware 1.4,so  we can not use 1.4 boards from our application )

therefore, I think kernel  4.19  work well for the touchscreen  ,but updated kernel (5.10,5.15) not work well. Maybe the issue is from HDMI drivers in higher kernel?
so do you know where are the HDMI driver files  in kernel? how to modified thems (I mean where is the key code ？）
any help?
I change different resolutions ,but same error.

this is no error phone picture ,you can see at right side  the letters "MN "are same dark, and at left side   letter"c" the upper angle is clear. green and blue background are same clear(sorry, the camera can not catch  clearly ,but it  still  can show it)
[image]

the following  pictures are error , you can see  at right side  the letters "MN "are  NOT same dark(ink is not even), and at left side   letter"c" the upper angle is Not  clear(fuzzy). green and blue background are clear
[image]
ll
[image]

[image]

[image]

agherzan, 2022-07-06:
@gitgonewithwind As you mentioned that this is reproducible on Raspbian, could you please raise this issue against the RaspberryPi kernel repository?

gitgonewithwind, 2022-07-06:
@agherzan 
sure.  how to raise this issue against the RaspberryPi kernel repository? Yes. I think it's the kernel issue
before I tried to question it on here https://forums.raspberrypi.com/viewforum.php?f=28 ,but it never display to show it on the forum.

agherzan, 2022-07-06:
That is exactly what I meant. I'll leave this issue open and if you figure a fix in upstream, let's try to follow up with a patch backport or a kernel bump.

gitgonewithwind, 2022-07-06:
@agherzan  great! hope someone can find a solution. I am trying to use the older HDMI source code to fix it.