agherzan/meta-raspberrypi #1070: u-boot unable to read uboot.env

WalterPrech, 2022-06-17:
I'm using the kirkstone branch and want to start the a Raspberry 4 b with u-boot bootloader. 

I bitbake an image with:
`bitbake -k core-image-full-cmdline`

But the system is not starting while I get the following message in the serial console:

```
-Boot 2022.01 (Jan 10 2022 - 18:46:34 +0000)

DRAM:  3.9 GiB
RPI 4 Model B (0xc03111)
MMC:   mmcnr@7e300000: 1, mmc@7e340000: 0
Loading Environment from FAT... Unable to read "uboot.env" from mmc0:1... In:    serial
Out:   serial
Err:   serial
Net:   eth0: ethernet@7d580000
PCIe BRCM: link up, 5.0 Gbps x1 (SSC)
starting USB...
Bus xhci_pci: Register 5000420 NbrPorts 5
Starting the controller
USB XHCI 1.00
scanning bus xhci_pci for devices... 3 USB Device(s) found
       scanning usb for storage devices... 0 Storage Device(s) found
Hit any key to stop autoboot:  0
U-Boot>
```

local.conf settings:
```
MACHINE ??= "raspberrypi4-64"

IMAGE_FSTYPES = "ext4 rpi-sdimg"
SDIMG_ROOTFS_TYPE = "ext4"
 
KERNEL_IMAGETYPE = "Image"

PREFERRED_VERSION_linux-raspberrypi = "5.15.%"

## enable some hardware
VC4DTBO ?= "vc4-fkms-v3d"
DISABLE_OVERSCAN = "1"
SERIAL_CONSOLES = "115200;ttyS0"

# I2C
ENABLE_I2C = "1"
KERNEL_MODULE_AUTOLOAD:rpi += "i2c-dev i2c-bcm2708"
 
# UART
ENABLE_UART = "1"

## enable uboot
RPI_USE_U_BOOT = "1"   
#PREFERRED_PROVIDER:virtual/bootloader = "u-boot"

## enable video camera
RASPBERRYPI_CAMERA_V2 = "1"

# for no root passwd uncomment the following and comment the two extra user lines
EXTRA_IMAGE_FEATURES = "debug-tweaks"
 
RM_OLD_IMAGE = "1"
#INHERIT += "rm_work"
```

I've also tried other images I can't get a running system with u-boot.

agherzan, 2022-07-06:
It sounds like we are missing `rpi-u-boot-scr` in your build/image. Could you try to see if it got compiled/deployed?

agherzan, 2022-07-13:
`rpi-u-boot-scr` was not pulled in the build?

WalterPrech, 2022-07-14:
No there is no file `* rpi-u-boot-scr*` in the directory `/rpi4-64-uboot/tmp/work/raspberrypi4_64-poky-linux/u-boot/1_2022.01-r0/build`, `/rpi4-64-uboot/tmp/work/raspberrypi4_64-poky-linux/u-boot/1_2022.01-r0/deploy-u-boot` and in `/rpi4-64-uboot/tmp/deploy/images/raspberrypi4-64`.

But in the log files I found the following:
```
log/cooker/raspberrypi4-64/20220622114225.log:NOTE: recipe rpi-u-boot-scr-1.0-r0: task do_install: Started
log/cooker/raspberrypi4-64/20220622114225.log:NOTE: recipe rpi-u-boot-scr-1.0-r0: task do_install: Succeeded
log/cooker/raspberrypi4-64/20220622114225.log:NOTE: Running task 3096 of 3154 (/workdir/poky-kirkstone/meta-raspberrypi/recipes-bsp/rpi-u-boot-scr/rpi-u-boot-scr.bb:do_populate_sysroot)
log/cooker/raspberrypi4-64/20220622114225.log:NOTE: recipe rpi-u-boot-scr-1.0-r0: task do_populate_sysroot: Started
log/cooker/raspberrypi4-64/20220622114225.log:NOTE: recipe rpi-u-boot-scr-1.0-r0: task do_populate_sysroot: Succeeded
log/cooker/raspberrypi4-64/20220622114225.log:NOTE: Task /workdir/poky-kirkstone/meta-raspberrypi/recipes-bsp/rpi-u-boot-scr/rpi-u-boot-scr.bb:do_populate_sysroot unihash changed to 8d82e49aedd0674590fcaf40075c342ec1f18e0ac2b456f79b8bcd1da8e5b77d
```

agherzan, 2022-07-16:
Can you check with `core-image-base`? I just saw that you are using the cmdline one and I can't remember what that was pulling in.

WalterPrech, 2022-07-22:
Sorry for the late reply. Now I found time to test with `core-image-base`.

I can confirm, that with this image u-boot starts the system correct.

closed 2022-07-22 by WalterPrech