agherzan/meta-raspberrypi #1071: Guidance required : generate boot.img and boot.sign using yocto project for secure boot

Vishwasrao1, 2022-06-18:
As of now , I was manually making boot.img, signing it using usbboot(https://github.com/raspberrypi/usbboot) tool for my custom yocto based image.

I would like to incorporate above step in yocto toolchain. So that I will get final signed image from yocto itself. The hardware is rpi CM4. Once I have achieved this I would like to enhance this and generate complete chain of trust including rootfs with help of yocto project.

I was not sure how to start with this, I was wondering if someone from the community has already given it a try or know how to proceed with this. If anyone can guide me or would like to contribute that will be really helpful.

agherzan, 2022-07-06:
This looks like something useful to have in. We would need to have a way to generate the images that create a signed fat fs image. It shouldn't be that complicated, to be honest. We would need to move things from the boot partition to a filesystem image file with the ability to expose the key for signing. Obviously, the integration of the RPI sign tools will also be needed. This sounds like a variable which would enable the above. How does that sound to you?

Vishwasrao1, 2022-07-06:
Thank You @kraj for your comment.
@agherzan Yes, you understood it correctly  and that's what I am currently working on.

Right now I am trying to achieve it using post-build script in yocto. People have already used postbuild.sh in buildroot to achieve the same. (https://github.com/raspberrypi/usbboot/issues/147)
So my goal is: 
1. copy content of boot partition to specific path. 
2. Use this content to generate the boot.img utilising usbboot make-boot-image.   
3. Sign this image using usbboot rpi-eeprom-digest.
4. In the end copy this newly generated boot.img and boot.sign into original boot partition 
5. Make the image using this updated boot partition.

I will use the same key (used in point 3) to sign the EEPROM OTP Fuses. Doing this we can achieve a secure root of trust with signed boot and kernel image. I am yet to figure out about signing the rootFS and adding it to the chain of trust. This is something I will do next.

So any inputs regarding post build script or if you know any alternative in yocto that will be also helpful.

kraj, 2022-07-06:
my suggestion is to bundle the final artifacts using wic process. Having custom SD card generation scripts or process should be avoided for best results.

agherzan, 2022-07-06:
@kraj of course. I didn't propose external tooling to wic.

The way I see it right now is the following. When `SIGN = 1` (variable name to be defined):

1. Filter out assets that are meant to be deployed as a signed fat image from `IMAGE_BOOT_FILES`.
2. Have `wic` depend on a `recipe:do_deploy` (which depends on the recipes providing/deploying all the assets needed in the signed image). This new recipe will generate the signed image by taking the input from the deploy directory and generating the signed image based on a key defined in another variable.
3. Add the signed file in `IMAGE_BOOT_FILES`.

agherzan, 2022-07-06:
As a side note, 1 above can be optional but I don't see why not do this clean and avoid duplication (that has an impact on handling updates for example).

BuriKizilkaya, 2025-03-06:
What is the state of this issue?