agherzan/meta-raspberrypi #1072: How to add PCF8523TK RTC module and autoload it?

ricbrue, 2022-06-23:
Hi, I've already added the ENABLE_I2C = "1" in the local.conf configuration file in Yocto. Once built, I've got this error when trying to scan the I2C bus:
$ sudo i2cdetect -y 1
Error: Could not open file `/dev/i2c-1' or `/dev/i2c/1': No such file or directory

I want to load the driver for the PCF8523(TK) RTC module which is installed on I2C bus (on the expansion connector, GPIO2/3), how can I do that? Thank you.

agherzan, 2022-07-06:
Could you check the kernel configuration for the `CONFIG_I2C_CHARDEV` config?

vulpes2, 2022-07-08:
Merely setting `ENABLE_I2C` is not sufficient, you also need to autoload the i2c drivers, as well as adding a dtoverlay for your I2C RTC chip for the kernel to autoload the correct RTC driver. Full solution will be something like this in your `local.conf`:

```
ENABLE_I2C = "1"
KERNEL_MODULE_AUTOLOAD:rpi += "i2c-dev i2c-bcm2708"

RPI_EXTRA_CONFIG = ' \n \
dtoverlay=i2c-rtc,pcf8523 \n \
'
```

Note that you may need to add additional params to the dtoverlay line if you need to change configuration of the RTC (if applicable). See https://github.com/raspberrypi/firmware/blob/stable/boot/overlays/README#L1551= for details.

closed 2022-07-13 by agherzan