agherzan/meta-raspberrypi #1079: Screen turn black when vc4 is enabled

aurelihein, 2022-08-06:
**Screen turn black when vc4 is enabled**

When I compile a core-image-base for MACHINE=_raspberrypi4-64_ on **kirkstone** branch, with poky also on **kirkstone** branch, the device boot and at the very beginning of the boot the HDMI screen turn off.

To solve this I need to edit config.txt and comment the line `dtoverlay=vc4-kms-v3d`

Probably a **_RDEPENDS_** is missing somewhere

agherzan, 2022-08-11:
Could you compare this with Raspbian?

aurelihein, 2022-08-27:
Hello,
The latest `2022-04-04-raspios-bullseye-armhf-lite.img.xz` have in config.txt **dtoverlay=vc4-kms-v3d**
I just have retested the kirkstone branch and it is still failing
This is how to reproduce it : 
```
mkdir one_test_dir/
cd one_test_dir/
git clone --depth 1 -b kirkstone https://github.com/agherzan/meta-raspberrypi.git
git clone --depth 1 -b kirkstone git://git.yoctoproject.org/poky

source poky/oe-init-build-env rpi-build
echo "BBLAYERS += \"../meta-raspberrypi\"" >> conf/bblayers.conf
echo "MACHINE = \"raspberrypi4-64\"" >> conf/local.conf

bitbake core-image-minimal

#later
SDCARD_IMAGE=core-image-minimal-raspberrypi4-64-20220826125133.rootfs.wic.bz2
bzip2 -dc ${SDCARD_IMAGE} |  sudo dd of=${SDCARD_SLOT} bs=4M conv=notrunc,noerror status=progress ; sync;fin
```

barolo, 2022-09-04:
I have the same issue, works only with fkms [which is a no go for me]

agherzan, 2022-09-14:
Did anyone investigate this against RaspberryPi OS? Kernel version, config etc.

barolo, 2022-09-15:
Yes, to sum it up, it works there, with the same config.

agherzan, 2022-09-15:
What about the kernel version?

barolo, 2022-09-15:
On Yocto, whichever is the most recent for kirkstone, on piOS whichever is the newest, which corresponds to master here [tried it too, no go]

agherzan, 2022-09-16:
So are you saying that the exact same kernel version, same exact fw config (config.txt) and same exact rpi-firmware version behave differently on RPi OS and Yocto?

barolo, 2022-09-16:
Yes, that's what I'm saying.

agherzan, 2022-09-22:
I would be very surprised if that would be the case because I know Oniro has demos on Raspberry Pi 4 (64) and with the current layer it works for HDMI use cases too. Do you have a minimum setup that I can use to reproduce?

barolo, 2022-09-23:
> I would be very surprised if that would be the case because I know Oniro has demos on Raspberry Pi 4 (64) and with the current layer it works for HDMI use cases too. Do you have a minimum setup that I can use to reproduce?

hdmi works with fkms-3d [which is incredibly laggy], just not with "full kms" [and piOS does]

agherzan, 2022-09-23:
I see. That makes sense now. This needs a bit of debugging.

FederAndInk, 2023-06-01:
has this issue been solved? I have the same issue on most recent kirkstone

agherzan, 2023-06-05:
There was no progress on this as far as I am aware.

FederAndInk, 2023-06-29:
So after a bit of testing, it is working with rpi-test-image but not with core-image-minimal. rpi-test-image is based on core-image-base, so I tested it, and it is working too.

FederAndInk, 2023-06-30:
Alright, I found it!

After some binary search: `kernel-module-i2c-brcmstb-5.15.34-v7l` needs to be installed.

Here are now the added packages (depends):.

```diff
> kernel-5.15.34-v7l - 1:5.15.34+git0+e1b976ee4f_0086da6acd-r0                                            
> kernel-image-5.15.34-v7l - 1:5.15.34+git0+e1b976ee4f_0086da6acd-r0                                      
> kernel-image-zimage-5.15.34-v7l - 1:5.15.34+git0+e1b976ee4f_0086da6acd-r0                               
> kernel-module-i2c-brcmstb-5.15.34-v7l - 1:5.15.34+git0+e1b976ee4f_0086da6acd-r0
```

By the way, I also found out that kernel-image-zimage is installing zImage-5.15.34-v7l in `/boot` but yocto do it separatly so the it is in the rootfs while `/boot` isn't mounted. on the target I have kernel7l.img instead:

```
root@raspberrypi4:~# opkg files kernel-image-zimage-5.15.34-v7l
Package kernel-image-zimage-5.15.34-v7l (1:5.15.34+git0+e1b976ee4f_0086da6acd-r0) is installed on root and has the following files:
/boot/zImage-5.15.34-v7l

root@raspberrypi4:~# ls /boot/
bcm2708-rpi-b-plus.dtb        bcm2709-rpi-2-b.dtb           bcm2711-rpi-400.dtb           fixup4.dat                    fixup_x.dat                   start4cd.elf
bcm2708-rpi-b-rev1.dtb        bcm2710-rpi-2-b.dtb           bcm2711-rpi-cm4.dtb           fixup4cd.dat                  kernel7l.img                  start4db.elf
bcm2708-rpi-b.dtb             bcm2710-rpi-3-b-plus.dtb      bootcode.bin                  fixup4db.dat                  overlays                      start4x.elf
bcm2708-rpi-cm.dtb            bcm2710-rpi-3-b.dtb           cmdline.txt                   fixup4x.dat                   rpi-bootfiles-20220830.stamp  start_cd.elf
bcm2708-rpi-zero-w.dtb        bcm2710-rpi-cm3.dtb           config.txt                    fixup_cd.dat                  start.elf                     start_db.elf
bcm2708-rpi-zero.dtb          bcm2711-rpi-4-b.dtb           fixup.dat                     fixup_db.dat                  start4.elf                    start_x.elf
```

Awarty, 2023-09-22:
Hi!

I am currently moving from using the Dunfell branch to Kirkstone. The reason is that QtWebEngine needs Hardknott or newer. Before the change, we had no problem with our graphics. Now, after boot the HDMI output stops and the screen turns off. I tried to comment out *dtoverlay=vc4-kms-v3d* from config.txt and this allows the device to boot and show a console but Qt can't render to the screen. 

Did you find a solution @FederAndInk?

Is this fixed by any newer versions @agherzan? 

Would appreciate any help you can give on this issue.

kraj, 2023-09-22:
@Awarty see this chage

https://github.com/YoeDistro/yoe-distro/blob/master/conf/projects/rpi4-64/config.conf#L53

MultiXaNos, 2024-10-13:
> Alright, I found it!
>
> After some binary search: `kernel-module-i2c-brcmstb-5.15.34-v7l` needs to be installed.
>
> Here are now the added packages (depends):.
>
> ```diff
> ```
>
> By the way, I also found out [...]

I am new in learning Yocto, so can you explain how you add those packages to the image ? I am facing the same issue and I want to use core-image-minimal as a base to get the lighter OS. Thanks by advance for the help!

kraj, 2024-10-13:
@MultiXaNos `core-image-base` is perhaps better than `core-image-minimal` since you want to include basic building blocks and not wrestle with these util you get some experience with the system. When building final image, boot elements are emitted into its own partition. the `/boot` that comes with main rootfs is not used so perhaps its best to remove that from image, you can do so by setting

```
RDEPENDS:${KERNEL_PACKAGE_NAME}-base = ""
```

davidlekve, 2024-10-23:
I have had the same issue with scarthgap, raspberrypi5 and core-image-full-cmdline

trieu1162000, 2025-03-22:
I encountered the similar issue. Are there any solutions so far?