agherzan/meta-raspberrypi #1088: [dunfell] Sporadic freeze due to MMC issues

sahib, 2022-09-11:
Hello,

I'm currently debugging a nasty freeze on a CM3+ with a 4.19 kernel and the dunfell branch of this repo
and I hope to find someone pushing me in the right direction. In my case the freeze symptoms look like this:

* Everything from the ui to all other services of the device lock up.
* No new SSH connections are accepted (but previously existing ones, most of the time, persist)
* Attached USB keyboards do not get recognized (no num lock, SysRQ keys seemingly not working - but will try again)
* This usually happens within the first few minutes of the device running (most I/O happens there).
* More I/O over a long time range seems to trigger the bug more often (I don't have clear data on this yet),
  but disabling quite some of our applications make the freeze less likely.
* MemFree & MemAvailable seem to rise in moment of freeze. So not a typical OOM situation.
* If it happens, it often happens in batches, then often stays away for a few days.
* The CM3 still gets hot during the freeze, indicating load.
* Hard shutdown makes the system behave like nothing happened. No logs.

## System information

<details><summary>System info:<summary>

```bash
$ vcgencmd version  # also newer/older firmware version was tried without any change.
May 27 2021 14:04:13
Copyright (c) 2012 Broadcom
version 7d9a298cda813f747b51fe17e1e417e7bf5ca94d (clean) (release) (start)

$ uname -a
Linux <hostname> 4.19.127-v7 #1 SMP Thu Jul 28 07:59:25 UTC 2022 armv7l GNU/Linux

$ cat /proc/cmdline
8250.nr_uarts=1 bcm2708_fb.fbwidth=480 bcm2708_fb.fbheight=800 bcm2708_fb.fbswap=1 dwc_otg.lpm_enable=0 usbhid.mousepoll=0 vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000 cma=512M@128M coherent_pool=6M fbcon=vc:2-4 logo.nologo quiet video=HDMI-A-1:480x800MR-24@60 ostree=/ostree/boot.1/poky/2fb5f4d11ab0a08ad01ad2b44fee499ee3d129c276dd0f41184fdc79850e413e/0  ostree_root=/dev/mmcblk0p2 root=/dev/ram0 rw rootwait rootdelay=2 ramdisk_size=8192 panic=1

$ cat config.txt
disable_overscan=1
gpu_mem=128
boot_delay=0
boot_delay_ms=0
disable_splash=1
dispmanx_offline=1
dtparam=i2c1=on
dtparam=i2c_arm=on
enable_uart=1
dtoverlay=vc4-kms-v3d
hdmi_edid_file=1
avoid_warnings=2
mask_gpu_interrupt0=0x400
dtparam=audio=on
```

</details>

## Anything similar on the internet?

Yes. There are plenty of reports of this behavior on the internet (google: "mmc_rescan freeze").
Here's a small selection:

* https://github.com/raspberrypi/linux/issues/1885
* https://github.com/raspberrypi/firmware/issues/731 (linux 4.4.38, 4.14.50, 5.10.3)
* https://github.com/raspberrypi/firmware/issues/1522 (linux 5.15)

*Summary:* What all of those have sadly in common: There is no clear fix. Some had luck with up- or downgrading their kernel, some with playing around with `sysctl vm.{swappiness,dirty_ratio,min_free_kbytes}`. It also happens from Pi2 up to Pi4. In my case it's a CM3+, so likely a common (software?) issue and none of the proposed fixes seem to work.

## Debugging attempts:

When disabling swap, the system seems a little more "stable" during the freeze. SSH connections started before the freeze seem to still run if they do not use any I/O. For example this command in a while loop reported that the memory usage of the system is quite okay:

```
cat /proc/meminfo | grep '\(MemAvailable\|MemFree\)' | tr -d '\n' | awk '{ print $2 / 1024 " " $4 / 1024 }'
```
 the official power supply (as well as in our own custom hardware).
[... 484 lines cut ...]
[  367.370597] [<802af1a4>] (__swap_writepage) from [<802af43c>] (swap_writepage+0x40/0x90)
[  367.378692] [<802af43c>] (swap_writepage) from [<80279e00>] (shrink_page_list+0x828/0xfb4)
[  367.386961] [<80279e00>] (shrink_page_list) from [<8027ad6c>] (shrink_inactive_list+0x244/0x714)
[  367.395742] [<8027ad6c>] (shrink_inactive_list) from [<8027bab4>] (shrink_node_memcg+0x394/0x770)
[  367.404617] [<8027bab4>] (shrink_node_memcg) from [<8027bf7c>] (shrink_node+0xec/0x4f0)
[  367.412625] [<8027bf7c>] (shrink_node) from [<8027d474>] (kswapd+0x41c/0x808)
[  367.419764] [<8027d474>] (kswapd) from [<80142f64>] (kthread+0x140/0x170)
[  367.426549] [<80142f64>] (kthread) from [<801010ac>] (ret_from_fork+0x14/0x28)
[  367.433771] Exception stack(0xb73d7fb0 to 0xb73d7ff8)
[  367.438824] 7fa0:                                     00000000 00000000 00000000 00000000
[  367.447006] 7fc0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[  367.455177] 7fe0: 00000000 00000000 00000000 00000000 00000013 00000000
```
</details>

One could suspect undervoltage here, but it happens also in the [Compute Module IO Board](https://www.raspberrypi.com/products /compute-module-io-board-v3) with the official power supply (as well as in our own custom hardware).

## Questions

1. What can make the mmc get stuck in mmc_rescan?
2. Will upgrading the kernel likely help? Some people report this issue with 5.x too,
   and upgrading will be hard for us for various reasons.
3. And most importantly: **How to continue debugging this?**

Also thank you all for the work on this layer. Hope this is the (a?) right place to ask.

agherzan, 2022-09-14:
I would firstly try to reproduce this with an io stress program. Once that is easy to trigger, attach a serial and reproduce over serial and see if that freezes too. Try to see if that gives you a bit more insight. Nevertheless, this is most probably a kernel bug so I'd encourage you to raise it there as well. Have you tried to reproduce it on later kernel versions? If it works, we could try to bisect too.

sahib, 2022-09-15:
Hi @agherzan,

thanks for the input. I tried to run ``stress-ng --hdd 16 --io 16`` on the system. This caused quite a bit of lockup, but nothing fatal yet. I still could connect and (very slowly) execute file system related commands. Maybe it's more caused by specific IO patterns? I will do more testing here.

What had more effect though was adding this dtoverlay (I guess you can do it also with ``dtparam`` in config.txt):

```c
/dts-v1/;
/plugin/;

/ {
	compatible = "brcm,bcm2708";
	fragment@0 {
		target = <&sdhost>;
		__overlay__ {
			non-removable;
                        brcm,force-pio;
		};
	};
};
```
This disables DMA in the sdhost driver and also disables the periodic mmc_rescan check that seemed to be part of the freeze. This had the effect of having  ~3-4 freezes in a day on a few test systems to only having 1 in 2 days. Could be due to the slower I/O in general though due to disabled DMA. I will check what half of the change actually triggered the benefits.

> Have you tried to reproduce it on later kernel versions? 

Not yet, we had a bit of trouble updating to 5.x due to some u-boot issues last time which were hard to debug too. Maybe we need to go that route though. Others reported same issues in 5.15 though so I had low hopes. But likely one of the next steps...

agherzan, 2022-09-15:
@sahib Is there anyway you can try to reproduce on RaspberryPi OS?

sahib, 2022-09-15:
Hmm, would be kinda hard. We have quite a bit of changes in our OS that is required for the apps we run on top of it. Rebuilding this on-top RaspberryPi OS would require considerable work and time. Or put differently: Maybe as desperate action if there's no progress, but not as next step. But yes, I consider it likely that we (or some other layer) did something that differs from RPi OS.

sahib, 2022-09-16:
We tried different firmware versions already, but to no avail. The firmware config is rather basic, as you can see in the first post. No overclocking and no special settings really. Yes, we do use u-boot from [meta-updater-raspberrypi](https://github.com/advancedtelematic/meta-updater-raspberrypi). Where do you see a connection there?

sahib, 2022-09-23:
Hmm, it really seems that the device tree overlay I posted above fixed the issue for us. We have only one device doing strange stuff after some days, but we suspect some other (maybe hardware) issue on that one. The symptoms differ a bit (full lockup, not even reacting to sysrq keys, no activity LED) and 5 other devices run fine through the whole time. I'll post again if that changes - really hard to confidently claim the freeze does not happen again.

From some testing we also know that the ``non-removable`` part does not seem strictly necessary (although testing time was relatively short), but the ``brcm,force-pio`` part is definitely necessary since we had a freeze again after removing this. It is yet unclear if this is a real fix or just makes the freeze less likely due to the slower IO. It probably makes sense to share this info over on the issue tracker for the rpi-kernel and test again when we eventually updated. I'll do this when I have some spare minutes.

While I'm happy this seems fixed, I would still be grateful for any pointers on why forcing PIO seems to fix this.

agherzan, 2022-09-23:
This is an interesting find and I remember seeing this set for RPi4 in the main dt. We would need to ask the rpi kernel team for some pointers here. I wouldn't know without debugging this a bit.

agherzan, 2022-09-23:
Also, have you seen https://github.com/raspberrypi/linux/issues/1536? Looks similar.

sahib, 2022-09-30:
> This is an interesting find and I remember seeing this set for RPi4 in the main dt. We would need to ask the rpi kernel team for some pointers here. I wouldn't know without debugging this a bit.

Alright, I still need to create a ticket over there. Got sidetracked a little & hope to get time for that soon.

> Also, have you seen https://github.com/raspberrypi/linux/issues/1536? Looks similar.

Thanks for the link. Not exactly sure if related, since their issue is rather binary: either it works or it doesn't.

In the meantime I captured another log of blocked tasks with only ``non-removable`` set. They differ quite a lot from the one without ``non-removable`` (which had ``mmc_rescan`` and only a few other tasks in them). There are a lot of tasks and the memory info at the top seems to indicate that there's enough free pages. A bit of swap was used and applications/tasks using it seem to be stuck completely too. Another bunch of tasks are stuck in ``rpi_firmware_transaction`` which is interesting since they should not be blocked by MMC.

[freeze-with-non-removable-but-without-force-pio.log](https://github.com/agherzan/meta-raspberrypi/files/9682527/freeze-with-non-removable-but-without-force-pio.log)

sahib, 2022-09-30:
Posted the issue now on the kernel issue tracker: https://github.com/raspberrypi/linux/issues/5190