agherzan/meta-raspberrypi #1095: cma=64M in /boot/cmdline.txt causes libcamera-apps to fail for full resolution image capture

tewarid, 2022-10-03:
Is there any need to set `cma=64M` in `/boot/cmdline.txt` by default any more given that it [causes libcamera-apps to fail for full resolution image capture](https://github.com/raspberrypi/libcamera-apps/issues/375)?

https://github.com/agherzan/meta-raspberrypi/blob/c57b464b88769c1336695e48e1eb8f35e258b9ae/recipes-bsp/bootfiles/rpi-cmdline.bb#L16

agherzan, 2022-10-03:
@tewarid We would need to dig a bit into history to figure that out. I'm not using the camera directly so I haven't checked it out. Are you able to dig a bit into it?

tewarid, 2022-10-03:
@agherzan Thanks - will do that.

tewarid, 2022-10-03:
This was added by 8827040d9c218f68ae3ee6a5814929888ed61581 and hasn't been changed since. It sets `cma=64M` value when `RASPBERRYPI_CAMERA_V2 = "1"`.

Raspberry Pi [camera docs](https://www.raspberrypi.com/documentation/accessories/camera.html) and [cmdline.txt docs](https://www.raspberrypi.com/documentation/computers/configuration.html#command-line-options) do not mention [cma](https://developer.toradex.com/linux-bsp/how-to/linux-features/contiguous-memory-allocator-cma-linux/), so whatever the historical reason was to use that particular value no longer seems relevant.

The default cma values in kernel are way higher than the value specified in `cmdline.txt`

```log
cat /proc/meminfo | grep -i cma
CmaTotal:         262144 kB
CmaFree:          242468 kB
```

Looks like we can let the kernel handle cma.

PR #1096 (2022-10-10): rpi-cmdline: Leave cma value to kernel default
Fixes #1095 where setting cma to a lower value affects the ability of libcamera-apps to capture higher resolution images.
commit d3cdc309 (2022-10-10): rpi-cmdline: Leave cma value to kernel default

Fixes #1095 where setting cma to a lower value affects the ability of
libcamera-apps to capture higher resolution images.
closed 2022-10-10 by agherzan