agherzan/meta-raspberrypi #1102: Cannot run omxplayer properly

esesyn01, 2022-10-20:
**Description**

After migrating to kirkstone, omxplayer on RPI 0-2w (32-bit version) cannot be run properly. Some libraries aren't visible for
omxplayer binary file, even if they are present in correct directory. After directly specifying the path to the libraries, omxplayers
exits with "Aborted" communicate

**Steps to reproduce the issue:**
1. run `omxplayer.bin video.mp4`
2. run `LD_LIBRARY_PATH="/usr/lib:/opt/vc/lib:/usr/lib/omxplayer" omxplayer.bin video.mp4 `

**Describe the results you received:**
Output for 1:
```
omxplayer.bin: error while loading shared libraries: libavutil.so.56: cannot open shared object file: No such file or directory
```
Output for 2:
```
Aborted
```

**Describe the results you expected:**

Properly playing the video

**Additional details (revisions used, host distro, etc.):**

meta-raspberrypi rev: 722c51647c75d63e8c6c53ee214c4e3f0a263ed2
poky rev: 453be4d258f71855205f45599eea04589eb4a369
meta-openembedded rev: 0b78362654262145415df8211052442823b9ec9b

TomaszAIR, 2022-10-21:
Looks like omxplayer sources were updated ~3 months ago https://github.com/agherzan/meta-raspberrypi/commit/3122d5291da5c8583e7f4b1d063f28eac90b6f96. Was it tested on devices?

agherzan, 2022-10-21:
Omxplayer was deprecated upstream in favour of VLC - see https://github.com/agherzan/meta-raspberrypi/pull/985. This means that I don't think there are many users of the current omxplayer support in this layer so unless somebody tests it and pushes eventually required fixes (upgrades), I will most probably remove the support "soon".

TomaszAIR, 2022-11-30:
@agherzan Ok, we will try and use VLC from https://github.com/agherzan/meta-raspberrypi/pull/985 . Was it only tested on rpi4 with X11? We would like to run this on rpi0w2 with framebuffer only.

agherzan, 2022-12-06:
It was only community-tested.