agherzan/meta-raspberrypi #1105: raspberrypi0 2w 64 Wifi not detected

smzahraee, 2022-10-29:
I build  with  `MACHINE ??= "raspberrypi0-2w-64"` with `core-image-minimal` but my board can not detect wifi and i2c 
here is my local.conf config file

```
DISTRO_FEATURES:remove = "x11 3g nfc ppp directfb ptest nfs smbfs zeroconf perf irda"

EXTRA_IMAGE_FEATURES ?= "debug-tweaks"

INIT_MANAGER = "systemd"

TOOLCHAIN_TARGET_TASK:append = " openssl-dev"

IMAGE_INSTALL:append = "\
	 i2c-tools \
	 python3 python3-pip python3-psutil \
	 openssh \
	 openssl libcrypto libssl openssl-dev \
	 libusb1 \
	 wpa-supplicant linux-firmware-bcm43430 \
	 bluez5 udev-rules-rpi \
	"
ENABLE_I2C = "1"
ENABLE_UART = "1"
ENABLE_DWC2_PERIPHERAL = "1"
ENABLE_DWC2_HOST = "1"
```

agherzan, 2022-10-31:
@smzahraee please try `core-image-base`.

smzahraee, 2022-11-01:
Thank you, It works.
Could you say why does it not work with `core-image-minimal` ?

agherzan, 2022-11-01:
Sure. The minimum image is meant only for minimum boot requirements. Technically, it won't pull in the recommended packages defined in the machine configuration - the firmware for WiFi for example.

closed 2022-11-01 by agherzan