agherzan/meta-raspberrypi #1108: How can add python Built distribution(bdist) in yocto build

jahanzaib1234, 2022-11-09:
I  am trying to append some packages in yocto build with dunfell branch but the issue is these python packages required setuptools but setuptools not sported by dunfell.

My question is can i add  python package Built distribution(bdist) rather then Source distribution(sdist)  directly to append packages in yocto build if yes then how. I am new in yocto and sorry if its a very simple question.

agherzan, 2022-11-09:
@jahanzaib1234 I would redirect you to the Yocto mailing list as this is not a Raspberry Pi-specific question. On the other hand, I don't know much about the current status of python support in dunfell as things have changed a fair bit lately so I'd follow up with the lads on the mailing list for this question.

jahanzaib1234, 2022-11-09:
@agherzan thanks for your time