agherzan/meta-raspberrypi #1112: Kirkstone on raspberry pi won't boot

weiqj, 2022-11-23:
I am using this repo with meta-rpi. Honister boots perfectly. However kirkstone won't.

Also, how to enable serial print out of kernel and rootfs logs? Adding ENABLE_UART = "1" in local.conf only enables login terminal on serial.

nickbetteridge, 2022-11-24:
I can confirm this - using a rpi4b, I get a quick bios raspberry image then the hdmi dies. Uart throws out an 'ERR'

agherzan, 2022-11-25:
@weiqj have you tried a "dry" build with a minimum set of kirkstone layers, ENABLE_UART=1, one a basic image (core-image-base)?
@nickbetteridge have you checked your board against a RaspberryPi OS image?

nickbetteridge, 2022-11-25:
@agherzan I haven't checked against a RaspberryPi OS image, but I have just tried a rpi4_64 buildroot image, which also failed - however, the buildroot non-64 image worked - I suspect it's the same with meta-raspberrypi - I'll try and find time to try meta-raspberrypi out with machine=raspberrypi4

agherzan, 2022-11-25:
Can you update your initial report with a clear and minimum set of build steps that reproduces this issue?

nickbetteridge, 2022-11-25:
I used docker -

```
RUN git clone -b kirkstone git://git.yoctoproject.org/poky.git

WORKDIR /home/nick/poky

RUN git clone -b kirkstone https://github.com/openembedded/meta-openembedded.git

RUN git clone -b kirkstone git://git.yoctoproject.org/meta-virtualization

RUN /bin/bash -c "source ./oe-init-build-env \
    && bitbake-layers add-layer ../meta-openembedded/meta-oe \
    && bitbake-layers add-layer ../meta-openembedded/meta-python \
    && bitbake-layers add-layer ../meta-openembedded/meta-networking \
    && bitbake-layers add-layer ../meta-openembedded/meta-filesystems \
    && bitbake-layers add-layer ../meta-openembedded/meta-perl \
    && bitbake-layers add-layer ../meta-virtualization" || true

WORKDIR /home/nick/poky/build

RUN echo 'DISTRO_FEATURES += " virtualization"' >> conf/local.conf
RUN echo 'ENABLE_UART = "1"' >> conf/local.conf

WORKDIR /home/nick/poky

RUN git clone -b kirkstone git://git.yoctoproject.org/meta-raspberrypi.git

RUN /bin/bash -c "source ./oe-init-build-env \
    && bitbake-layers add-layer ../meta-raspberrypi" || true

WORKDIR /home/nick/poky/build

RUN echo 'MACHINE = "raspberrypi4-64"' >> conf/local.conf

WORKDIR /home/nick/poky

RUN /bin/bash -c "source ./oe-init-build-env && bitbake core-image-minimal" || true
```

agherzan, 2022-11-25:
@nickbetteridge That is not a minimum set of reproducible steps. I would expect a setup that has oe-core + meta-rpi with specific revisions, poky distro and a core image.

weiqj, 2022-11-25:
I am building 32 bit image of Raspberry Pi 4.
I noticed that the green led flashes 7 times, which indicates "kernel.img not found".
But I already put zImage onto the SD card.

weiqj, 2022-11-25:
Nevermind, I renamed zImage into kernel.img, now the Rpi4 is able to boot correctly.
meta-rpi script needs to be modified
copy_boot.sh, line 199: from
sudo cp "${src}/${KERNEL_IMAGETYPE}" "${mnt}/${KERNEL_IMAGETYPE}"
to
sudo cp "${src}/${KERNEL_IMAGETYPE}" "${mnt}/kernel.img"

I am wondering why it worked before.

agherzan, 2022-11-26:
What script is that? I'm missing a bit of context here.

weiqj, 2022-11-26:
> What script is that? I'm missing a bit of context here.

The script is "copy_boot.sh" from meta-rpi.
It literally copies "zImage" to the boot partition. Apparently somehow the new bootloader expects a file name of "kernel.img"

Renaming zImage into kernel.img solves the problem.
I am wondering why the name zImage worked before.

agherzan, 2022-11-26:
Sorry, what is meta-rpi?

weiqj, 2022-11-26:
> 

https://github.com/jumpnow/meta-rpi

I-am-Harsh, 2023-01-17:
I have run into this same issue, dunfell through honister work fine. 
The rpi starts booting, I am able to see the boot screen for about 1 sec or less, and there is blackout, HDMI output stops, there is no ACT light blinks. I have been trying to use core-image-minimal and core-image-minimal-dev, 
my bb layers are --> 

```
\
  /local/home/harshmav/rpi_builds/honister/poky/meta \
  /local/home/harshmav/rpi_builds/honister/poky/meta-poky \
  /local/home/harshmav/rpi_builds/honister/poky/meta-yocto-bsp \
/local/home/harshmav/rpi_builds/honister/poky/meta-raspberrypi  "
```

The only other change that I have done except for adding the BSP is specifying the machine and image type

FederAndInk, 2023-06-02:
This seems to be linked to https://github.com/agherzan/meta-raspberrypi/issues/1079