agherzan/meta-raspberrypi #1113: [raspberrypi4-64][honister] timedatectl timesync not working

bojidartonchev, 2022-11-23:
Hello,

After migrating from dunfell to honister, time synchronization stopped working for `raspberrypi4-64`. On dunfell, it takes about 3 seconds after connecting to Wifi to sync the time. On honister it never does.

I have the following patch for systemd's timesyncd.conf:
```
[Time]
NTP=ntp.ubuntu.com
RootDistanceMaxSec=30s
```

The `ntp.ubuntu.com` is reachable.

This is the `timedatectl` output:
```
root@raspberrypi4-64:~# timedatectl
               Local time: Fri 2021-11-19 17:20:49 UTC
           Universal time: Fri 2021-11-19 17:20:49 UTC
                 RTC time: n/a
                Time zone: Universal (UTC, +0000)
System clock synchronized: no
              NTP service: active
          RTC in local TZ: no
```

agherzan, 2022-11-25:
Is the network interface up?

Piocky, 2023-01-06:
It works well for you on kirkstone? I've got kind of the same result on my side (switching from dunfell to kirkstone), except that it takes approximately 3-4 minutes to get timesync. 
The only solution I found to reduce this time is to add:
```
ExecStartPre=/usr/sbin/ntpd -g -q
ExecStart=/usr/sbin/ntpd -u ntp:ntp -p /run/ntpd.pid -g
```
in /lib/systemd/system/ntpd.service

From what I understand, it seems that systemd-timesyncd.service is used on dunfell:
```
root@raspberrypi4:~# journalctl -u  systemd-timesyncd
Jan 01 01:00:04 localhost systemd[1]: Starting Network Time Synchronization...
Jan 01 01:00:04 localhost systemd[1]: Started Network Time Synchronization.
Jan 01 01:00:10 Piccolo-Noire systemd-timesyncd[208]: Network configuration changed, trying to establish connection.
Jan 06 12:42:25 Piccolo-Noire systemd-timesyncd[208]: Initial synchronization to time server 216.239.35.0:123 (time1.google.com).
```

On kirkstone i've got nothing:
```
root@raspberrypi4:~# journalctl -u  systemd-timesyncd
Jan 01 01:00:05 localhost systemd[1]: Starting Network Time Synchronization...
Jan 01 01:00:05 localhost systemd[1]: Started Network Time Synchronization.
```

But I do see stuff with ntpd.service:
```
root@raspberrypi4:~# journalctl -u ntpd
Jan 01 01:00:06 raspberrypi4 systemd[1]: Starting Network Time Service...
Jan 01 01:00:06 raspberrypi4 ntpd[293]:  1 Jan 01:00:06 ntpd[293]: ntpd 4.2.8p15@1.3728-o Tue 23 Jun 2020 09:22:22 AM UTC (1): Starting
Jan 01 01:00:06 raspberrypi4 ntpd[293]:  1 Jan 01:00:06 ntpd[293]: Command line: /usr/sbin/ntpd -g -q
Jan 01 01:00:06 raspberrypi4 ntpd[293]:  1 Jan 01:00:06 ntpd[293]: ----------------------------------------------------
Jan 01 01:00:06 raspberrypi4 ntpd[293]:  1 Jan 01:00:06 ntpd[293]: ntp-4 is maintained by Network Time Foundation,
Jan 01 01:00:06 raspberrypi4 ntpd[293]:  1 Jan 01:00:06 ntpd[293]: Inc. (NTF), a non-profit 501(c)(3) public-benefit
Jan 01 01:00:06 raspberrypi4 ntpd[293]:  1 Jan 01:00:06 ntpd[293]: corporation.  Support and training for ntp-4 are
Jan 01 01:00:06 raspberrypi4 ntpd[293]:  1 Jan 01:00:06 ntpd[293]: available at https://www.nwtime.org/support
Jan 01 01:00:06 raspberrypi4 ntpd[293]:  1 Jan 01:00:06 ntpd[293]: ----------------------------------------------------
Jan 01 01:00:06 raspberrypi4 ntpd[293]:  1 Jan 01:00:06 ntpd[293]: proto: precision = 6.926 usec (-17)
Jan 01 01:00:06 raspberrypi4 ntpd[293]:  1 Jan 01:00:06 ntpd[293]: basedate set to 2020-06-11
Jan 01 01:00:06 raspberrypi4 ntpd[293]:  1 Jan 01:00:06 ntpd[293]: gps base set to 2020-06-14 (week 2110)
Jan 01 01:00:06 raspberrypi4 ntpd[293]:  1 Jan 01:00:06 ntpd[293]: Listen and drop on 0 v6wildcard [::]:123
Jan 01 01:00:06 raspberrypi4 ntpd[293]:  1 Jan 01:00:06 ntpd[293]: Listen and drop on 1 v4wildcard 0.0.0.0:123
Jan 01 01:00:06 raspberrypi4 ntpd[293]:  1 Jan 01:00:06 ntpd[293]: Listen normally on 2 lo 127.0.0.1:123
Jan 01 01:00:06 raspberrypi4 ntpd[293]:  1 Jan 01:00:06 ntpd[293]: Listen normally on 3 lo [::1]:123
Jan 01 01:00:06 raspberrypi4 ntpd[293]:  1 Jan 01:00:06 ntpd[293]: Listening on routing socket on fd #20 for interface updates
Jan 01 01:00:14 raspberrypi4 ntpd[293]:  1 Jan 01:00:14 ntpd[293]: Listen normally on 4 wlan0 [fe80::da5b:a660:4196:a985%3]:123
Jan 01 01:00:14 raspberrypi4 ntpd[293]:  1 Jan 01:00:14 ntpd[293]: new interface(s) found: waking up resolver
Jan 01 01:00:21 raspberrypi4 ntpd[293]:  1 Jan 01:00:21 ntpd[293]: Listen normally on 5 wlan0 192.168.0.101:123
Jan 01 01:00:21 raspberrypi4 ntpd[293]:  1 Jan 01:00:21 ntpd[293]: new interface(s) found: waking up resolver
Jan 06 16:13:10 raspberrypi4 ntpd[293]:  6 Jan 16:13:10 ntpd[293]: ntpd: time set +1673017958.292877 s
Jan 06 16:13:10 raspberrypi4 ntpd[293]: ntpd: time set +1673017958.292877s
Jan 06 16:13:10 raspberrypi4 systemd[1]: Started Network Time Service.
```

Is there a reason for systemd-timesyncd not being used anymore? The service is still up and running though.