agherzan/meta-raspberrypi #1117: VLC on RaspberryPI0w2 with framebuffer only

TomaszAIR, 2022-11-30:
**Description**

Want to use vlc without X11 and run it on fb only.

**Steps to reproduce the issue:**
1. Add `vlc` to IMAGE_INSTALL
2. Set machine to `raspberrypi0-2w`
3. Do not set x11 or wayland in DISTRO_FEATURES
4. Remove `vc4graphics` from MACHINE_FEATURES and use userland
5. Run build using kas-container v3.0.2 https://raw.githubusercontent.com/siemens/kas/3.0.2/kas-container

**Describe the results you received:**

We are using custom layer which uses meta-raspberrypi. Previously our project used omxplayer but we were told to try and switch to VLC as it gets more support https://github.com/agherzan/meta-raspberrypi/issues/1102.

rpidistro-vlc got some switches on x11 https://github.com/agherzan/meta-raspberrypi/blob/master/dynamic-layers/multimedia-layer/recipes-multimedia/rpidistro-vlc/rpidistro-vlc_3.0.12.bb#L53 so we try build it to run on fb. What we got are following build errors:

1. First is that running `bitbake ffmpeg` start ffmpeg from poky/meta

When force to build rpidistro-ffmpeg in return we get

```
ERROR: Nothing PROVIDES 'rpidistro-ffmpeg'
rpidistro-ffmpeg was skipped: incompatible with host arm-project-linux-gnueabi (not in COMPATIBLE_HOST)
```

This is probably because of https://github.com/agherzan/meta-raspberrypi/blob/master/recipes-multimedia/rpidistro-ffmpeg/rpidistro-ffmpeg_4.3.2.bb#L193, should not be there just `COMPATIBLE_HOST:rpi = "(.*)"` - so the same for https://github.com/agherzan/meta-raspberrypi/blob/master/dynamic-layers/multimedia-layer/recipes-multimedia/rpidistro-vlc/rpidistro-vlc_3.0.12.bb#L157 - after that `rpidistro-ffmpeg` starts build but fails on opengl check (mesa-gl successfully build with this fix https://github.com/agherzan/meta-raspberrypi/issues/987#issuecomment-1020306027)

```
| DEBUG: Python function extend_recipe_sysroot finished
| DEBUG: Executing shell function autotools_preconfigure
| DEBUG: Shell function autotools_preconfigure finished
| DEBUG: Executing python function autotools_aclocals
| DEBUG: SITE files ['endian-little', 'bit-32', 'arm-common', 'arm-32', 'common-linux', 'common-glibc', 'arm-linux', 'arm-linux-gnueabi', 'common']
| DEBUG: Python function autotools_aclocals finished
| DEBUG: Executing shell function do_configure
| ERROR: opengl not found.
| 
| If you think configure made a mistake, make sure you are using the latest
| version from Git.  If the latest version fails, report the problem to the
| ffmpeg-user@ffmpeg.org mailing list or IRC #ffmpeg on irc.freenode.net.
| Include the log file "ffbuild/config.log" produced by configure as this will help
| solve the problem.
| WARNING: exit code 1 from a shell command.
ERROR: Task (/work/meta-raspberrypi/recipes-multimedia/rpidistro-ffmpeg/rpidistro-ffmpeg_4.3.2.bb:do_configure) failed with exit code '1'
```

**Describe the results you expected:**

VLC and ffmpeg distro compiled correctly, able to run vlc on rpi0w2 (at least help command)

**Additional information you deem important (e.g. issue happens only occasionally):**

**Additional details (revisions used, host distro, etc.):**

```
Build Configuration:                                                                                                                                                                                               
BB_VERSION           = "2.0.0"                                                                                                                                                                                     
BUILD_SYS            = "x86_64-linux"                                                                                                                                                                              
NATIVELSBSTRING      = "universal"                                                                                                                                                                                 
TARGET_SYS           = "arm-project-linux-gnueabi"                                                                                                                                                                  
MACHINE              = "raspberrypi0-2w"                                                                                                                                                                           
DISTRO               = "project-fb"                                                                                                                                                                             
DISTRO_VERSION       = "4.5.3"                                                                                                                                                                                     
TUNE_FEATURES        = "arm vfp cortexa7 neon vfpv4 thumb callconvention-hard"                                                                                                                                     
TARGET_FPU           = "hard"                                                                                                                                                                                      
repo                 = "add-vlc:a7f9c12f1b7b8d8b230ad960db6830e52f593499"                                
meta-multimedia                                                                                          
meta-networking      
meta-oe              
meta-python          = "HEAD:0b78362654262145415df8211052442823b9ec9b"
meta-raspberrypi     = "HEAD:fc5f80a47ed7761509ac0700c81581c97dfe135a"
meta-swupdate        = "HEAD:2faeb01a7a95e06eb477f70dcd46f5d508c861eb"
meta                 
meta-poky            
meta-yocto-bsp       = "patched-453be4d258f71855205f45599eea04589eb4a369:deebfbf9315ff8f0ac8a2793429785c91eae6f0f"
```

TomaszAIR, 2022-11-30:
@EasyIP2023 looks like you did majority of work for VLC support in meta-raspberrypi. Maybe you will be able to support here?

EasyIP2023, 2022-11-30:
@TomaszAIR Does running it like it show in `Usage example:` not produce output? https://meta-raspberrypi.readthedocs.io/en/latest/extra-build-config.html#raspberry-pi-distro-vlc.

Also, may you add some of the `vlc` flags you are testing with? My main concern was the use of mmal_vout plugin.

TomaszAIR, 2022-12-01:
@EasyIP2023 thanks for quick response.

The problem is that I am not able to compile VLC for rpi0w2 without x11. The revision used of meta-raspberrypi is `fc5f80a47ed7761509ac0700c81581c97dfe135a`. I will test from dropping off our custom layer and test rpi0w2 build just with meta-raspberrypi, x11 removed from DISTRO_FEATURES and DISABLE_VC4GRAPHICS set to `1`.

Will post the results.

EasyIP2023, 2022-12-02:
@TomaszAIR Thanks for helping on finding this. Appears the issue is with https://github.com/RPi-Distro/ffmpeg/blob/pios/bullseye/configure#L6455. `./configure` is looking for `<recipe-sysroot>/usr/include/ES2/gl.h` which doesn't exists. It does however exists in `<recipe-sysroot>/usr/include/GLES/gl.h`.

Also appears I didn't include `virtual/libgles2` in `PACKAGECONFIG[opengl] = "--enable-opengl,--disable-opengl,virtual/libgl"`.

Will try to have add patch that fixes ffmpeg configure bugs.

EasyIP2023, 2022-12-05:
Hey @TomaszAIR when you get the chance can you try out https://github.com/EasyIP2023/meta-raspberrypi/tree/testing. Contains relevent open MR's + rpidistro-vlc version update. 

The VLC recipe update is still a WIP running into a pretty nasty bug when testing. It has the mmal_vout plugin like we all want. You should be able to at least compile both ffmpeg and vlc without x11 or wayland distro features enabled on branch. Just my update will have more time next week on vlc upgrade.

```
roott@raspberrypi4:~# ffmpeg -i 210329_06B_Bali_1080p_013.mp4 -c:v h264_v4l2m2m -b:v 8M -c:a copy encoded_video.mp4
ffmpeg version 4.3.2 Copyright (c) 2000-2021 the FFmpeg developers
Input #0, mov,mp4,m4a,3gp,3g2,mj2, from '210329_06B_Bali_1080p_013.mp4':
  Metadata:
    major_brand     : mp42
    minor_version   : 0
    compatible_brands: mp42mp41
    creation_time   : 2021-04-22T22:58:58.000000Z
  Duration: 00:00:30.00, start: 0.000000, bitrate: 59933 kb/s
    Stream #0:0(eng): Video: h264 (High) (avc1 / 0x31637661), yuv420p(tv, bt709), 1920x1080, 59927 kb/s, 25 fps, 25 tbr, 25k tbn, 50 tbc (default)
    Metadata:
      creation_time   : 2021-04-22T22:58:58.000000Z
      handler_name    : ?Mainconcept Video Media Handler
      encoder         : AVC Coding
Stream mapping:
  Stream #0:0 -> #0:0 (h264 (native) -> h264 (h264_v4l2m2m))
Press [q] to stop, [?] for help
[h264_v4l2m2m @ 0xe42ca0] Using device /dev/video11
[h264_v4l2m2m @ 0xe42ca0] driver 'bcm2835-codec' on card 'bcm2835-codec-encode' in mplane mode
[h264_v4l2m2m @ 0xe42ca0] requesting formats: output=YU12 capture=H264
[h264_v4l2m2m @ 0xe42ca0] Failed to set gop size: Invalid argument
Output #0, mp4, to 'encoded_video.mp4':
  Metadata:
    major_brand     : mp42
    minor_version   : 0
    compatible_brands: mp42mp41
    encoder         : Lavf58.45.100
    Stream #0:0(eng): Video: h264 (h264_v4l2m2m) (avc1 / 0x31637661), yuv420p, 1920x1080, q=-1--1, 8000 kb/s, 25 fps, 12800 tbn, 25 tbc (default)
    Metadata:
      creation_time   : 2021-04-22T22:58:58.000000Z
      handler_name    : ?Mainconcept Video Media Handler
      encoder         : Lavc58.91.100 h264_v4l2m2m
[h264_v4l2m2m @ 0xe42ca0] ff_v4l2_buffer_buf_to_avpkt00.00 bitrate=N/A speed=   0x
[mp4 @ 0xe50bc0] Timestamps are unset in a packet for stream 0. This is deprecated and will stop working in the future. Fix your code to set the timestamps properly
[mp4 @ 0xe50bc0] Encoder did not produce proper pts, making some up.
[h264_v4l2m2m @ 0xe42ca0] ff_v4l2_buffer_buf_to_avpkt:00.00 bitrate=4923.1kbits/s speed=5.8e-05x
    Last message repeated 5 times
[h264_v4l2m2m @ 0xe42ca0] ff_v4l2_buffer_buf_to_avpkt:00.20 bitrate=   1.9kbits/s speed=0.0935x
    Last message repeated 6 times
[h264_v4l2m2m @ 0xe42ca0] ff_v4l2_buffer_buf_to_avpkt:00.48 bitrate=4369.2kbits/s speed=0.16x
    Last message repeated 4 times
[h264_v4l2m2m @ 0xe42ca0] ff_v4l2_buffer_buf_to_avpkt:00.68 bitrate=3084.3kbits/s speed=0.184x
    Last message repeated 6 times
```

```
root@raspberrypi4:~# ffmpeg -decoders | grep mmal
ffmpeg version 4.3.2 Copyright (c) 2000-2021 the FFmpeg developers
  built with gcc 12.2.0 (GCC)
  libavutil      56. 51.100 / 56. 51.100
  libavcodec     58. 91.100 / 58. 91.100
  libavformat    58. 45.100 / 58. 45.100
  libavdevice    58. 10.100 / 58. 10.100
  libavfilter     7. 85.100 /  7. 85.100
  libavresample   4.  0.  0 /  4.  0.  0
  libswscale      5.  7.100 /  5.  7.100
  libswresample   3.  7.100 /  3.  7.100
  libpostproc    55.  7.100 / 55.  7.100
 V..... h264_mmal            h264 (mmal) (codec h264)
 V..... mpeg2_mmal           mpeg2 (mmal) (codec mpeg2video)
 V..... mpeg4_mmal           mpeg4 (mmal) (codec mpeg4)
 V..... vc1_mmal             vc1 (mmal) (codec vc1)
```

```
[... 95 lines cut ...]
[  112.114376] 1f60: c0200264 c3bb0000 c3bb1fa4 c3bb1f78 c020a5ac 2415a519 aeffe700 b4c61178
[  112.122679] 1f80: 76767676 00000004 00000036 c0200264 c3bb0000 00000036 00000000 c3bb1fa8
[  112.130983] 1fa0: c0200244 c048a7dc b4c61178 76767676 00000004 80404a5b aeffe720 aeffe700
[  112.139288] 1fc0: b4c61178 76767676 00000004 00000036 00000000 0000001b af0024c8 0000001a
[  112.147593] 1fe0: 00000036 aeffe6e8 b6ee6fa5 b6e5e576 280f0030 00000004 00000000 00000000
[  112.155892] Backtrace:
[  112.158379] [<c08e73c4>] (dma_buf_dynamic_attach) from [<c08e7634>] (dma_buf_attach+0x20/0x24)
[  112.167150]  r9:c2a366c0 r8:bf1612c0 r7:c3bb0000 r6:76767676 r5:00000000 r4:fffffff7
[  112.175009] [<c08e7614>] (dma_buf_attach) from [<bf15d35c>] (vc_sm_cma_import_dmabuf_internal+0xc8/0x400 [vc_sm_cma])
[  112.185825] [<bf15d294>] (vc_sm_cma_import_dmabuf_internal [vc_sm_cma]) from [<bf15dcc4>] (vc_sm_cma_ioctl+0x104/0x52c [vc_sm_cma])
[  112.197880]  r10:00000004 r9:c2a366c0 r8:bf000000 r7:c3bb0000 r6:c36ed9c0 r5:aeffe720
[  112.205828]  r4:c3bb0000
[  112.208398] [<bf15dbc0>] (vc_sm_cma_ioctl [vc_sm_cma]) from [<c048a8ec>] (sys_ioctl+0x11c/0xb44)
[  112.217354]  r10:00000004 r9:c2a366c0 r8:aeffe720 r7:c3bb0000 r6:c2a366c1 r5:00000000
[  112.225300]  r4:80404a5b
[  112.227869] [<c048a7d0>] (sys_ioctl) from [<c0200244>] (__sys_trace_return+0x0/0x1c)
[  112.235746] Exception stack(0xc3bb1fa8 to 0xc3bb1ff0)
[  112.240877] 1fa0:                   b4c61178 76767676 00000004 80404a5b aeffe720 aeffe700
[  112.249181] 1fc0: b4c61178 76767676 00000004 00000036 00000000 0000001b af0024c8 0000001a
[  112.257482] 1fe0: 00000036 aeffe6e8 b6ee6fa5 b6e5e576
[  112.262615]  r10:00000036 r9:c3bb0000 r8:c0200264 r7:00000036 r6:00000004 r5:76767676
[  112.270563]  r4:b4c61178
[  112.273137] Code: 0a00004f e5d63000 e88400a0 e5c43018 (e5953010)
[  112.279438] ---[ end trace 8ed776ff1e5f5bf7 ]---
```

TomaszAIR, 2022-12-06:
Hey @EasyIP2023, thanks for your effort. I was able to compile and check VLC on rpi0w2 without x11 in distro.

```
# cvlc -H | grep mmal                                                                                                                                                                       
VLC media player 3.0.17.4 Vetinari (revision 1.0.6-1687-gb6a28bbbe)
 MMAL-based decoder plugin for Raspberry Pi (mmal_codec)
      --mmal-opaque, --no-mmal-opaque 
      --mmal-decode-enable, --no-mmal-decode-enable 
                                 Enable mmal decode even if normally disabled
          Enable mmal decode even if normally disabled. MMAL decode is normally
      --mmal-resize, --no-mmal-resize 
                                 Use mmal resizer rather than hvs.
          Use mmal resizer rather than isp. This uses less gpu memory than the
      --mmal-isp, --no-mmal-isp  Use mmal isp rather than hvs.
          Use mmal isp rather than hvs. This may be faster but has no blend.
 mmal_vout (mmal_vout)
      --mmal-layer <integer>     VideoCore layer where the video is displayed.
      --mmal-adjust-refreshrate, --no-mmal-adjust-refreshrate 
      --mmal-native-interlaced, --no-mmal-native-interlaced 
      --mmal-display <string>    Output device for Rpi fullscreen.
      --mmal-vout-transform <string> 
      --mmal-vout-window <string> 
      --mmal-vout-transparent, --no-mmal-vout-transparent 
 MMAL buffered avcodec  (mmal_avcodec)
      --mmal-avcodec-buffers <integer> 
 MMAL-based deinterlace filter plugin (mmal_deinterlace)
      --mmal-deinterlace-no-qpu, --no-mmal-deinterlace-no-qpu 
      --mmal-deinterlace-adv, --no-mmal-deinterlace-adv 
      --mmal-deinterlace-fast, --no-mmal-deinterlace-fast 
      --mmal-deinterlace-none, --no-mmal-deinterlace-none 
      --mmal-deinterlace-half-rate, --no-mmal-deinterlace-half-rate 
      --mmal-deinterlace-full-rate, --no-mmal-deinterlace-full-rate 
      --glconv {any,drm_gl_conv,mmal_converter,none} 
  -V, --vout {any,mmal_xsplitter,gles2,fb,drm_vout,mmal_vout,vmem,yuv,flaschen,vdummy,vdummy,omxil_vout,none}
```

Running vlc ends with similar effect

```
# DISPLAYNUM=$(tvservice -l | tail -c 2)
# MMAL_DISPLAY=$(expr $DISPLAYNUM + 1)
# VLC_SETTINGS="-I dummy --vout=mmal_vout --mmal-resize --mmal-display hdmi-$MMAL_DISPLAY --no-dbus"
# vlc $VLC_SETTINGS video.mp4                                                                                                                                                       [226/1149]
VLC media player 3.0.17.4 Vetinari (revision 1.0.6-1687-gb6a28bbbe)                                      
[01dd3aa0] main interface error: no suitable interface module                                            
[01cbb260] main libvlc error: interface "globalhotkeys,none" initialization failed                       
[01dd3aa0] dummy interface: using the dummy interface module...                                          
[6b833ea0] freetype spu text error: LoadFace: Error creating face for /usr/share/fonts/truetype/freefont/FreeSerifBold.ttf - 0 - 2 - 2
[6b833ea0] freetype spu text error: Error loading default face                                           
[6ce36098] main vout display error: option qt-fullscreen-screennumber does not exist                     
[6b833ea0] freetype spu text error: LoadFace: Error creating f[ 1213.300976] 8<--- cut here ---                                                                                                                    
ace for [ 1213.304870] Unable to handle kernel NULL pointer dereference at virtual address 00000007                                                                                                                
/usr/sha[ 1213.313906] pgd = 9bc9a421                                                                    
re/fonts[ 1213.317223] [00000007] *pgd=00000000                                                          
/truetyp[ 1213.321665] Internal error: Oops: 5 [#2] SMP ARM                                              
e/freefo[ 1213.326950] Modules linked in: brcmfmac bcm2835_codec(C) bcm2835_v4l2(C) brcmutil raspberrypi_hwmon snd_bcm2835(C) videobuf2_vmalloc sha256_generic bcm2835_isp(C) v4l2_mem2mem snd_pcm cfg80211 bcm2836
nt/FreeS[ 1213.369875] CPU: 2 PID: 351 Comm: vlc Tainted: G      D  C        5.15.56-v7 #1               
erifBold[ 1213.377979] Hardware name: BCM2835                                                                                                                                                                      
.ttf - 0[ 1213.382132] PC is at dma_buf_dynamic_attach+0x70/0x258                                        
 - 2 - 2[ 1213.388046] LR is at 0x0                                                                      
                                                                                                         
[1213.391315] pc : [<80794168>]    lr : [<00000000>]    psr: a80f0013                                    
[32;1m6b[ 1213.398373] sp : 851b9db0  ip : 85bd4f00  fp : 851b9dd4                                       
833ea0 1213.404373] r10: 00000004  r9 : 859a6cc0  r8 : 00000000                                          
0m] free[ 1213.410372] r7 : 82187810  r6 : 00000000  r5 : fffffff7  r4 : 85bd4ec0                        
type spu[ 1213.417690] r3 : 00000000  r2 : 00000000  r1 : 00000000  r0 : 85bd4ec0                        
[... cut ...]
[ 1213.786056] Exception stack(0x851b9fa8 to 0x851b9ff0)
[ 1213.791186] 9fa0:                   718a1160 00000004 00000004 80404a5b 6b3ce5f8 6b3ce5d8
[ 1213.799490] 9fc0: 718a1160 00000004 43464b4f 00000036 00000000 0000001b 6ce36098 0000001a
[ 1213.807789] 9fe0: 00000036 6b3ce5c0 76f07ba9 76e81ae6
[ 1213.812919]  r10:00000036 r9:851b8000 r8:80100264 r7:00000036 r6:43464b4f r5:00000004
[ 1213.820863]  r4:718a1160
[ 1213.823435] Code: 0a00004f e5d63000 e88400a0 e5c43018 (e5953010) 
[ 1213.829693] ---[ end trace 619244702d2e1552 ]---
```

EasyIP2023, 2022-12-10:
@TomaszAIR Thanks again for identifying this whenever you have time may you try https://github.com/EasyIP2023/meta-raspberrypi/tree/testing-two.

I'm able to build and see output without both x11 and opengl being defined in DISTRO_FEATURES

TomaszAIR, 2022-12-21:
@EasyIP2023 First of all, sorry that it took so long for me to check this out. Second of all.. Thank you! It is working now. I can confirm that using vlc compiled from https://github.com/EasyIP2023/meta-raspberrypi/tree/testing-two on my rpi0w2 with framebuffer only allow me to start video on screen connected via HDMI.

Are those changes already merged into master of meta-raspberrypi?

Also, got one minor issue. In my build I do not use libcamera so I cannot build if I do not remove `libcamera_%.bbappend` from dynamic-layers as it does not have recipe to use with. Can it be handled in some way?

EasyIP2023, 2022-12-26:
@TomaszAIR. The updates to the rpidistro-{vlc,ffmpeg} are merge into master.

`BBMASK` is your friend. https://docs.yoctoproject.org/ref-manual/variables.html?highlight=bbmask#term-BBMASK

TomaszAIR, 2022-12-27:
Thanks. I am not used to to use `BBMASK`. Will try that. Looks like it work so I think we can close that issue

closed 2022-12-27 by TomaszAIR