agherzan/meta-raspberrypi #1129: [Kirkstone] ERROR: Feature 'cfg/fs/vfat.scc' not found, this will cause configuration failures

jahanzaib1234, 2022-12-15:
i am building image for respberry pi4 and using kirkstone 4.04 branch and get this error.

`| DEBUG: Executing shell function do_kernel_metadata
| NOTE: do_kernel_metadata: for summary/debug, set KCONF_AUDIT_LEVEL > 0
| ERROR: Feature 'cfg/fs/vfat.scc' not found, this will cause configuration failures.
| ERROR: Check the SRC_URI for meta-data repositories or directories that may be missing
| ERROR: Set KERNEL_DANGLING_FEATURES_WARN_ONLY to ignore this issue
| WARNING: exit code 1 from a shell command.
ERROR: Task (/home/smart/Desktop/yocto_kirkstone/poky/smartGlass/build/../../../meta-raspberrypi/recipes-kernel/linux/linux-raspberrypi_5.15.bb:do_kernel_metadata) failed with exit code '1'
`

kraj, 2022-12-15:
Please try adding
```
KERNEL_FEATURES:remove = "cfg/fs/vfat.scc"
```
in `meta-raspberrypi/recipes-kernel/linux/linux-raspberrypi_5.15.bb`
and report back perhaps with a patch in a Pull request :)

agherzan, 2022-12-15:
Is that now part of some default cfg? Why was it removed? A reference to oe-core would be useful.

jahanzaib1234, 2022-12-16:
@kraj  by adding `KERNEL_FEATURES:remove = "cfg/fs/vfat.scc"` in 
`meta-raspberrypi/recipes-kernel/linux/linux-raspberrypi_5.15.bb` last error is removed 
`ERROR: Feature 'cfg/fs/vfat.scc' not found, this will cause configuration failures`

But during  `do_image_rpi_sdimg` i get this error 

`ERROR: core-image-full-cmdline-1.0-r0 do_image_rpi_sdimg: mcopy cannot copy /home/smart/Desktop/yocto_kirkstone/poky/smartGlass/build/tmp/deploy/images/raspberrypi4/bcm2708-rpi-zero.dtb into boot.img
ERROR: core-image-full-cmdline-1.0-r0 do_image_rpi_sdimg: ExecutionError('/home/smart/Desktop/yocto_kirkstone/poky/smartGlass/build/tmp/work/raspberrypi4-poky-linux-gnueabi/core-image-full-cmdline/1.0-r0/temp/run.do_image_rpi_sdimg.4189746', 1, None, None)
ERROR: Logfile of failure stored in: /home/smart/Desktop/yocto_kirkstone/poky/smartGlass/build/tmp/work/raspberrypi4-poky-linux-gnueabi/core-image-full-cmdline/1.0-r0/temp/log.do_image_rpi_sdimg.4189746
ERROR: Task (/home/smart/Desktop/yocto_kirkstone/poky/smartGlass/build/../../meta/recipes-extended/images/core-image-full-cmdline.bb:do_image_rpi_sdimg) failed with exit code '1'`

is that error linked with cfg/fs/vfat.scc. i am not making a pull request for last error because i want make it sure its working before making a pull request.

kraj, 2022-12-16:
The error is not related to missing `vfat.scc` it just means that we have mismatch of DTB named `bcm2708-rpi-zero.dtb`

It seems you are building form rpi4 32bit platform. in that case please try this with this patch
```diff
diff --git a/conf/machine/raspberrypi4.conf b/conf/machine/raspberrypi4.conf
index 86c57ed..b53a790 100644
--- a/conf/machine/raspberrypi4.conf
+++ b/conf/machine/raspberrypi4.conf
@@ -14,6 +14,13 @@ MACHINE_EXTRA_RRECOMMENDS += "\
     bluez-firmware-rpidistro-bcm4345c5-hcd \
 "

+RPI_KERNEL_DEVICETREE = " \
+    broadcom/bcm2711-rpi-4-b.dtb \
+    broadcom/bcm2711-rpi-400.dtb \
+    broadcom/bcm2711-rpi-cm4.dtb \
+    broadcom/bcm2711-rpi-cm4s.dtb \
+"
+
 # 'l' stands for LPAE
 SDIMG_KERNELIMAGE ?= "kernel7l.img"
 UBOOT_MACHINE = "rpi_4_32b_config"
```

jahanzaib1234, 2022-12-16:
@kraj okay i try with this patch

jahanzaib1234, 2022-12-28:
@kraj  still i get same error.

jahanzaib1234, 2022-12-30:
@kraj now i try to make image for pi4_64 to avoid previous error but during build i get this error.

`Loading cache: 100% |############################################| Time: 0:00:01
Loaded 4312 entries from dependency cache.
NOTE: Resolving any missing task queue dependencies

Build Configuration:
BB_VERSION           = "2.0.0"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "universal"
TARGET_SYS           = "aarch64-poky-linux"
MACHINE              = "raspberrypi4-64"
DISTRO               = "poky"
DISTRO_VERSION       = "4.0.4"
TUNE_FEATURES        = "aarch64 armv8a crc cortexa72"
TARGET_FPU           = ""
meta                 
meta-poky            
meta-yocto-bsp       
meta-raspberrypi     
meta-oe              
meta-multimedia      
meta-networking      
meta-python          
meta-filesystems     
meta-perl            
meta-qt5             
meta-security        
meta-rust            
meta-rpi             
meta-screeninfo      = "<unknown>:<unknown>"

Initialising tasks: 100% |#######################################| Time: 0:00:05
Sstate summary: Wanted 2048 Local 0 Mirrors 0 Missed 2048 Current 523 (0% match, 20% complete)
NOTE: Executing Tasks
WARNING: rust-hello-world-git-r0 do_fetch: URL: git://github.com/meta-rust/rust-hello-world.git;protocol=https does not set any branch parameter. The future default branch used by tools and repositories is uncertain and we will therefore soon require this is set in all git urls.
ERROR: linux-raspberrypi-1_5.15.82+gitAUTOINC+51e603e687-r0 do_compile: oe_runmake failed
ERROR: linux-raspberrypi-1_5.15.82+gitAUTOINC+51e603e687-r0 do_compile: ExecutionError('/home/smart/Desktop/yocto_kirkstone/poky/smartGlass/build/tmp/work/raspberrypi4_64-poky-linux/linux-raspberrypi/1_5.15.82+gitAUTOINC+51e603e687-r0/temp/run.do_compile.516533', 1, None, None)
ERROR: Logfile of failure stored in: /home/smart/Desktop/yocto_kirkstone/poky/smartGlass/build/tmp/work/raspberrypi4_64-poky-linux/linux-raspberrypi/1_5.15.82+gitAUTOINC+51e603e687-r0/temp/log.do_compile.516533
Log data follows:
| DEBUG: Executing shell function do_compile
| NOTE: KBUILD_BUILD_TIMESTAMP: Mon Dec 12 15:22:29 UTC 2022
| NOTE: make -j 4 CC=aarch64-poky-linux-gcc   -fuse-ld=bfd -fmacro-prefix-map=/home/smart/Desktop/yocto_kirkstone/poky/smartGlass/build/tmp/work/raspberrypi4_64-poky-linux/linux-raspberrypi/1_5.15.82+gitAUTOINC+51e603e687-r0=/usr/src/debug/linux-raspberrypi/1_5.15.82+gitAUTOINC+51e603e687-r0                      -fdebug-prefix-map=/home/smart/Desktop/yocto_kirkstone/poky/smartGlass/build/tmp/work/raspberrypi4_64-poky-linux/linux-raspberrypi/1_5.15.82+gitAUTOINC+51e603e687-r0=/usr/src/debug/linux-raspberrypi/1_5.15.82+gitAUTOINC+51e603e687-r0                      -fdebug-prefix-map=/home/smart/Desktop/yocto_kirkstone/poky/smartGlass/build/tmp/work/raspberrypi4_64-poky-linux/linux-raspberrypi/1_5.15.82+gitAUTOINC+51e603e687-r0/recipe-sysroot=                      -fdebug-prefix-map=/home/smart/Desktop/yocto_kirkstone/poky/smartGlass/build/tmp/work/raspberrypi4_64-poky-linux/linux-raspberrypi/1_5.15.82+gitAUTOINC+51e603e687-r0/recipe-sysroot-native=  -fdebug-prefix-map=/home/smart/Desktop/yocto_kirkstone/poky/smartGlass/build/tmp/work-shared/raspberrypi4-64/kernel-source=/usr/src/kernel -fdebug-prefix-map=/home/smart/Desktop/yocto_kirkstone/poky/smartGlass/build/tmp/work-shared/raspberrypi4-64/kernel-build-artifacts=/usr/src/kernel  --sysroot=/home/smart/Desktop/yocto_kirkstone/poky/smartGlass/build/tmp/work/raspberrypi4_64-poky-linux/linux-raspberrypi/1_5.15.82+gitAUTOINC+51e603e687-r0/recipe-sysroot LD=aarch64-poky-linux-ld.bfd    --sysroot=/home/smart/Desktop/yocto_kirkstone/poky/smartGlass/build/tmp/work/raspberrypi4_64-poky-linux/linux-raspberrypi/1_5.15.82+gitAUTOINC+51e603e687-r0/recipe-sysroot HOSTCC=gcc  HOSTCFLAGS=-isystem/home/smart/Desktop/yocto_kirkstone/poky/smartGlass/build/tmp/work/raspberrypi4_64-poky-linux/linux-raspberrypi/1_5.15.82+gitAUTOINC+51e603e687-r0/recipe-sysroot-native/usr/include -O2 -pipe HOSTLDFLAGS=-L/home/smart/Desktop/yocto_kirkstone/poky/smartGlass/build/tmp/work/raspberrypi4_64-poky-linux/linux-raspberrypi/1_5.15.82+gitAUTOINC+51e603e687-r0/recipe-sysroot-native/usr/lib                         -L/home/smart/Desktop/yocto_kirkstone/poky/smartGlass/build/tmp/work/raspberrypi4_64-poky-linux/linux-raspberrypi/1_5.15.82+gitAUTOINC+51e603e687-r0/recipe-sysroot-native/lib                         -Wl,--enable-new-dtags                         -Wl,-rpath-link,/home/smart/Desktop/yocto_kirkstone/poky/smartGlass/build/tmp/work/raspberrypi4_64-poky-linux/linux-raspberrypi/1_5.15.82+gitAUTOINC+51e603e687-r0/recipe-sysroot-native/usr/lib                         -Wl,-rpath-link,/home/smart/Desktop/yocto_kirkstone/poky/smartGlass/build/tmp/work/raspberrypi4_64-poky-linux/linux-raspberrypi/1_5.15.82+gitAUTOINC+51e603e687-r0/recipe-sysroot-native/lib                         -Wl,-rpath,/home/smart/Desktop/yocto_kirkstone/poky/smartGlass/build/tmp/work/raspberrypi4_64-poky-linux/linux-raspberrypi/1_5.15.82+gitAUTOINC+51e603e687-r0/recipe-sysroot-native/usr/lib                         -Wl,-rpath,/home/smart/Desktop/yocto_kirkstone/poky/smartGlass/build/tmp/work/raspberrypi4_64-poky-linux/linux-raspberrypi/1_5.15.82+gitAUTOINC+51e603e687-r0/recipe-sysroot-native/lib                         -Wl,-O1 -Wl,--allow-shlib-undefined -Wl,--dynamic-linker=/home/smart/Desktop/yocto_kirkstone/poky/smartGlass/build/tmp/sysroots-uninative/x86_64-linux/lib/ld-linux-x86-64.so.2 [... cut ...]
ERROR: Task (/home/smart/Desktop/yocto_kirkstone/poky/smartGlass/build/../../../meta-raspberrypi/recipes-kernel/linux/linux-raspberrypi_5.15.bb:do_compile) failed with exit code '1'
NOTE: Tasks Summary: Attempted 3380 tasks of which 2124 didn't need to be rerun and 1 failed.

Summary: 1 task failed:
  /home/smart/Desktop/yocto_kirkstone/poky/smartGlass/build/../../../meta-raspberrypi/recipes-kernel/linux/linux-raspberrypi_5.15.bb:do_compile
Summary: There were 2 WARNING messages.
Summary: There were 2 ERROR messages, returning a non-zero exit code.`

kraj, 2022-12-30:
sadly, I dont have a `kirkstone` setup handy, Can you try yo reproduce this on `master` branch as well ?

jahanzaib1234, 2023-01-02:
@kraj can i create new image with master branch ??

kraj, 2023-01-05:
Please try with https://github.com/agherzan/meta-raspberrypi/pull/1134 applied and report back

agherzan, 2023-01-11:
> okay

Did the above fix your issue?

jahanzaib1234, 2023-01-16:
> > okay
> 
> Did the above fix your issue?

i report back in 3 to 4 days because i move to dunfell layer for my project now if i test Kirkstone i need to make space in my system to test it after completing my current project i test it again and report back.

dYalib, 2023-07-18:
Hi, I stumbled across the same problem. 
The `cfg/fs/vfat.scc` is part of the [yocto-kernel-cache](https://git.yoctoproject.org/yocto-kernel-cache/tree/) repository . If you add this to the `SRC_URI` of the kernel, it works out of the box.

Here is my abbreviated kernel bbappend file (kirkstone version):

```
SRCREV_meta="0b65b80aa112614e8ab129f2d832b8cf050e7a4a"
SRC_URI+=\
  git://git.yoctoproject.org/yocto-kernel-cache;type=kmeta;name=meta;branch=yocto-5.10;destsuffix=kernel-meta \
"
```

If you don´t need the vfat support, you can just remove this like @kraj mentioned it [here](https://github.com/agherzan/meta-raspberrypi/issues/1129#issuecomment-1353586981) before.