agherzan/meta-raspberrypi #1137: Unable to see mini uart /dev/ttyS0 /dev/Serial1 on CM3+

newmedFabio, 2023-01-19:
Hello,
I see this issue on poky honister and poky kirkstone on a CM3+.

The problem is that  /dev/ttyS0 is not showing up. 
/dev/ttyAMA0 is working fine.
I've built a simply "core-image-minimal" system with only little modification to have a parallel display working with vc4-drm-v3d.
Adding/Removing ENABLE_UART="1" in local.conf does not change the behvoir.

dmesg doesn't tell anything about /dev/ttyS0

I have tested the same configuration with RaspberryPI OS and in that case /dev/Serial1 is working fine.
In order to have it work, I simply add to config.txt:
force_turbo=1
dtoverlay=uart1,txd1_pin=32,rxd1_pin=33_

This made me think that some configuration is wrong in meta-raspberry for cm3+.
According to Raspberry pi [documentation](https://www.raspberrypi.com/documentation/computers/configuration.html#configuring-uarts), on CM3+ uart1 should work without affecting bluetooth.

[image]

Can you please help me solve this issue? I don't know what to do anymore, I've carefully red many post but all didn't have a working  solution.
Thanks!

newmedFabio, 2023-01-20:
Looking more into this, raspi-gpio get gives:
`GPIO 32: level=1 fsel=0 func=INPUT
GPIO 33: level=1 fsel=0 func=INPUT`

It seems that
`force_turbo=1
dtoverlay=uart1,txd1_pin=32,rxd1_pin=33`
in config.txt has been ignored. Why?

Also, ENABLE_UART does not change the behavoir because I add a custom config.txt after flashing and according to meta-raspberry conf files, ENABLE _UART should only add enable_uart=1 in config.txt

newmedFabio, 2023-01-25:
Are my question bad posted? Can you please help me solving this?
I think that minuart malfunction on CM3+ can be of common interest, espacially if with raspberrypi OS is perfectly working...

agherzan, 2023-01-25:
As you can see there seem to be very few people using CM3+ these days. I would start by investigating the version differences between RaspberryPi OS and meta-raspberrypi. Maybe we miss some fixes? Have you tried the `master` branch?

newmedFabio, 2023-01-30:
Thank you for your reply Andrei.
I will soon follow your suggestion to try with the master branch, I have not tried yet.

I'm also trying to compare kernel .config between my distro and a Raspberry OS. I know I can get it from /proc/config.gz for raspberry OS but I don't know how to get it from a simple core-image-minimal build since that file is missing. Do you know another way to get kernel .config?

agherzan, 2023-01-31:
You can enable it in the kernel configuration 

https://cateee.net/lkddb/web-lkddb/IKCONFIG_PROC.html

newmedFabio, 2023-02-01:
Thank you Andrei.
In order to see /proc/config.gz I simply ended up to modprobe its module with:
modprobe configs

I also performed a kernel configuration comparison, finding nothing relevant different from Raspberrypi OS.
Then I struggled to run "vcdbg log msg", since it is a propetary software and a few more libraries are needed (libdebug_sym.so.0, libelftoolchain.so, libvcos.so.0).

All this to find out that... **uart1.dtbo was missing!** 
I made some stupid error in layer configurations and was not included in the distro.
Really big big shame on me!!

Thanks for your support, I close the thread.

closed 2023-02-01 by newmedFabio

agherzan, 2023-02-01:
Happy to hear you've figured it out.

carlonluca, 2024-06-24:
@newmedFabio do you remember what error caused this problem? I also just noticed I'm missing files from the overlay directory, in particular uart1.dtbo. Thanks!

newmedFabio, 2024-06-26:
Hello Luca,
the main issoue was that /dev/ttyS0, which was the second serial interface, was missing from /dev/ folder.

carlonluca, 2024-06-26:
Yes, that is clear. But do you remember why the uart1.dtbo file was not there? My understanding is that most dtbo files must be listed explicitly, which they are already there in Raspberry OS.

newmedFabio, 2024-06-27:
I don't know why it was not there, I suppose it has something to do with meta-raspberry choices.
Maybe it has been excluded from core-image-minimal.

To verify I used:
'
bitbake -e core-image-minimal | grep ^RPI_KERNEL_DEVICETREE_OVERLAYS
'

If it is not listed, you can add it creating a simple  linux-raspberrypi_%.bbappend and wrinting inside:
'
RPI_KERNEL_DEVICETREE_OVERLAYS:append = " \
       overlays/uart1.dtbo \
'

carlonluca, 2024-06-27:
Yes, that is what I did as well. Thanks.