agherzan/meta-raspberrypi #1138: Problem encountered: xlib based GLX requires at least one gallium driver

kkettinger, 2023-01-20:
When disabling VC4GRAPHICS with `DISABLE_VC4GRAPHICS = "1"` in local.conf i get this build error:

```
| ../mesa-22.2.3/meson.build:555:6: ERROR: Problem encountered: xlib based GLX requires at least one gallium driver
|
| A full log can be found at /home/devel/projects/example/kas/build/tmp/work/cortexa7t2hf-neon-vfpv4-example-linux-gnueabi/mesa-gl/2_22.2.3-r0/build/meson-logs/meson-log.txt
| ERROR: meson failed
| WARNING: exit code 1 from a shell command.
```

Build configuration:
```
Build Configuration:
BB_VERSION           = "2.2.0"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "universal"
TARGET_SYS           = "arm-steppermess-linux-gnueabi"
MACHINE              = "raspberrypi4"
DISTRO               = "steppermess"
DISTRO_VERSION       = "1.0"
TUNE_FEATURES        = "arm vfp cortexa7 neon vfpv4 thumb callconvention-hard"
TARGET_FPU           = "hard"
meta-chromium        = "master:7442631fab308c965bf9833a1a2eadeca7b66610"
meta-clang           = "master:d89e06ad94a46f6810d0a8787004b71b8ecaf87d"
meta-filesystems
meta-gnome
meta-initramfs
meta-multimedia
meta-networking
meta-oe
meta-python
meta-xfce            = "langdale:c354f92778c1d4bcd3680af7e0fb0d1414de2344"
meta-raspberrypi     = "langdale:6f5771d2bcfbfb8f8ce17b455c29a5703f2027c9"
meta-rauc            = "langdale:e3c08045b5ea7bb66612b7e43c21ecf230a1e869"
meta-poky            = "langdale:e911b760d279774d8ab24529a2ffd82c02976feb"
meta                 = "langdale:5a622a9171da941c705aedc88d487d7158955a8f"
```

agherzan, 2023-01-20:
The fix is in master. Would you be up for a cherry pick + a MR?

kkettinger, 2023-01-20:
Great, can you give me the commit hash?

agherzan, 2023-01-20:
I actually was wrong. We agreed that this should be fixed upstream. Checkout https://github.com/agherzan/meta-raspberrypi/pull/1119

kkettinger, 2023-01-20:
I have just created a bbappend file `recipes-graphics/mesa/mesa-gl_%.bbappend` which does the trick:
```
PACKAGECONFIG += "gallium"
PACKAGECONFIG:class-target += "gallium"
```
Thanks for your help.

closed 2023-01-20 by kkettinger