agherzan/meta-raspberrypi #1140: Image works only on one of two identical RPIs

kkettinger, 2023-01-24:
I'm building an image with yocto and meta-raspberrypi, which worked fine until I built the last image.

On one RPI it boots fine, but on the other I get this error at boot:
[image]

It seems that something is getting into the u-boot console and writing random stuff into it, preventing it from booting.

Raspbian works fine on both Raspberries.

Here is the build configuration:
```
Build Configuration:
BB_VERSION           = "2.2.0"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "universal"
TARGET_SYS           = "arm-custom-linux-gnueabi"
MACHINE              = "raspberrypi4"
DISTRO               = "custom"
DISTRO_VERSION       = "1.0"
TUNE_FEATURES        = "arm vfp cortexa7 neon vfpv4 thumb callconvention-hard"
TARGET_FPU           = "hard"
meta-chromium        = "master:f2d5539552b54099893a7339cbb2ab46b42ee754"
meta-clang           = "master:36d2b65f81542ea5e36664cfdc4d1b5763b9ba4b"
meta-filesystems
meta-gnome
meta-initramfs
meta-multimedia
meta-networking
meta-oe
meta-python
meta-xfce            = "langdale:c354f92778c1d4bcd3680af7e0fb0d1414de2344"
meta-raspberrypi     = "langdale:6f5771d2bcfbfb8f8ce17b455c29a5703f2027c9"
meta-rauc            = "langdale:e3c08045b5ea7bb66612b7e43c21ecf230a1e869"
meta-custom       = "main:c52eec681f642a60387ff578720bf4c874094a6d"
meta-poky            = "langdale:c7f82c7604c3af32635b8330f3edd4508e576fb3"
meta                 = "langdale:b2dfb2fe86de887d0239078ded0a4f9e8c677d83"
```

kkettinger, 2023-05-16:
I found out that one RPI was defective, i've applied too high voltage on one pin (12V). It still boots fine without U-Boot. It can be closed, sorry for the issue.

closed 2023-05-16 by kkettinger

agherzan, 2023-05-16:
How come they were both working with Raspbian though?

kkettinger, 2023-05-16:
My best guess is the UART is faulty and interrupts the boot process in u-boot by spitting out random chars, which u-boot picks up and enters the prompt (U-Boot>).
Do you know what raspbian is using as bootloader? I think it might be grub, but i'm not sure.