agherzan/meta-raspberrypi #1144: pps device on raspberry

epasinetti, 2023-01-27:
I'm trying to add pps device to my raspberry, using "Raspberry Pi OS" it is just needed to add this line in boot/config.txt
dtoverlay=pps-gpio,gpiopin=18

doing the same with RPI_EXTRA_CONFIG  variable in local.conf, 

RPI_EXTRA_CONFIG = ' \n \
dtoverlay=pps-gpio,gpiopin=18\n \
init_uart_baud=9600\n \
'

the text was succesfully added in config.txt but I can't see pps0 under dev.

Am I doing something wrong ?

agherzan, 2023-01-27:
Have you checked for support in the kernel? We may miss the kernel driver for it.

epasinetti, 2023-01-29:
Sorry I' dont understand your question, is there a solution to bring pps support on raspberry yocto ? 
Thank you

agherzan, 2023-01-29:
Take a look at https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/tree/drivers/pps/Kconfig

epasinetti, 2023-01-30:
sorry I'm not an expert compiling kernel, is there a way to use pps directly implemented with config.txt configuration ? 
Can you add the support in meta-raspberrypi ?

agherzan, 2023-01-31:
It is challenging to find time for that soon but having this ticket created is one step forward. I encourage you though to dig a bit into it and maybe end up with a PR yourself.