agherzan/meta-raspberrypi #1145: do fetch failing due to checksum mismatch

aliraza1944, 2023-01-30:
**Description**

When I try to build an image after specifying the image type in local config as rpi-sdimg. The following files fail to be fetched.

`IMAGE_FSTYPES = "tar.xz ext3 rpi-sdimg"`

It looks like the checksums are mismatched. The expected checksum is

`c687aa1b5127a8dc0773e8aefb1f009f24bf71ccb4c9e8b40a1d46cbbb7bee0c`  (mentioned in the raspberrypi-firmware.bb file)

whereas the one I get is

`702f10a5adca063df0d903af1e54c4925a6c5ab89cee5f33e42f3e17a386660f`

**Steps to reproduce the issue:**
1. In local.conf set IMAGE_FSTYPE to `rpi-sdimg`
2. Run `bitbake core-image-base`

**Describe the results you received:**
`Initialising tasks: 100% |##################################################################################################################| Time: 0:00:03
Sstate summary: Wanted 117 Local 78 Network 0 Missed 39 Current 806 (66% match, 95% complete)
Removing 1 stale sstate objects for arch acu_rpi4_64: 100% |################################################################################| Time: 0:00:00
NOTE: Executing Tasks
WARNING: rpi-bootfiles-20210421-r3 do_fetch: Checksum mismatch for local file /home/aliraza/acu-rpi-yocto/yocto_downloads/raspberrypi-firmware-2ac4de4eaac5c1d1b25acec4a5e0a9fdb16f0c91.tar.gz
Cleaning and trying again.
WARNING: rpi-bootfiles-20210421-r3 do_fetch: Renaming /home/aliraza/acu-rpi-yocto/yocto_downloads/raspberrypi-firmware-2ac4de4eaac5c1d1b25acec4a5e0a9fdb16f0c91.tar.gz to /home/aliraza/acu-rpi-yocto/yocto_downloads/raspberrypi-firmware-2ac4de4eaac5c1d1b25acec4a5e0a9fdb16f0c91.tar.gz_bad-checksum_702f10a5adca063df0d903af1e54c4925a6c5ab89cee5f33e42f3e17a386660f
WARNING: rpi-bootfiles-20210421-r3 do_fetch: Checksum failure encountered with download of https://github.com/raspberrypi/firmware/archive/2ac4de4eaac5c1d1b25acec4a5e0a9fdb16f0c91.tar.gz;downloadfilename=raspberrypi-firmware-2ac4de4eaac5c1d1b25acec4a5e0a9fdb16f0c91.tar.gz - will attempt other sources if available
ERROR: rpi-bootfiles-20210421-r3 do_fetch: Fetcher failure for URL: 'https://github.com/raspberrypi/firmware/archive/2ac4de4eaac5c1d1b25acec4a5e0a9fdb16f0c91.tar.gz;downloadfilename=raspberrypi-firmware-2ac4de4eaac5c1d1b25acec4a5e0a9fdb16f0c91.tar.gz'. Checksum mismatch!
File: '/home/aliraza/acu-rpi-yocto/yocto_downloads/raspberrypi-firmware-2ac4de4eaac5c1d1b25acec4a5e0a9fdb16f0c91.tar.gz' has sha256 checksum '702f10a5adca063df0d903af1e54c4925a6c5ab89cee5f33e42f3e17a386660f' when 'c687aa1b5127a8dc0773e8aefb1f009f24bf71ccb4c9e8b40a1d46cbbb7bee0c' was expected
If this change is expected (e.g. you have upgraded to a new version without updating the checksums) then you can use these lines within the recipe:
SRC_URI[sha256sum] = "702f10a5adca063df0d903af1e54c4925a6c5ab89cee5f33e42f3e17a386660f"
Otherwise you should retry the download and/or check with upstream to determine if the file has become corrupted or otherwise unexpectedly modified.
ERROR: rpi-bootfiles-20210421-r3 do_fetch: Fetcher failure for URL: 'https://github.com/raspberrypi/firmware/archive/2ac4de4eaac5c1d1b25acec4a5e0a9fdb16f0c91.tar.gz;downloadfilename=raspberrypi-firmware-2ac4de4eaac5c1d1b25acec4a5e0a9fdb16f0c91.tar.gz'. Unable to fetch URL from any source.
ERROR: Logfile of failure stored in: /home/aliraza/acu-rpi-yocto/build/tmp/work/acu_rpi4_64-poky-linux/rpi-bootfiles/20210421-r3/temp/log.do_fetch.2004924
ERROR: Task (/home/aliraza/acu-rpi-yocto/sources/meta-raspberrypi/recipes-bsp/bootfiles/rpi-bootfiles.bb:do_fetch) failed with exit code '1'
NOTE: Tasks Summary: Attempted 2886 tasks of which 2775 didn't need to be rerun and 1 failed.

Summary: 1 task failed:
  /home/aliraza/acu-rpi-yocto/sources/meta-raspberrypi/recipes-bsp/bootfiles/rpi-bootfiles.bb:do_fetch
Summary: There were 4 WARNING messages shown.
Summary: There were 2 ERROR messages shown, returning a non-zero exit code.
`

**Describe the results you expected:**
Smooth build process

**Additional information you deem important (e.g. issue happens only occasionally):**
meta-raspberrypi repo is on commit `b4ec97e4eb8e36efd1f7e2f8ae020a9e55cfc239`

agherzan, 2023-01-31:
@aliraza1944 seems like an annoying Github issue. I've seen people reporting this lately. Would you be able to push a patch?

agherzan, 2023-01-31:
Also, this is on master?

agherzan, 2023-01-31:
I suspect this is fixed - https://lore.kernel.org/git/df7b0b43-efa2-ea04-dc5b-9515e7f1d86f@gmail.com/T/#m656f88c00f1250b5b5bb7fa1ad7b3fecd3592d79

aliraza1944, 2023-01-31:
I was going to say the same, found this last night. 

https://github.blog/changelog/2023-01-30-git-archive-checksums-may-change/

closed 2023-01-31 by aliraza1944